libplctag/libplctag #6: undefined reference to 'plc-tags_create' ... in Code::Blocks (C++)

reyanvaldes, 2015-02-20:
Please it is a very nice work. I love it. I only has one challenge. I have tried to compile and link with Code::Blocks using C++, where I have all classes of our software created, but I got an error undefined reference to 'plc-tags_create'. I am pointing to the correct library after make install. I have tried unsuccessfully many things.

Please can you help me,

Thanks

kyle-github, 2015-02-21:
Hi,

Perhaps it is a typo, but "plc-tags_create" is not one of the API functions.  It looks like I somehow managed to leave some of the C++ definition code incomplete except on Windows!

What OS are you using?

The way I would suggest working with C++, until I fix the library, is to do the following:

1) build the library as C and install it.
2) include libplctag.h like this in your C++ code:

```c
extern "C" {
#include <libplctag.h>
}
```

C++ has different linkage (due to name mangling) than C, so when you are calling a C function from C++, you must make sure that the function declaration has extern "C" in front of it.

Have you been able to build any of the examples?  

I will make sure that it works without hacks from C++ this weekend.  

Thanks for the report!

reyanvaldes, 2015-02-21:
Kyle 
Thank you for your support and quickly response!. I am using Ubuntu 14.04 LTS, 32 bits.

Excellent! I tried already. Now with the extern "C", it is able to compile and linked without error!. Next week I will start testing your API with one AB PLC using Ethernet/IP, and let you know.

Kyle when we use plc_tag_create, it means create only the link to the tag in the PLC in the user program? Or does it actually created the tag in the PLC? In my case I want to read/write tags already existing in the AB PLC. So should I do following calling?:
for preparing:
----------------
- plc_tag_create
- while loop plc_tag_status

for reading:
--------------
- plc_tag_read
- plc_tag_get

for finishing:
----------
- plc_tag_destroy

Can I use plc_tag_create/destroy only one time when open/close comm? I have an idea to use vector of plc_tag to allow me handle tags list easily. Just sharing with you, I believe It would be nice to create C++ class driver as wrapper of your API for C++ developer and include in your example. Maybe I can try to develop some simple Driver example and you can use or improve as part of package. Are you ok with it? I was thinking to develop a class with following methods and simplify operations like reading/writing tag: openComm,closeComm, addTag, delTag, addTagsList, delTagsList, readTag, writeTag, getError. The reagTag/writeTag will have overloads for different data types. If you like we can discuss in detail this simple idea. Please let me know?

Thank you so much for support and nice work!!

Looking forward to heard from you

Reyan

closed 2015-02-21 by reyanvaldes

kyle-github, 2015-02-21:
I checked in some code to the head branch.   This should automatically make
the library header file work for C++ code.  If you get a chance, I would
appreciate it if you could try it.  If you write a good C++ wrapper, I
would love to include it in the wrappers I have.

The library does not create tags in the PLC.  Perhaps I should have used a
different name for that function.

What model of PLC are you using?

kyle-github, 2015-02-21:
The examples directory has several examples that show reading/writing different types of AB PLCs as well as multithreaded and single threaded examples.  The API can be synchronous or async, as you wish.

You first open a tag to the PLC (this does a lot of work that is not visible).  This is done with plc_tag_create().  Once open, you can call plc_tag_read/write as much as you want.  When you are finally done with the tag, you call plc_tag_destroy() to clean up the resources.   We have systems that keep tags open for weeks.  

Note that when you read and write, the plc_tag_read/plc_tag_write functions only bring in/out data from/to the PLC.  In order to get the data out of the tag, you still need to call the plc_tag_get_XintY functions (for read) or plc_tag_set_XintY functions for write.   Or you can call the floating point functions.  These functions move data in and out of the buffer in memory in the PC.  The read/write functions get the raw (bytes) from/to the PLC.

The read/write functions take *byte* offsets!  So, be careful.  Look carefully at the examples.  You can see where I multiply by a constant to read DINTs in a loop, for instance.

Note that AB PLCs (and possibly others) have very limited resources for Ethernet connections.  An ENBT module only has 64 (IIRC) TCP connection slots.  So, unless there is really some need, you should probably leave the default TCP connection sharing in place.

The function pair plc_tag_create()/plc_tag_destroy() will set up and tear down the connection to a PLC only if they are the first and last calls to set up any tags to a specific PLC.  If you open two tags on a PLC, the second call to plc_tag_create() will reuse the TCP socket used by the first (unless you add in a parameter preventing sharing in the string defining the tag).  Similarly, the plc_tag_destroy() function will not close the TCP socket until the last tag is closed.  All the resource control happens automatically for you.

Most of the outline of setting up comms etc. is not needed.  All of that happens inside of plc_tag_create.  What is returned is effectively an object in C with a few methods/functions you can call.

A C++ wrapper could set up the tag as a sort of bytestream with >> and << operators or something to make getting/setting data easier.

reyanvaldes, 2015-02-22:
Kyle
Thank you for in-deep information, and I appreciate a lot your work. About the simple cpp example, it worked fine using make outside the IDE but when I tried with CodeBlocks shown same 'undefined reference to' <API's function name>. I had to put extern "C" you recommended to allow build it. The reason I am using CodeBlocks is because all the software and other drivers I have developed already on it. I will try to make a wrapper from my point of view to make it simple for my own use for your beautiful API and also to share with you. 

I will tell you AB PLC type on Monday, because the PLC is at the office. 

I have a couple of questions or topics:
- Please when you mentioned "So, unless there is really some need, you should probably leave the default TCP connection sharing in place.", do you means on PLC side? or do you means the first common part of TAG_PATH for all tags?

-  I don't know what is going on behind the scene, maybe the question is related with something very hard to use. Please Is it possible to open one PLC connection kind of open one socket on regular Ethernet TCP/IP, and use/create multiple tags link (or tags connection) to read/write tags that share the same PLC connection? Or when using plc_tag_create if you don't specify the protocol means shared same PLC connection. The reason I asked you is because it will fit with the same philosophy when working with other drivers that I have, for example with modbus tcp where you open the connection to device only one time, and have multiple reading/writing for different registers, and at the end close the connection to release resources used for it. Can I use same philosophy with your API. If is not possible, it is ok, just thinking about some ideas. 

- The plc_tag_create demands a lot of computing effort? Kyle still I have my doubts how handle a lot of tags, for example > 100 tags with your API that need to read/write from/to PLC. Maybe I have two options, (a) or (b). Which one as your consideration could be better for performance and also considering resources? Please can you help me clarify it: 
 a) Do plc_tag_create at the beginning for all tags (100 tags), only one time, when OpenComm. And after on every read/write use the combination of the plc_tag_read/plc_tag_get_XintY. And when CloseComm call plc_tag_destroy for all tags.

 b) Do nothing at beginning. And after on every read/write use the combination of the plc_tag_create/plc_tag_read/plc_tag_get_XintY/plc_tag_destroy. My concern here is with performance.

- The offset you mentioned is basically used for the arrays, correct? If it is a single tag will be offset of 0, correct? For example if I am reading a single tag named temperature, the TAG_PATH "protocol=ab_eip&gateway=192.168.1.42&path=1,0&cpu=LGX&elem_size=4&elem_count=1&name=temperature", and plc_tag_get_float32 (tag, 0);  

- If the tag I want to read is individual element of the arrays, can I use ".... &name=temperature[2]"? 

I was thinking in general for the wrapper class use following methods as starting point. Maybe you would like to talk or discuss about it at front during the design of it to make it good for you and for me also:
- OpenComm (protocol, cpu, ip, path, timeout)
- CloseComm()
- AddTag ( name, elemsize, count) [ for array]
- AddTag (name, type, count) [ for array, with tag type explicitly entered]
- AddTag (name, elemsize) [for single tag]
- AddTag (name, type) [for single tag, with tag type explicitly entered]
- GetTagElemCount (name) // get how many elements has the tag
- DelTag (name) // delete or destroy the tag from tag list and tag connections
- DelTagList ()
- ReadTag (name) 
- WriteTag (name); 
- GetTagValue (name, index) [for array] - use stream or overload for different types
- GetTagValue (name) [for single tag]  - use stream or overload for different types
- SetTagValue (name, index, value) [for array] - use stream or overload for different types
- SetTagValue (name, value) [for single tag] - use stream or overload for different types
- GetTagPath (name); // used inside the cip class to get the entire tag path
- GetTagsTotal(); // get total tags defined on tag list

* Attributes
   - string commpath; // just include the open comm parameters
   - int timeout; 
   - vector <tciptag> taglist; // ordered by name to allow binary searching by name later on

struct tciptag { // Tag Data Structure used on vector class on CIP driver
   string tagname; // simple tag name
   string tagpath; // entire tag path including communication
   bool  linkdone; // indicate if the tag was created successfully or not
   int     elemsize; // it is element size: bytes in PLC memory used for the tag type
   bool  isarray;  // allow validate some operations only applicable when uses get or set with index
   long  count; // get total of elements. in case of array will be total of elements, otherwise 1
};

Please see at end example that attempt to mimic simple.c. Below are some ideas:

  - Call OpenComm (protocol, cpu, ip, path, timeout) only one time, and store these parameters as attribute of the class. This will allow to set the completely tag path every time AddTag is called.

- Where CloseComm will delete tag list and its content, and release all resources.

- Then the method AddTag ( name, elemSize, count) for array, and for single tag just AddTag (name, elemSize). [It will set the count =0]. Where name is just an individual tag name, for example "temperature", and we can use the name as simple of way, and the [... cut ...]
Please notice that also OpenComm do nothing with communication, only prepare communication path. But maybe will be better if at front this can do more work related with PLC communication.

Please this is just draft design or initial idea. If we have a good detail definition of wrapper, and responsibilities of each method and its attributes next week I can start implementation on it and testing also, but if possible I would like to discuss with you at front to review the design conception.

Thanks

kyle-github, 2015-02-22:
See inline.

> Kyle
> Thank you for in-deep information, and I appreciate a lot your work. About
> the simple cpp example, it worked fine using make outside the IDE but when
> I tried with CodeBlocks shown same 'undefined reference to' . I had to put
> extern "C" you recommended to allow build it. The reason I am using
> CodeBlocks is because all the software and other drivers I have developed
> already on it. I will try to make a wrapper from my point of view to make
> it simple for my own use for your beautiful API and also to share with you.

Hmm, that is odd.  I will need to load CodeBlocks and try it.  I have not
used CodeBlocks.  The sequence I put into the .h file is the standard one
to make the contents work correctly with both C and C++, so there must be
something else I am doing wrong.

> I will tell you AB PLC type on Monday, because the PLC is at the office.
>
> I have a couple of questions or topics:
>
>    -
>
>    Please when you mentioned "So, unless there is really some need, you [...]
The library looks at the gateway (PLC) IP and port.  If there is another
tag that uses the same gateway, then it will reuse that open TCP channel.
This is the default behavior to keep the resource usage low.

This happens automatically.  If you want to use the same TCP channel, just
make sure all the tags you open with plc_tag_create() use the same IP and
port.    There is a special param you can set, per tag, to prevent that tag
from sharing TCP connections.

So,

plc_tag tag1 = plc_tag_create("...gateway=1.2.3.4&...name=my_tag");
plc_tag tag2 = plc_tag_create("...gateway=1.2.3.4&...name=another_tag");
plc_tag tag3 = plc_tag_create("...gateway=1.2.3.4&...name=another_tag2");

Will open one TCP channel to the PLC at 1.2.3.4.  This channel will be used
for all three tags.   When the last tag using the TCP channel is closed,
the TCP channel will be closed too.

> -
>
>    I don't know what is going on behind the scene, maybe the question is
>    related with something very hard to use. Please Is it possible to open one
>    PLC connection kind of open one socket on [...]
Yes, the sharing happens automatically.  However, this is a different model
than you have below because you do not need to implement some of the
objects/classes you have.  The only object needed is for each individual
tag.

The library keeps track of things that can be shared internally and reuses
what it can.

The library is designed to support multiple protocols.  In fact, while
there is some sharing, there is a lot of difference between the C*Logix
code and the PLC 5 code.  The first protocol I am going to add next is
ModBus TCP.  The API will not change but the params in the tag string will.

> -
>
>     -
>
>    The plc_tag_create demands a lot of computing effort? Kyle still I
>    have my doubts how handle a lot of tags, for example > 100 tags with your
>    API that need to read/write from/to [...]
Creating a tag is not that "heavy" on the PC side.  However, creating a new
TCP connection is fairly heavy on the PLC side, so I would do a).  That is
what we do.
[... 237 lines cut ...]
stable across several fairly large internal changes of the library and our
applications have not had to change even when we added things like Windows
support, PLC 5 support, read caching etc.

C does not allow optional parameters, so if I had parameters that
controlled something that normally did not need to be set (such as sharing
or not sharing a TCP channel), they would always have to be there in the
API function signatures.   And, if I had to add a new parameter, I would
have to change the API and thus all my applications.

The drawback with using a single string as the parameters for a tag is that
there is no type checking.

You are welcome to build whatever class API you wish, but I wanted to share
the pain that I went through before I got to the API I have now :-)

Most of our work is actually in Java now using a wrapper very similar to
the one that is in the source in the wrappers directory.  It has an OO API
that is really just a slim wrapper around the C API.

This discussion has been very useful to me.  I am opening another issue for
the C++ wrapper so that this does not get lost.

Best,
Kyle

reyanvaldes, 2015-02-24:
Thank you for your recommendations, I appreciate them. Later when I am able to read/write some tags, and advance on it, I will share with you other ideas related with c++ wrapper, which should be organized in different way based on the driver base class  developed as base for all drivers. 
I tried today to read some tag but apparently I am doing something wrong or missing some parameter. I opened an issue to follow it. 
Thanks for your kindly attention and help
Reyan