libplctag/libplctag #7: C++ Wrapper discussion

kyle-github, 2015-02-22:
This issue is being opened to gather ideas on a C++ wrapper.  Feel free to add to this!

kyle-github, 2015-02-22:
To kick this off, please read issue #6.

One other way to do this would be to make an API similar to that used in Java.  Something like this:

```c++
class Tag {
public:
    Tag(const std::string attribs);

    int read(void);
    int read(int timeout);
    int write(void);
    int write(int timeout);

   int8_t getInt8(int byteOffset);
   uint8_t getUInt8(int byteOffset);
   ...
   
}
```

It would be nice to have some convenience methods that would allow you to treat a tag like a stream:

```c++
Tag tag("protocol=....");

int32_t first, second, third;

tag >> first >> second >> third;
```

So something like 

```c++
Tag& operator>>(int32_t &arg);
```

It would be really convenient to have the following be possible:

```c++
Tag tag("protocol=....");
int32_t foo;

tag.read(500);
foo = tag[42];
foo++;
tag[42] = foo;
// or just tag[42] = int32_t(tag[42])++;
tag.write(500);
```

This would require a semi-hidden proxy class to make this work. It could be really useful.  I do not think that you could do the commented out line without the cast.  This could easily be a source of bugs...

It also would not be possible to use this with UDTs in general, but that may be moot as other protocols like ModBus TCP do not support structured data.

Alternatively, it may make more sense to have specific subclasses of Tag, or other wrappers, for things like DINT arrays etc.  This could work:

```c++
DINTTag tag("protocol=....");

tag.read(500);
for(int i=0; i < tag.numElements(); i++) {
    tag[i] = tag[i] ++;
}
tag.write(500);
```

Only a few such wrapper classes using Tag would be needed to provide a lot of benefit.  I think I like this approach more than trying to add on complicated proxies and trying to make the core Tag class do  more.  This spreads out the complexity and thus the cost.  You only pull in what you need.

reyanvaldes, 2015-03-03:
Kyle
I will send you email with the simple wrapper for C++ that fit the software I have. First of all this is my way to appreciate your beautiful work and nice attention. Please feel free to use it and comments. You will see it is simple wrapper but a little different on the approaching based on the drivers structure that I already have developed for other protocols.

Thank you so much

kyle-github, 2015-10-21:
I am going to move this to the mailing list.  I want to find out if there are any other C++ users.

The mailing list is at:

[libplctag](https://groups.google.com/group/libplctag)

Please sign up.  I would rather have the discussion there.

Best,
Kyle

closed 2015-10-21 by kyle-github

reyanvaldes, 2015-10-21:
Kyle
Ok, thank you