libplctag/libplctag #8: Addressing tags with : in the tag name

rh2ygit, 2015-02-23:
Hi, I'm trying to read some tags in a Control Logix PLC, which are not global in scope. I was previously able to read them using TuxEIP, addressing the tags with names such as:

Program:Supervisor.pidLocal_PV
Program:Zone1.flow_CV
etc.

How can I address these tags with libplctag? When I try to read them, I get error -33 returned. Reading other controller-wide tags works fine. 

Thanks!

kyle-github, 2015-02-24:
Can you please modify the tag string to include "&debug=1", capture the output and add that to the issue?  This will print out everything that is going back and forth between the PLC and the PLC (and a lot more).

-33 means that the PLC returned an error to the library.  It is possible that there is a bug in the parsing, but that will show up in the packets.

Thanks for the bug report!

rh2ygit, 2015-02-24:
Thanks, I will add the &debug=1 tomorrow and report back. By the way, THANKS for actively developing this much needed library!

kyle-github, 2015-02-24:
I had another thought.  What CPU and firmware version are you running?  I thought I read somewhere that AB has started to enforce the restrictions on tag access more.

I also have a question to ask: what is the next protocol that I should add?  I am leaning toward Modbus TCP.  There are some internal simplifications and changes I want to make to the library first.

kyle-github, 2015-02-24:
Ah, in looking in the code, I am not handling ":" characters at all!  

Oops!  

Can you copy the debugging output of TuxEIP too?  I need to see how to encode that.  My guess is that I encode it much like I handle "." which is just skipping over it and putting down the remaining parts as ASCII names, but I need to make sure.  AB is not the most consistent...

kyle-github, 2015-02-24:
I found the TuxEIP code and looked at how they are doing it.  It looks like they do not handle colons at all!  So, maybe it is copied directly into the name.  I modified my code and checked it in to GitHub.  Please give that a try.

I also saw another difference: they handle multiple values within arrays but I do not.  

TuxEIP: fooTag[12,42]
libplctag: fooTag[12][42]

I may modify libplctag if it is enough of a problem.

rh2ygit, 2015-02-24:
Ok, I had to modify cip.c to also allow ':' characters in the case NAME: clause of the switch/case in cip_encode_tag_name(), but now it works!

I'm no expert at github, so if you want to make the change shown below and check it in, that would be great!

 while(*p) {
        switch(state) {
            case START:
                /* must start with an alpha character. */
              if(isalpha(*p) || (*p == '_') || (*p == ':')) {
                rutabaga++;
                state = NAME;
              } else if(*p == '.') {
                state = DOT;
              } else if(*p == '[') {
                state = ARRAY;
              } else {
                return 0;
              }
              
              break;

            case NAME:
                *dp = 0x91; /* start of ASCII name */
                dp++;
                name_len = dp;
                *name_len = 0;
                dp++;

                while(isalnum(*p) || (*p == '_') || (*p == ':')) {  /// ADD COLON SUPPORT HERE TOO
                    *dp = *p;
                    dp++;
                    p++;
                    (*name_len)++;
                }

[ ... ]

kyle-github, 2015-02-24:
D'oh, I forgot the second one!  I will make sure that that is in GitHub.  Give me a little while, I'm on a conference call.

kyle-github, 2015-02-24:
OK, I updated GitHub.  Can you please repull and do the test again?  

Thanks again for the bug report!

rh2ygit, 2015-02-24:
Done. New code pull works like a charm. Thanks!

kyle-github, 2015-02-24:
I will close this now.  Thanks for the bug report!

closed 2015-02-24 by kyle-github