libplctag/libplctag #10: Next feature discussion

kyle-github, 2015-02-24:
For the next changes to the library there are several features I am considering.

* Modbus TCP - just read/write registers for now.  But, it may be simple to add the other things.
* read-only produced tags.  This is a big bandwidth saver if all you are doing is reading some data over and over from the PLC.    The trick here is that there does not seem to be any complete public documentation about how these work.
* multi-tags.  AB's protocol has the ability to read more than one tag at the same time per request.  This would lower the number of network back-and-forth trips a lot.

There are a few other things that I would like to fix too:

* the initial TCP connect() can delay several seconds and block everything in some circumstances (right after the Ethernet cable is pulled on the PLC).  That needs to be changed to be the same as the timeout.
* simplify some of the internals of the library.  It would be much nicer to have one thread per tag, but I think the resource overhead would be far too big.  I can go half-way on that and keep one thread or perhaps one thread per TCP channel.

Please comment.

rh2ygit, 2015-02-24:
Multi-tags support would help a lot -- My primary use of this library is for data collection from a set of PLCs on the network. Any reduction in network traffic can only help increase the rate at which data can be collected.  Thanks.

johanfforsberg, 2015-08-17:
I have a similar use case. The ability to read many tags per request would be useful.

Support for "produced" tags would be even better, if I understand correctly how they work. That would require some setup on the PLC to produce particular tags, right?

By the way, if you haven't seen it, there is a pure python library that supports communication with AB PLCs: https://github.com/pjkundert/cpppo

reyanvaldes, 2015-08-17:
Thank you for share, yes it is kind of challenge now for handling implicit messages. 

Thanks

johanfforsberg, 2015-08-18:
Cool. I might be interested in working on a python wrapper for libplctag, but multi-tag reading (or possibly implicit msgs) is pretty much a requirement for my use case as we need to monitor large numbers of tags reasonably quickly.

kyle-github, 2015-08-29:
Can you please open a new issue for reading multiple tags at once?

I am slowly digging my way out of about two months worth of backlog due to travel and conferences.  I will try to address this more over the weekend.

Best,
Kyle

kyle-github, 2015-08-29:
Thanks for the pointer to the Python library.  It can simulate a PLC!  That is really helpful.  It is costly to have a PLC just for debugging.  Very useful.

kyle-github, 2015-09-01:
Please see the discussion I posted to issue #15 about reading multiple tags.  There are some issues that make it difficult to decide how to proceed.  I would appreciate feedback.

kyle-github, 2015-10-21:
I have moved all discussion to a Google group:

[libplctag](https://groups.google.com/forum/#!forum/libplctag)

Please continue there.

closed 2015-10-21 by kyle-github