libplctag/libplctag #9: Reading error from PLC LOGIX 5573

reyanvaldes, 2015-02-24:
Hi Kyle
I am using Logix 5573, ip address: 192.168.1.207. I created different tags in a very simple program, one tag for each data type, and tried to read Real, it is version of simple.c. I am able to ping PLC. I setup the connection string as shown below and tried to read a tag named 'Real' as float value but I am getting an error message "ERROR: Unable to read the data! Got error: " -5 (PLC_TAG_ERR_BAD_PARAM). Please can you help me to identify what I am doing wrong. Quickly question for Logix 5573 the cpu name I have to use is LGX, correct? Thanks

Please see below the configuration: 
[image]

Please see code below:

extern "C" {
#include "libplctag.h"
}

#include <unistd.h>

#define TAG_PATH "protocol=ab_eip&gateway=192.168.1.207&path=1,0&cpu=LGX&elem_size=4&elem_count=1&name=Real"
#define ELEM_COUNT 10
#define ELEM_SIZE 4
#define DATA_TIMEOUT 5000

int testingAB_CIP() // Testing Allen Bradley PLC
{
   plc_tag tag = PLC_TAG_NULL;
    int rc;
    int i;

    // create the tag
    tag = plc_tag_create(TAG_PATH);

    // everything OK?
    if(!tag) {
        fprintf(stderr,"ERROR: Could not create tag!\n");

        return 0;
    }

    // let the connect succeed we hope
    while(plc_tag_status(tag) == PLCTAG_STATUS_PENDING) {
    	sleep(1);
    }

    if(plc_tag_status(tag) != PLCTAG_STATUS_OK) {
    	fprintf(stderr,"Error setting up tag internal state.\n");
    	return 0;
    }

    // get the data
    rc = plc_tag_read(tag, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d\n",rc);

        return 0;
    }
    cout << plc_tag_get_float32(tag,0);

    // we are done
    plc_tag_destroy(tag);
};

int main(int argc, char *argv[])
{
 testingAB_CPI()
return 0;
}

kyle-github, 2015-02-24:
Can you please turn on debugging and capture the output?

Modify the tag string to:

"protocol=ab_eip&gateway=192.168.1.207&path=1,0&cpu=LGX&elem_size=4&elem_count=1&name=Real&debug=1"

(debug=1 at the end)

Unfortunately, bad param errors are not very useful.  I really should do something about that...

Once you have the output, please attach it to this issue and I can look at it to see what is going on.  It may let me know what part is considered a bad parameter.  That seems a bit strange to get that error that late.  By the time you are doing a read, everything is already parsed.

kyle-github, 2015-02-24:
I put up issue #10 for discussion of next features in the library.  I would appreciate if you would look over the list and comment.   Thanks!

reyanvaldes, 2015-02-24:
Kyle please below is the screen shot. Thanks
[image]

kyle-github, 2015-02-24:
I will try to debug from the screen shot, but it would be very helpful if you could capture the output into a text file and paste that into the issue tracker.  Thanks!

kyle-github, 2015-02-24:
I can see the part of the packet that is the converted tag name:

```
03  91  04  52  65  61  6c  
^   ^   ^   ^   ^   ^   ^
|   |   |   |   |   |   | 
|   |   |   |   |   |   +-- l
|   |   |   |   |   +------ a
|   |   |   |   +---------- e
|   |   |   +-------------- R
|   |   +------------------ length in bytes
|   +---------------------- Hex 91 means start string name segment
+-------------------------- 3 = 3 16-bit words for IOI/EPATH segment length, not counting this byte.
```

Hmm, this looks correct. So why is the PLC complaining about the IOI path?

Can you either rename this tag to something other than a type name (i.e. not DINT, INT, SINT, REAL etc.) and try again or make another tag and try it?  I wonder if there is a limitation on reading tags that have the same name as types?  I have not run into such a problem before.

Alternatively, maybe this is not the part of the path that is the problem.  What slot is the CPU in?  The path for that you have is 1,0 and that means "go out the back plane port on the Ethernet module, go to slot 0 on the back plane."  The 1 part is correct I think, but the 0 part might not be.

IOI/EPATH really only tells us that the PLC doesn't like something about how the request is being routed or the tag is named.  I assumed it was the tag name, but maybe it is not.

Go through and double check the whole path:

gateway=192.168.1.207
path=1,0
cpu=LGX
elem_size=4
elem_count=1
name=Real

Based on your first screen shots, this looks correct.  

I will get a hold of a PLC as soon as I can and test this too.  Please try to the above things and see if that makes a difference.

kyle-github, 2015-02-25:
I got a ControlLogix L61 system.  

I created a tag named "Real" of type "REAL".

I changed the value from 0.0 to 42.0.

I used the tag_rw example program and read the tag.

I do not get the error.  I got 42.0.

I will try to recreate this by copying the code you have above.

kyle-github, 2015-02-25:
Hmm, you definitely have found a bug.  I might take a bit to find this one...  Give me a few hours.

I modified the simple.c and simple_cpp.cpp examples to read a float named "Real" just like you have.  I enabled debug.  simple.c works and simple_cpp.cpp does not.  I get different packets generated for the request.  It looks like just one byte is different.

kyle-github, 2015-02-25:
Huh.  My mistake.  I accidentally made the count=11 rather than 1 and I got an error.  I include simple_cpp.cpp here:

```c++
#include <stdio.h>
#include <unistd.h>
#include "../lib/libplctag.h"

#define TAG_PATH "protocol=ab_eip&gateway=10.206.1.39&path=1,0&cpu=LGX&elem_size=4&elem_count=1&name=Real&debug=1"
#define ELEM_COUNT 1
#define ELEM_SIZE 4
#define DATA_TIMEOUT 5000

/*
 * There really isn't much to this example other than it being compiled
 * by g++ instead of gcc.  This is just to test that the API functions are
 * correctly exported for C++ code.
 */

int main(int argc, char **argv)
{
    plc_tag tag = PLC_TAG_NULL;
    int rc;
    int i;

    /* create the tag */
    tag = plc_tag_create(TAG_PATH);

    /* everything OK? */
    if(!tag) {
        fprintf(stderr,"ERROR: Could not create tag!\n");

        return 0;
    }

    /* let the connect succeed we hope */
    while(plc_tag_status(tag) == PLCTAG_STATUS_PENDING) {
    	sleep(1);
    }

    if(plc_tag_status(tag) != PLCTAG_STATUS_OK) {
    	fprintf(stderr,"Error setting up tag internal state.\n");
    	return 0;
    }

    /* get the data */
    rc = plc_tag_read(tag, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d\n",rc);

        return 0;
    }

    /* print out the data */
    for(i=0; i < ELEM_COUNT; i++) {
        fprintf(stderr,"data[%d]=%f\n",i,plc_tag_get_float32(tag,(i*ELEM_SIZE)));
    }

    /* now test a write */
    for(i=0; i < ELEM_COUNT; i++) {
    	float val = plc_tag_get_float32(tag,(i*ELEM_SIZE));

    	val = val+1.0;

    	fprintf(stderr,"Setting element %d to %f\n",i,val);

    	plc_tag_set_float32(tag,(i*ELEM_SIZE),val);
    }

    rc = plc_tag_write(tag, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d\n",rc);

        return 0;
    }

    /* get the data again*/
    rc = plc_tag_read(tag, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d\n",rc);

        return 0;
    }

    /* print out the data */
    for(i=0; i < ELEM_COUNT; i++) {
        fprintf(stderr,"data[%d]=%f\n",i,plc_tag_get_float32(tag,(i*ELEM_SIZE)));
    }

    /* we are done */
    plc_tag_destroy(tag);

    return 0;
}

```

This works fine.  It correctly reads the value from the PLC, increments it, writes it, and reads it back again.  I made sure that simple_cpp.cpp is compiled with g++, not gcc.

I am going to tackle your code next.

kyle-github, 2015-02-25:
The program you put in the example did not survive cut-and-paste very well.  I got it to compile.  The program I ran was:

```c++
extern "C" {
#include "libplctag.h"
}

#include <stdio.h>
#include <unistd.h>
#include <iostream>

#define TAG_PATH "protocol=ab_eip&gateway=10.206.1.39&path=1,0&cpu=LGX&elem_size=4&elem_count=1&name=Real&debug=1"
#define ELEM_COUNT 10
#define ELEM_SIZE 4
#define DATA_TIMEOUT 5000

int testingAB_CIP(void) // Testing Allen Bradley PLC
{
	plc_tag tag = PLC_TAG_NULL;
	int rc;

	// create the tag
	tag = plc_tag_create(TAG_PATH);

	// everything OK?
	if(!tag) {
	    fprintf(stderr,"ERROR: Could not create tag!\n");

	    return 0;
	}

	// let the connect succeed we hope
	while(plc_tag_status(tag) == PLCTAG_STATUS_PENDING) {
	    sleep(1);
	}

	if(plc_tag_status(tag) != PLCTAG_STATUS_OK) {
	    fprintf(stderr,"Error setting up tag internal state.\n");
	    return 0;
	}

	// get the data
	rc = plc_tag_read(tag, DATA_TIMEOUT);

	if(rc != PLCTAG_STATUS_OK) {
	    fprintf(stderr,"ERROR: Unable to read the data! Got error code %d\n",rc);

	    return 0;
	}

	std::cout << "got data: " << plc_tag_get_float32(tag,0) << std::endl;

	// we are done
	plc_tag_destroy(tag);

	return 0;
}

int main(int argc, char *argv[])
{
	testingAB_CIP();
	return 0;
}

```

The command line I used to compile this was:

```
g++ -o test test.cpp -fno-strict-aliasing -g -I../lib -fms-extensions -Wall -I/usr/local/include -L../lib -lplctag -lpthread -pthread
```

I don't think you need the -fms-extensions part.  The part that seems to include pthreads twice was needed on some platform, but I can't remember if it is still needed.  Maybe it was due to an older version of g++?

I was able to read the value of the "Real" tag in the PLC with this.

Please check this against your code.  Are there possibly some differences in the compiler settings?

reyanvaldes, 2015-02-25:
Kyle 
Thank you so much for all your effort and help with the troubleshooting of it. Today I was very busy in another project, I couldn't advance on this as I would like with more testing. I will try to do tomorrow. If I understand correctly what you recommend is:
- check path <nbr1>,<nbr2>: could you please remind me what means <nbr1>,<nbr2>. Maybe is here the problem. I will check first thing in the morning the current hardware configuration.

-  I will review the codeblocks compiler options to make sure there is no special compiler setting.
 
I will let you know first thing tomorrow morning.
Thanks
Reyan

reyanvaldes, 2015-02-25:
Kyle good news!
You was right, the issue was with the path!. I mistake the second parameter has to match with the cpu slot. Still I am not sure about first path parameter. When I changed to 11, where the cpu slot is, it read the real data correctly. 

Thank you so much for your support. I will do a test for different tags and start organize the idea for the class wrapper as per driver base I have. I will share with you the idea, although maybe will be little different of java wrapper conception.

Thanks.

kyle-github, 2015-02-25:
Glad it worked.  I will put more information in the project Wiki pages about the path.  At least as much as I understand it.  The publicly available documentation (not much) is not very clear on exactly how they work.

I actually cheat a bit in the library by breaking out the path into two pieces: the gateway and the "path" elements.  If you were setting up a MSG instruction, these would be in one EPATH/IOI path instead of two pieces.  Perhaps I should have done that originally.

reyanvaldes, 2015-02-25:
Yes, the path seems very detail. I had to work with other project, so I only tested successfully: Real, Int, Dint, but at the end when I was trying to read Sint, it didn't return any value, nothing, just empty string, not sure what happened. When I am back to the office today, no matter what time, I will keep testing the sint, and boolean. Quickly question how you handle the Boolean? and it has elemsize = 1 (byte), correct even if I we use only one bit? Do you have any plc_tag_get for boolean or should I use the get_int8 and compare to get the boolean result? The other that maybe we need in future is 64 bits Dint/real. 
I will let you know today late how was test.
Thanks for your excellent support.

kyle-github, 2015-02-25:
Booleans end up coming back as SINT values (8 bits).  Reading individual booleans seems to work fine.  I can't remember if boolean true == SINT 1 or SINT 127.  I think it is 1.

Hmm, I will check to make sure that I did not break SINT values at some point.  We certainly read them in our production deployments.

kyle-github, 2015-02-25:
Oops, I forgot that there is an example program to toggle a boolean value.  It reads the value, changes it and writes it back.  Look for toggle_bool.c in the examples.

kyle-github, 2015-02-25:
OK, I retested.  When you read a boolean you get all bits zero or all bits one in an 8-bit byte.  If you want a signed value, you will get 0 for false and -1 for true.  If you want an unsigned value you will get 0 for false and 255 for true.

By the way, I test with the tag_rw tool in the examples.  It lets you enter the tag string on the command line and print out the results.  I find it is a good way to make sure you have the IP address and path etc. right first before you try to write your own code.

kyle-github, 2015-02-25:
I also checked SINT values.  They work fine.  As with booleans, elem_size=1.

reyanvaldes, 2015-02-25:
Kyle
The problem was with the cout << plc_tag_get_int8 (tag4,0) << end; because the compiler was interpreting it is a character, and try to print like that, that is why didn't show any characters. Now after get it,  I moved first to integer and then print it (was 14 for Sint, -1 for bool). shows OK for both sint and bool. As you said the bool has -1 as int8. Also I tested with Array[i]. So we are reading excellent !!, Thanks!, on Friday I will test writing with all data types and starts the design of class wrapper, and give you feedback of it. After I will do test using class.
Thank you so much

closed 2015-02-25 by reyanvaldes

reyanvaldes, 2015-02-26:
So far for the connection path it has following parameters (cpu-related):
- protocol: ab_eip or ab_cip
- gateway: ip
- cpu slot: #
- path: 1 <hard coded to 1 always?>, <cpu slot>
- cpu type: lgx (for Logix). I will check for different cpu which cpu identification is used on API.

I will keep you posted.
Thanks

reyanvaldes, 2015-03-02:
Hi Kyle
I hope you are doing well.
I was out from Wednesday last week doing other projects with related with Siemens/AB. It is just to let you know that I am back in the testing process and start working on the wrapper class. As soon as I have the wrapper ready and testing I will share with you as my way to consider and appreciate your work.

I have a couple of questions: 
1- I have noticed that the ip address you have it as gateway parameter name, but gateway usually is using for router. Is it for a particular reason? Do you have ip and gateway parameters, or only gateway (where we entered IP)?

2- The first parameter of path number indicate: 1- backplane, 2- Ethernet/IP, 3- Serial. Is it correct?

3- As summary, the API path parameter has two numbers separated by comma. The first one indicate "connection way" and second slot number.It is correct? 

4- Although as path it is just small portion of the truly completed path parameter. Are you supporting real path from AB, for example using following format: port, next address, port, next address?
 
Thank you so much

kyle-github, 2015-03-02:
Hi Reyan,

The gateway was originally to make the C code a little simpler.  I was trying to keep things similar for different protocols like Modbus TCP.  I called it a gateway because that is really what it is for AB.  You can enter any kind of other AB network using a PLC as a gateway.  You can hop from Ethernet to ControlNet.  You can even go to Data Highway.

A "real" AB path would be something like 10.1.2.3,1,4,192.168.1.2,1,13.

That would be:

* create an initial TCP connection to 10.1.2.3
* on that Ethernet module, route the connection to the backplane (1) (there can be multiple "ports" per module, but on an Ethernet module, port 1 is the backplane)
* on the backplane, route the connection to slot 4.
* in the module in slot 4, route to 192.168.1.2 via TCP.  This is now on a completely different network.
* in that module at 192.168.1.2, route out to the backplane.
* on the backplane, route to the module in slot 13.

I just peeled off the first IP and called it a gateway.

I do not support most of the AB IOI/EPATH types yet.  It is not particularly hard to do, but I will need docs or code that does it to figure it out.  There are multiple types of path segment that can go there.  We did not need the other types yet.  

I will possibly pull support for Ethernet to DH+ because there is so little of it left in the world.  There are several places in the code that have to know about it and I could simplify things it I remove it.

Let me try to answer your questions:

1) As I mentioned above, a PLC in an AB network _is_ a router.  So, the term gateway is fairly accurate.

2) Er, not really.  The path is an IOI/EPATH (I see both referenced some what interchangeably even though I think they are probably a bit different).  The first number is the port on the Ethernet module out which the connection will go.  For instance a DH+ module has at least three ports.  1 is the back plane, 2 is DH channel A, and 3 is DH channel B.  

3) the examples only show direct connections to PLCs, not using a PLC as a gateway.  So there are no examples of longer paths.

4) not yet.  I may change this if I can find sufficient code/documentation and get a good test bench.  Right now, all I have is a single PLC.  The rest are all in production :-)

kyle-github, 2015-03-08:
Hi Reyan,

I saw that you emailed me your C++ wrapper.  I think I accidentally erased it or otherwise lost it :-(

Could you please send it again?

kyle-github, 2015-03-08:
Ah, wait, I found it.  I must have an old email address embedded somewhere in the docs...  I need to update that.   Where did you find that email address?

reyanvaldes, 2015-03-08:
Hi Kyle
If i remember correctly I found email in the headers (.h) I appreciate all your help and nice attention. I already integrated cip into another class named drivercip that contains cip object + driver virtual methods ( or standard methods) and tasks related with db, and in general does:
- reading configuration of cip channel from sqlite db.
- Reading tags configuration (name, type, scaling info if needed) from sqlite db
- the reading frames or buffer [with readtag]
- parsing frame or get tag value. [with get]
- store into the sqlitedb with saving seq. 

This match, thanks God with the driver base, and so far everything is smooth, and looks simple. Next week probably I will work wrapper for opc ua and next for siemens. If you like I can share some of the work of this with you. If likes also I can send you what I have as driver base, it is simple base or class for all drivers. And so far everything can be adapt or integrated into this framework. It is something simple but so far works, will see with opc ua. 
Thank you so much
Please if possible I would like to keep intouch with you

Bests

kyle-github, 2015-03-08:
I am happy to hear that the library is useful for you.

I tried to reply to your direct address, but I got a 500 error that the
address was "in quarantine".  Please let me know if you got my other
emails.  Thanks!

If you have other protocols that you have good code for, could you send me
links?  I want to make my library support as many as possible.  I do not
have any Siemens equipment to test with.

reyanvaldes, 2015-03-08:
Hi Kyle
Last email I received from you was the one you mentioned found wrapper class. Please feel free to use in case of problem my other email account: reyanvaldes@yahoo.com

About protocols links, sure. I will send you shortly all library links I have for different protocols: modbus tcp, can bus,j1587, opc ua, siemens. The Siemens library seems also beautiful work made by Italian eng. I didn't have chance to test it yet, but will do in few weeks and let you know. 
 
Kyle I believe Siemens 1200, cost around $230, the starter kit which include TIA portal and license is around $450, and it has very easily to setup as modbus tcp server or for small applications also. 

Best regards
Reyan

reyanvaldes, 2015-07-17:
Hello Kyle
Please does your library support unsolicited messages from AB PLC, I means allow open socket and listening clients (PLC) and read message (package)? In other words, instead of pulling data, the PLC is pushing it. If doesn't would you please recommend something to address this properly as per your experience. Thank you so much for any insight.

kyle-github, 2015-07-18:
Hi Reyan,

What you are talking about are implicit messages.  The library does not
support those yet.  I know of no open source library that does.

It is a feature I am working on.  Slowly :-(

There are some internal changes I need to make to the library that must be
finished before I can got implicit messages to work. I think I have enough
information on how to do them to complete the changes.

I do not have an ETA for these changes.

Best,
Kyle

reyanvaldes, 2015-07-20:
Hi KyleThank you for information. I really appreciate your help.
Please I would like to consult you one little challenge I have to read string tag. I need a function to read strings, so I was thinking to include it in the wrapper (written in C++), following same general idea and error checking the others get function you created but with copy bytes from t->data to the buffer to get string value.
//*********************************************************char* plc_tag_get_string (plc_tag t, int offset, int count){ //<same idea that other get but using t->data and copy the content of t->data into a buffer based on count and offset to get string>  plc_tag t;  ...
/* is there data? */ if(!t->data) { t->status = PLCTAG_ERR_NULL_PTR; return res; }
...}
But when I tried to build it I couldn't get compiling successfully and return an error on the line trying to parsing t->data pointer, even if the libplctag.h is part of the including. It shows an error "invalid use of incomplete type 'struct plc_tag_t'". 

//*********************************************************** 
Please do I am missing something. Would you recommend any idea to address this.
Thank a lot for any insightReyan

kyle-github, 2015-07-21:
Hi Reyan,

The plc_tag struct is an opaque pointer in libplctag.h.  It is only defined
in another file that is "private" to the implementation.  It is done that
way precisely to prevent people from getting to it :-)  This is not to make
life harder for you, but to make sure that if I decide to completely change
the implementation of a tag "under the hood", that I won't break your code.

There are two problems that adding a string function would face:

1) strings are represented many ways (zero-terminated, leading length...)

2) even within something like an AB PLC, I found that the size of a string
could change depending on the context.  Within a user-defined struct, it
could be smaller than 88 bytes.  And, people can define their own
string-like structs.

Those are the reasons I left it out.

reyanvaldes, 2015-07-21:
Kyle
Thank you for explanation, makes sense. The protection way works perfect :-((( 

Please, if possible, would you consider in near future to include in the library the reading of string tag, to be more completed.  It is important for the application we are using. Maybe handling max length & string type as parameters in the Get (for example:0- null terminating, 1-leading length in 32 bits). 

Thank you for any input and excellent support, even if it is not possible, you have been done great work with library

Thanks
Reyan

kyle-github, 2015-07-22:
Could you open another issue on GitHub?  This discussion in buried in a
closed issue.  I want to make sure other people can find it.

I am travelling in India right now, so my responses may be a bit late.

Let's talk about possible designs in the new issue.  Thanks.

Best,
Kyle

KutapiAise, 2016-08-29:
Hi Kyle and Reyan,
Hope you are doing fine. I have tested your library using contrologix. It works fine for tags of type INT, DINT,BOOL, REAL. Can you tell me how to deal with BOOL arrays?. I have checked for DINT arrays , the library works fine with the C++ wrapper as well. However, working with BOOL array is throwing the error PLCT_AG_ERR_REMOTE_ERR. I cant seem to get through this issue. Please provide me with suggestion or example.

Best,
Aishwarya

reyanvaldes, 2016-08-29:
Hi AishwaryaThank you for contacting me. Actually I am simple user  of this great library developed by Kyle that kindly shared with community, but let me try to help you a little bit. It sounds to me that could be range, size or naming issue, because you mentioned individual tags reading worked great .  The example of DINT arrays provided  I think is good starting point, If I were you, I would modify a little bit this example to works with BOOL arrays instead of DINT arrays, which is similar but as we know the items of arrays are BOOL (size of 1 byte) instead of DINT (size of 4 bytes), and plc_tag functions to use are plc_tag_get_int8 / plc_tag_set_int8 for reading /writing. Another important key is the tag path when the tag is created, where is specified its element size and total of elements, this has to match with the definition of tag.
Please you can use this example: https://github.com/kyle-github/libplctag/blob/master/examples/simple_cpp.cpp
Please change on (I avoided mention other changing to make it simple for you):- line #6 (Update IP, and PLC attributes properly, and very important element size and count in tag path, and BOOL array name): #define TAG_PATH "protocol=ab_eip&gateway=192.168.1.42&path=1,0&cpu=LGX&elem_size=1&elem_count=10&name=myBOOLArray"- line #7 (Total of items of the BOOL array): ELEM_COUNT 10.  This has to match with the elem_count used on line #6 and the actual size of the BOOL Array. In other words in this case it is assuming will read BOOL array defined as myBOOLArray [10] (BOOL) in PLC.- line #8:  ELEM_SIZE 1 
- Replace in all lines where it is used the plc_tag_get_int32 by plc_tag_get_int8.- Replace the plc_tag_set_int32 by plc_tag_set_int8. - In line #62 (if I remember correctly the BOOL in AB is defined as 1 as true, 0 as false, so 1- val should switch value): val = 1 - val;
I hope It should works. Please let me know if I can help with something else.
Thanks

KutapiAise, 2016-08-30:
Hi Reyan/ Kyle, 
I had earlier conducted my tests in line with your suggestions.And today i retested after reading your input! bool size as 1 (elem_size=1)and array count(elem_count=1024). However, i fail to get through the error. Finally, i set the debugger to get  command line debug info. Have attached the debug info:
[image]

Here is the code:

#define TAG_PATH "protocol=ab_eip&gateway=192.168.2.243&path=1,3&cpu=LGX&elem_size=1&elem_count=4&name=data4&debug=1"
#define ELEM_COUNT 4
#define ELEM_SIZE 1
#define DATA_TIMEOUT 5000

int _tmain(int argc, _TCHAR* argv[])
{
	

  plc_tag tagbool = PLC_TAG_NULL;
    int rc;
    int i;

    /* create the tag */
    tagbool = plc_tag_create(TAG_PATH);

    /* everything OK? */
    if(!tagbool) {
        fprintf(stderr,"ERROR: Could not create tag!\n");

        return 0;
    }

    /* let the connect succeed we hope */
    while(plc_tag_status(tagbool) == PLCTAG_STATUS_PENDING) {
    	_sleep(1);
    }

    if(plc_tag_status(tagbool) != PLCTAG_STATUS_OK) {
    	fprintf(stderr,"Error setting up tag internal state.\n");
    	return 0;
    }

    /* get the data */
    rc = plc_tag_read(tagbool, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d\n",rc);

        return 0;
    }

    /* print out the data */
    for(i=0; i < ELEM_COUNT; i++) {
        fprintf(stderr,"data[%d]=%d\n",i,plc_tag_get_int32(tagbool,(i*ELEM_SIZE)));
    }

    /* now test a write */
    for(i=0; i < ELEM_COUNT; i++) {
    	int8_t val = plc_tag_get_int8(tagbool,(i*ELEM_SIZE));

    	val =1- val;

    	fprintf(stderr,"Setting element %d to %d\n",i,val);

    	plc_tag_set_int8(tagbool,(i*ELEM_SIZE),val);
    }

    rc = plc_tag_write(tagbool, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d\n",rc);

        return 0;
    }

    /* get the data again*/
    rc = plc_tag_read(tagbool, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d\n",rc);

        return 0;
    }

    /* print out the data */
    for(i=0; i < ELEM_COUNT; i++) {
        fprintf(stderr,"data[%d]=%d\n",i,plc_tag_get_int32(tagbool,(i*ELEM_SIZE)));
    }

    /* we are done */
    plc_tag_destroy(tagbool);

    //char* charTime = "";/*ctime(&result);*/
 

	//long vl=testingDINTCIP("ab_eip", "192.168.2.243", "LGX", 1, 3, "data2");
	//bool test = testingBOOLCIP("ab_eip", "192.168.2.243", "LGX", 1, 3, "MBTU_EnMBTCP",charTime);

	getchar();
	return 0;
}

kyle-github, 2016-08-30:
Hi,

Are you sure of the size of your tag, data4?  Is it an array of 32 booleans?

When reading a boolean array, the PLC will pack the results.  You need to
send an element count that is the number of boolean bits divided by 8 and
rounded up.  In your text you mention 1024, but the code shows 8x4=32.

Is your PLC CPU in slot 3?  This would be the fourth slot from the left.
The first one is slot 0.  Then next is 1 etc.

I would suggest using the tag_rw tool to get the correct tag attribute
string.  It is a lot easier to use that instead of recompiling all the time.

Your example above would be .\tag_rw.exe -t int8 -p "
protocol=ab_eip&gateway=192.168.2.243&path=1,3&cpu=LGX&
elem_size=1&elem_count=4&name=data4&debug=1"

Is the tag a controller tag?

Can you try to read just the first bit to make sure the path and other
parts are all correct?

Best,
Kyle

KutapiAise, 2016-08-30:
Hi kyle
thank you for your probe in to my mistakes!.I have run the "tag_rw.exe" as per your recommendation. I have enclosed the outputs:
[image]
[image]
A you can see the controller path and the slot both are correct !!
And i get output for DINT array, single BOOL values.

On another note apologies for my mistake with the code. My BOOL array is defined as per this list:
[image]

[image]

kyle-github, 2016-08-30:
The problem is in your tag attribute string:

32 bits = 4 bytes => elem_count=4
1024 bits = 128 bytes => elem_count=128

The element count is the number of bits divided by 8 (rounded up to the
nearest byte).  You are printing it out as DINTs.  You could print it as
int8, int16 or even floats if you like pain :-)  The type you select to
output is not related to the actual element size.  For boolean arrays, the
element size is always 1.

The element count is something I wish I could skip, but the AB protocol
requires it

It looks like your tag attribute string to read MBTU_MB_0xx should be:

"protocol=ab_eip&gateway=192.168.2.243&path=1,3&cpu=LGX&elem_size=1&elem_count=128&name=MBTU_MB_0xx&debug=1"

Best,
Kyle

KutapiAise, 2016-08-31:
Kyle...Thank you very much for your prompt responses!!
And i hope to get an insight in to my issue. Though i did check with your suggestions, it seems that there isn't a way through this issue. i checked,  the tag is a controller tag array.
This is my latest debug output.I'm not able to decode the dumping bytes, albeit i could gain understanding about other parts!!
[image]

Best 
Aishwarya

kyle-github, 2016-08-31:
AArgh.  I completely forgot.  For some reason, I keep forgetting how to
read arrays of booleans.

Sigh.  The element size is not one byte, it is four.  So,

1024 bools -> 32 words of size 4 ->

"protocol=ab_eip&gateway=192.168.2.243&path=1,3&cpu=LGX&
elem_size=4&elem_count=32&name=MBTU_MB_0xx&debug=1"

Please try that.

The answer was on the forum:

https://groups.google.com/forum/#!forum/libplctag

Please send further responses to the forum. It is easier for people to find
answers.  Thanks!

I should probably make a Wiki page about this as it is really not obvious.

Best,
Kyle

KutapiAise, 2016-09-01:
kyle , thank you very much for this last input. I think i was close to analysing this nature of bool tag array.
I got the out put!!
Currently i'm testing with micrologix 1100. Will keep the queries posted in the google forum from here on.
Thank you very much for your support once again!!
Best,
Aishwarya

rgowda703, 2024-04-02:
hi, kyale im trying to connect micro850 plc using libpltag libraray on windows platform, but im getting error, plz let me know how to solve the issuses.
#include <stdio.h>
#include <stdlib.h>
#include "libplctag.h"

#define TIME 5000
#define REQUIRED_VERSION 2,1,0

int main() {
    int status;
    const char* tag = "protocol=ab-eip&gateway=192.168.1.215&path=1,0&elem_size=4&cpu=LGX&elem_count=1&name=Data2";

    // Initialize the library
    /* check the library version. */
    if (plc_tag_check_lib_version(REQUIRED_VERSION) != PLCTAG_STATUS_OK) {
        fprintf(stderr, "Required compatible library version %d.%d.%d not available!", REQUIRED_VERSION);
        exit(1);
    }
    plc_tag_set_debug_level(PLCTAG_DEBUG_DETAIL);

    // Create a tag
    int tag_id = plc_tag_create(tag, TIME);
    if (tag_id < 0) {
        printf("Error creating tag: %s\n", plc_tag_decode_error(tag_id));
        plc_tag_shutdown();
        return 1;
    }

    // Read the value from the input tag
    status = plc_tag_read(tag_id, TIME);
    if (status != PLCTAG_STATUS_OK) {
        printf("Error reading tag: %s\n", plc_tag_decode_error(status));
        plc_tag_destroy(tag_id);
        plc_tag_shutdown();
        return 1;
    }

    // Retrieve and print the value
    uint32_t value = plc_tag_get_uint32(tag_id, 0);
    printf("Value: %d\n", value);

    // Clean up
    plc_tag_destroy(tag_id);
    plc_tag_shutdown();

    return 0;
}
this is my code, 
and my output is:2024-04-02 14:46:55.568 thread(1) tag(0) INFO plc_tag_create_ex:838 Starting
2024-04-02 14:46:55.583 thread(1) tag(0) INFO initialize_modules:177 Starting.
2024-04-02 14:46:55.586 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2024-04-02 14:46:55.586 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-04-02 14:46:55.586 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-04-02 14:46:55.586 thread(1) tag(0) INFO initialize_modules:207 Initializing library modules.
2024-04-02 14:46:55.587 thread(1) tag(0) INFO lib_init:136 Starting.
2024-04-02 14:46:55.587 thread(1) tag(0) INFO lib_init:140 Setting up global library data.
2024-04-02 14:46:55.587 thread(1) tag(0) INFO lib_init:142 Creating tag hashtable.
2024-04-02 14:46:55.587 thread(1) tag(0) INFO hashtable_create:75 Starting
2024-04-02 14:46:55.587 thread(1) tag(0) INFO hashtable_create:99 Done
2024-04-02 14:46:55.587 thread(1) tag(0) INFO lib_init:148 Creating tag hashtable mutex.
2024-04-02 14:46:55.587 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-04-02 14:46:55.587 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-04-02 14:46:55.587 thread(1) tag(0) INFO lib_init:154 Creating tag condition variable.
2024-04-02 14:46:55.587 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2024-04-02 14:46:55.588 thread(1) tag(0) DETAIL cond_create:1114 Done.
2024-04-02 14:46:55.588 thread(1) tag(0) INFO lib_init:160 Creating tag tickler thread.
2024-04-02 14:46:55.588 thread(1) tag(0) DETAIL thread_create:884 Starting.
2024-04-02 14:46:55.588 thread(1) tag(0) DETAIL thread_create:917 Done.
2024-04-02 14:46:55.589 thread(1) tag(0) INFO lib_init:166 Done.
2024-04-02 14:46:55.589 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
[... 624 lines cut ...]
2024-04-02 14:47:00.885 thread(1) tag(11) INFO plc_tag_shutdown:1095 Starting.
2024-04-02 14:47:00.885 thread(1) tag(11) DETAIL plc_tag_shutdown:1101 Closing all tags.
2024-04-02 14:47:00.886 thread(1) tag(11) WARN mutex_lock_impl:767 null mutex pointer.
2024-04-02 14:47:00.887 thread(1) tag(11) WARN mutex_unlock_impl:818 null mutex pointer.
2024-04-02 14:47:00.887 thread(1) tag(11) DETAIL plc_tag_shutdown:1133 All tags closed.
2024-04-02 14:47:00.889 thread(1) tag(11) DETAIL plc_tag_shutdown:1135 Cleaning up library resources.
2024-04-02 14:47:00.891 thread(1) tag(11) INFO ab_teardown:142 Releasing global AB protocol resources.
2024-04-02 14:47:00.891 thread(1) tag(11) INFO ab_teardown:153 IO thread already stopped.
2024-04-02 14:47:00.895 thread(1) tag(11) INFO ab_teardown:156 Freeing session information.
2024-04-02 14:47:00.895 thread(1) tag(11) INFO session_teardown:142 Starting.
2024-04-02 14:47:00.896 thread(1) tag(11) DETAIL session_teardown:169 Destroying session mutex.
2024-04-02 14:47:00.896 thread(1) tag(11) INFO session_teardown:176 Done.
2024-04-02 14:47:00.897 thread(1) tag(11) INFO ab_teardown:162 Done.
2024-04-02 14:47:00.897 thread(1) tag(11) INFO mb_teardown:2427 Starting.
2024-04-02 14:47:00.898 thread(1) tag(11) DETAIL mb_teardown:2456 Modbus mutex destroyed.
2024-04-02 14:47:00.899 thread(1) tag(11) INFO mb_teardown:2458 Done.
2024-04-02 14:47:00.899 thread(1) tag(11) INFO lib_teardown:176 Tearing down library.
2024-04-02 14:47:00.899 thread(1) tag(11) INFO lib_teardown:212 Done.
2024-04-02 14:47:00.900 thread(1) tag(11) DETAIL plc_tag_unregister_logger:1368 Starting
2024-04-02 14:47:00.901 thread(1) tag(11) DETAIL plc_tag_unregister_logger:1372 Done.
2024-04-02 14:47:00.901 thread(1) tag(11) INFO plc_tag_shutdown:1142 Done.

D:\Roopesh\Adapter\Adapter\adapter\x64\Release\adapter.exe (process 16800) exited with code 1.
To automatically close the console when debugging stops, enable Tools->Options->Debugging->Automatically close the console when debugging stops.
Press any key to close this window . . .

kyle-github, 2024-04-04:
First, you need to use the PLC type "micro800" instead of LGX.   The two types of PLC do not support the same CIP commands.

Second, you will need to change the path.   I think that the path is either empty or something like 0,0.

See if that helps.