libplctag/libplctag #11: String support

reyanvaldes, 2015-07-22:
Kyle, 
It would be great to have capability of reading/writing string tags in the library to be more completed. In some applications the strings manners and it is so important as numeric data. Maybe when Get String we can have as parameters: offset, and max. length, and type of string (null terminating, leading length), to allow read/write string at least for most recently AB PLCs.

Thank you a lot for very nice work and sharing with community

kyle-github, 2015-08-28:
Hi Reyan,

I've been travelling at at conferences for most of the last two months. I will get to your request in a bit.  Please be patient while I catch up.  Thanks!

Best,
Kyle

reyanvaldes, 2015-08-29:
Hi Kyle
I hope you had good experience.

Thanks
Reyan

kyle-github, 2015-09-21:
I have opened a new mailing list in Google groups: libplctag.  Please bring this discussion there.  I would like to get a wider audience.  Thanks!

kyle-github, 2015-10-21:
Here is the new Google forum:

[libplctag](https://groups.google.com/forum/#!forum/libplctag)

I want to have the discussion there.  Please sign up!

closed 2015-10-21 by kyle-github