libplctag/libplctag #12: Implicit messages

reyanvaldes, 2015-07-22:
Kyle
It would be very helpful in some applications to allow work with implicit messages in the library, specially if need to collect the tags value from group at exact time when the trigger was made. Also maybe will be useful to provide functions for mapping or getting values from package received based on offset inside the buffer.

Thank you

kyle-github, 2015-08-28:
This is one of the most interesting things to add to the library.  Unfortunately, it requires a fair amount of replumbing in the library internals.  This will ultimately make the library more clean, but it is not a simple task.  Also, there is very, very little documentation about how to do this but I have some ideas.  It cannot be that hard since implicit messages are just UDP packets as are the return ACKs.

More later when I get caught up.

Best,
Kyle

kyle-github, 2015-08-31:
Hmm, these will need to be read-only too.  That means that we will need to introduce the concept of read-only tags at some point.

kyle-github, 2015-09-21:
I have opened a new mailing list in Google groups: libplctag.  Please join that as I am going to move discussions of future features there (with hopefully a larger audience).  Thanks!

kyle-github, 2015-10-21:
The Google group is here:

[libplctag](https://groups.google.com/forum/#!forum/libplctag)

I will close this and move discussion there until we come up with a solid idea.

closed 2015-10-21 by kyle-github