libplctag/libplctag #13: Tag Path structure for different PLCs

reyanvaldes, 2015-07-22:
Kyle
Please can you help me to understand what is correct tag path structure to use in the library based on cpu, I am facing some little challenges with it in the wrapper because my unknown about it, and unfortunately don't have experience with different types of AB PLC. 

So far we have following CPUs supported by the library:

1-PLC5
2-SLC
3-Micrologix
4-Compactlogix
5-Controllogix
6-Flexlogix

Also I was thinking it could be useful, as suggestion, to have two functions like GetCPUPath that receive as input parameters (protocol, ip, port type, slot, cpu) and returns the part of the tag path related with the cpu, and GetTagPath that receive (protocol, ip, port type, slot, cpu, tagName, elemCount, elemSize) returns the completed tagPath.

Thank you

kyle-github, 2015-08-28:
One of the difficulties with the AB protocol is that it is very flexible.  This is good for users and bad for me :-)

A path can transition from one kind of PLC (e.g. ControLogix) through the backplane, out a DH/RIO module and to a PLC5 for instance.  The data sent then is different from the data sent directly to a PLC5.  So, the "type" of CPU is a bit strange because it depends somewhat on the how you get to that CPU.

kyle-github, 2015-08-31:
BTW, this is why I ended up with an API that takes a string that looks a bit like part of a URL.  I realized that having a function with fixed inputs was:

1. really limiting when the protocol allowed some strange things like CIP does.
2. painful for wrappers like Java, Python etc. that do not really have pointers exposed and do not necessarily do arrays like C does.
3. painful for supporting multiple protocols.  Each protocol would either need its own tag creation functions (possible, but annoying) or I would need to have all possible arguments all the time.  I opted for a formatted string which is really a key-value pair array in string form.

I did start with an API that had specific arguments for the different parts.

kyle-github, 2015-10-21:
Please move this to the new forum:

[libplctag](https://groups.google.com/forum/#!forum/libplctag)

Thanks!

closed 2015-10-21 by kyle-github