libplctag/libplctag #14: Help request with DH+ path

mmcpheeters, 2015-08-28:
Hi Kyle,

We are using the library, and it works well for AB PLC direct connection, with standard path of '1,0'

Could you provide some guidance on what path should be if we are now trying to interface with a AB
PLC5 through a DH+ gateway?  We know the IP address of the gateway (I'm just not posting all of it).

We know the PLC is supposed to be on channel A, node 27, through the DH+.  Not sure what
the source address should be.  In fact, we aren't even sure if we need the colons instead of
commas, and something like '1,,1.A.27' which we tried first.

So recap, have tried:

protocol=ab_eip&gateway=xxx.yy.23.39&path=A:1:27&cpu=LGX&elem_size=2&elem_count=1&name=N15:40

protocol=ab_eip&gateway=xxx.yy.23.39&path=1,,1.A.27&cpu=LGX&elem_size=2&elem_count=1&name=N15:40

We get good status when creating the session, but bad status when attempting the first ever plc_tag_read call.

Thanks for any help you can provide!
-Matt

kyle-github, 2015-08-29:
Ah paths...

First, to be completely up front, I am not sure that the LGX-to-PLC5 path works any more.  I thought I had removed it from the wiki just in case.  

I do not have anything I can test with, but if you are willing to be a guinea pig, I can try to help with this.  The code was completed and it did work at one point a few years ago.  

Back to paths.  So, the path is the sequence of ports and slots that packets/data will go through.  So, for instance in a fairly standard LGX system, you have the CPU in slot zero.  The data comes into the ENBT/EN2T and hops onto the backplane.  That is usually via port 1 (not sure what port 0 does).  Then the CPU is in slot zero (0).  So, the path in this case would be path=1,0.

Now, in your case, you want to send the data back out through the DH/RIO module on a LGX?  I will assume that is the case below.

Assumptions:

* The gateway PLC is a LGX-class system with the CPU in slot 0.
* The gateway ENBT has IP address 10.0.0.42.
* The DH/RIO card is in slot 6 on the LGX chassis.
* The DH+ network is attached to channel A of the DH/RIO module.
* The LGX system is at node 12 (decimal!) on the DH+ network.
* The PLC5 is at node address 13 (decimal!) on the DH+ network.

The path should be something like "1,6,A:12:13"  

The full string should be:

"protocol=ab_eip&gateway=1.0.0.42&path=1,6,A:12:13&cpu=PLC&elem_size=2&elem_count=1&name=N15:40"

As I said far above, I am not sure that the code is still working for this particular path.  I currently have no way to test this.  

Changes from what you had include the path and the CPU.  

Over the weekend I will try to look through the code and see if there is any chance of it working or if bit rot has destroyed it all :-/

mmcpheeters, 2015-08-29:
Thanks!

Please look over the weekend.  We are willing to be a guinea pig, but the DH+ gateway and PLC
attached to it our remote.  A company we are working with made it visible to us over the internet.

I get a good status on creating the session, and we can interface with the PLC directly using
a program called KEPServerEX.  This particular program could have valuable info in it.  We
see station on the gateway configuration page is set to 22 (octal) so we use 18 below for
the source part of A:18:27

This is the particular gateway model:

http://www.prosoft-technology.com/Products/Gateways/EtherNet-IP/EtherNet-IP-to-Allen-Bradley-Remote-I-O-or-DH-Gateway

Last attempt was this:

protocol=ab_eip&gateway=xxx.yy.23.39&path=1,1,A:18:27&cpu=PLC&elem_size=2&elem_count=1&name=N15:40

And the error code we are getting is -11, the timeout error.  I have been trying with a 20 second
timeout just to be sure.

We aren't sure about the second '1'.  Are there some cases where you skip that field and use '1,,A:18:27' instead?

The PLC is a PLC5, not an LGX as you pointed out.  But are you sure cpu should be PLC instead of PLC5 ?

Have a great weekend!

kyle-github, 2015-08-31:
TL;DR - this isn't going to work until I fix the code.  

It looks like the code suffered quite a bit of bit rot when I transitioned from connected to unconnected as the base a couple of years ago.  While this ended up making things more robust, I must not have ported that code over.  

I can work on this in a few days, but I have no hardware to test this on.  I don't have a DH/RIO card. I will try to get one, but we rarely even see them anymore.  That said, the old code is relatively clear as to what I was doing, but my worry is that it has been so long since I have had a viable way to test this configuration that there might be other issues.

kyle-github, 2015-08-31:
BTW, adding "debug=1" to the tag string will cause the library to dump out a bunch of debugging information including dumps of all the packets received.  

You can use the tag_rw tool in the examples directory to try this out:

tag_rw -t sint16 -p "protocol=ab_eip&gateway=xxx.yy.23.39&path=1,1,A:18:27&cpu=PLC&elem_size=2&elem_count=1&name=N15:40&debug=1"

In the above path, is the DHRIO module in slot 1 (slots count from zero)?

Best,
Kyle

kyle-github, 2015-09-14:
The beta 1.0 branch has code that "works for me."  I do not have any hardware other than a DHRIO that I borrowed.    With that hardware, I can now reliably  get and set tags in a PLC5 attached via DH+ to a LGX chassis with a DHRIO module.

Please see if this works for you.  Thanks for your patience.

kyle-github, 2015-09-21:
Can I close this issue?

mmcpheeters, 2015-09-21:
I think so.  Thanks again!

Matt McPheeters
Systems Engineer
Renaissance Sciences Corporation
Cell:  (480) 254-0078

closed 2015-09-23 by kyle-github