libplctag/libplctag #15: Reading multiple tags

johanfforsberg, 2015-08-29:
It is possible to request several tags in one request through eip (although the size of a single request is limited to ~500 bytes, at least in in AllenBradley PLCs). When polling many tags this is a big performance win.

This involves figuring out how to split a given list of tags into several requests of limited size.

kyle-github, 2015-08-31:
(Side note: 544 bytes is what I saw in a couple of docs that AB left public.  However, I also saw lower numbers. I think I limit it to something like 512 just in case.)

There are a few things that need to be decided:

1. how do we create one of these multiple tag tags?
2. is there any limitation on them other than size?  I.e. read-only?
3. should the library allow arbitrary numbers of tags to be read at once or limit this to whatever the underlying protocol supports (remember I want to support things like ModBus)?
4. should the library allow arbitrary tags to be read as one bundle (i.e. not enforce that they are in the same PLC)?

My first goal here is to keep any API changes to only API additions and to a minimum to limit breakage of existing code.

Here are my thoughts per point:

1. **how do we create these?** I am not sure on this one.  We could allow the tag strings to have a delimiter for additional tags.  We could extend the API with another function that would take existing tags and merge them.  Pros and cons either way.
2. **Are there additional limitations?** I think read-only makes sense.  It is pretty common to read some tags that are read-only in the PLC like inputs.  The library could read including all sub-tags and write only including the ones that are writable.  This is already going to take a lot of code.
3. **Allow arbitrary numbers of sub-tags or limit this to some number?**  I am also a bit ambivalent on this.  I am leaning toward allowing arbitrary sub-tags and making the library figure it out to keep the client code simple.
4. **Should arbitrary tags be allowed in a multi-tag?** Various protocols enforce various rules about reading multiple tags at once.  I am leaning toward allowing this, even including a multi-tag that includes tags from multiple PLCs and multiple protocols.  However, I am not looking forward to implementing that.

Feedback?  Depending on the answers to these questions, this could take a lot of code.

Note that if these tags are ready-only we could also do some extensions such as making a read-only tag automatically reread from the PLC on a periodic basis.  That would eliminate some boilerplate logic in client code.

johanfforsberg, 2015-08-31:
My experience is from using https://github.com/EPICSTools/ether_ip which is sort of tied to the EPICS control system but can be modified to work without it. This library works for reading and writing multiple tags, but it has some issues regarding connection handling. It's also pretty old code that seems not actively developed any more. Hence my interest in alternatives :)

Here's some feedback off the top of my head:

1. The ether_ip library uses something it calls "scanlists" (I think this is an EPICS concept) which is just a list of arbitrary tags that is read/written as a unit. Any tags that have been changed in the scanlist since last scan are written to the PLC on the next scan. The tags in the scanlist are automatically chunked into pieces that fit in a request. This seems pretty efficient, but is probably a bit tricky to do right.

2. For my particular use case, reading is definitely the most important thing, writing is much more rare and could be done one tag at a time.

3. With the ether_ip library, I've not been able to reliably use a larger request size than 496 bytes (I think, need to double check the number) with AB ControlLogix PLCs. I'd suggest to make this limit configurable.

4. For me, reading/writing to several PLCs in a single go is not important, we use different processes to speak to different PLCs anyway.

The ether_ip library does not do any polling by itself, the user of the library needs to do that periodically. Which makes sense to me, client code may not play well with whatever method the library uses for background tasks.

kyle-github, 2015-09-05:
Thanks for the feedback.  I appreciate the notes about your experience with ether_ip!

I have done a (small) amount of thinking about this and it is definitely easier to make this protocol-specific and not try to make it work across multiple PLCs or protocols.

So, for a first cut, how does this sound:

1) use the existing C API but extend the tag string itself to support multiple tags with AB LGX-class PLCs (this does not work with PLC5 IIRC).  I am still thinking about the syntax, but the path and gateway would be shared.   Reading data out of these could be a bit annoying as you'll have to handle all the offsets.

2) the request would be grouped up to the maximum size of a request.  After that, the library would build another request.

3) We will have to see if there are any unknown limitations in total response size.  I am not aware of any, but that does not mean that there aren't any.

Thoughts?

johanfforsberg, 2015-09-08:
Sounds good to me!

kyle-github, 2015-09-21:
I have opened a Google group: libplctag.  I will move discussion of this feature to that mailing list.  Please sign up if you want to get updates.

kyle-github, 2015-10-21:
The Google group is here:

[libplctag](https://groups.google.com/forum/#!forum/libplctag)

I will close this.  There is more discussion on the forum about this.  I think I have come to a decision on how to do this that is both flexible and backward compatible.

whoschcekingin, 2017-03-21:
We are using libplctag for one of a test script for reading controller tags from Contollogix.

We are in requirement to read multiple tags in single request, in this discussion we found this functionality is under development.

Please let us know whether this functionality is supported in latest release , If not can you kindly suggest a method using which this can be implemented.

kyle-github, 2017-03-21:
It is not developed yet.   Most of the work since this discussion has been changes to the core of the library to allow this to be a lot easier to do.  There is still some work to be done before the work can start.

I will reopen this feature request.

kyle-github, 2018-02-14:
This is in progress.   I have a (barely) working prototype, but the ideas above are probably not going to make the cut.

I think what I will do instead is simply have an opt-out flag in the tag string.  If you do not opt out, requests against the tag will automatically be combined into a single packet with any other outstanding requests that fit.   There are limitations with this, but it seems like the least intrusive thing to do.

kyle-github, 2018-12-17:
This has been implemented in version 2.0 which is now in Beta.   Closing.

closed 2018-12-17 by kyle-github

franklupo, 2019-02-06:
Hi,
how read multiple tags?

best regards

kyle-github, 2019-02-07:
You do not need to do anything directly.  The library does it all automatically. This only works if you are connecting to a Control/CompactLogix.

Just create multiple tags and read/write them all at the same time.   The library will automatically pack the requests and unpack the responses.

See the async.c example.   It reads multiple tags.  It used the asynchronous method of calling the API to make it easier to see what is being done.   The stress_test.c program is one that does this with a single thread per tag.

Now that the library is doing request packing, the async method is a lot more efficient.

franklupo, 2019-02-07:
Ok, I'll try the solutions

fluisgirardi, 2021-03-26:
@kyle-github example?

kyle-github, 2021-03-28:
You can see examples in the async.c, async_stress.c and stress_test.c
programs.  The first two use async mode to start reading multiple tags in a
single thread.  The last one uses multiple threads.

There is nothing you need to do in the tag string.   The library does all
the work to make multiple requests pack into a single request packet to the
PLC.   Only Control/CompactLogix support this operation.

Best,
Kyle

fluisgirardi, 2021-03-29:
In your examples, you read the fragments/chunks of the same tag. Is possible to read different tags using/grouping at the same request?

kyle-github, 2021-03-29:
Yes, as long as they are in the same PLC the library will try to combine to
requests into one packet.

The library only allows one request packet to be in flight at a time.  So
any individual requests that get queued with the library is waiting for a
response will be packed together.

I can go into more detail if you need it.

Best,
Kyle

fluisgirardi, 2021-03-29:
Well, for knowledge, I'm using ab_eip2 

One thing that I have noted, is that the library opens/closes lots of connections. So every time it has to negotiate the params of a new connection, that slows down the scan process. 

You can see a screenshot of how many connections are being made. 
[image]

kyle-github, 2021-03-29:
Ah, the new code does not yet support multi-request packets.   Just the old
code right now.  That is the part I am working on as I get time.

If you copy the new list_tag.c code it is fairly dumb.   It is opening and
closing a single tag and that opens and closes connections to the PLC each
time.   If you keep even one of the tags open, the PLC connection will
remain open.

Best,
Kyle

fluisgirardi, 2021-03-29:
Well, isn't very good to hear that. I'll try to merge these two branches, while I keep at least one tag active.

kyle-github, 2021-09-17:
Just as an update, @fluisgirardi, I have a new branch, add_udt, that contains updated UDT and tag listing.  It works for requests larger than a single packet.

IgorIIV, 2024-05-23:
So how do I poll two different tags (not an array element) in the same controller?
Here is an example simple.c. What to write in TAG_PATH to poll for example PROGRAM:MyVar1 and PROGRAM:MyVar2.
Or how to do it differently?