libplctag/libplctag #16: libplctag on Windows

Bolkano, 2015-09-17:
Hello, 
Does the current version of libplctag work on Windows ? I tried to compile it but VS didn't find the correct references to some .h files. 

I'm not very familiar wit Linux so if the library is compiled on Windows with VS , after that can I use the compiled dlls on windows without any additional Linux specific frameworks etc ?

Thanks

kyle-github, 2015-09-17:
Hi,

Yes, the library works on Windows.  I am not a Windows dev, so it isn't as easy to use as the Linux makefile.  I last built it with Visual Studio 2013 and it worked fine.  Caveat: Windows is not our primary platform, so I am much less able to help with that than Linux.

Once you have the dll, you can use it without any other parts.  The dll does not depend on any other libraries (other than the standard Windows ones).

If you are able, I would love to have some contribute a working command line makefile for VS.  I have no idea how to do that and it would be better than the hacks I went through :-/

Best,
Kyle

kyle-github, 2015-09-21:
Have you been able to get the library built under Windows?

Bolkano, 2015-09-21:
I'll test it in a few days and will let you know.

Bolkano, 2015-09-26:
Hi Kyle, 

I'm using VS2013. However, the version of VS is not so important here. 
1) You should add correct reference path to the .h files in each project. Project Property-C/C++/General/Additional Include Directories - For example for VS2012 project it should be ..\lib;..\lib\ab;..\lib\util;..\lib\windows;%(AdditionalIncludeDirectories)
2) There are some missing files in VS2012 project - common.c, ab.c, ab_defs.h,  common.h

When you fix it, I'll be able to continue with the testing. 

Regards

Bolkano, 2015-09-26:
P.S. Where is that new Google libplctag group ?

kyle-github, 2015-09-27:
I am close to finishing a switch to using nmake instead of the GUI.  This seems to be more stable.  I am trying to get tag_rw compiling and tested now.

kyle-github, 2015-09-27:
Hi Don,

The Google group is here:

https://groups.google.com/forum/#!forum/libplctag

Or, it might be here:

https://groups.google.com/group/libplctag

I started with the second and ended up on the first.

Best,
Kyle

kyle-github, 2015-09-27:
I have checked in (master) the nmake set up for building libplctag.  It is not quite ready though.

To build this, you will need Visual Studio.  Do the following:

1. check out libplctag from GitHub.
2. go to the Start menu (or equivalent) and look in the Visual Studio folder.  There should be a folder for tools.  Go into that folder.
3. run the tool called something like Native Tools Command Prompt.  This opens a cmd.exe window in which all the environment variables are set up for nmake and the compiler and linker.
4. cd into the directory in which you checked out libplctag.  Go into the vs directory.
5. run nmake -f vs.mk all.  That should build the library.
6. move the library to where the system will find it.  Usually that is in c:\windows\system32 or something similar.
7. go into the examples directory.  (still in the command line) cd ..\examples
8. type nmake -f vs.mk all.  This should make the tag_rw tool.
9. if you did not copy the library, run set PATH=< path to vs folder >;%PATH% to allow the system to search for the library where it was built.
10.  run tag_rw without any arguments.  If you see an error about the .dll missing, make sure to either copy it where the system already searches or set up the PATH as above.
11. run tag_rw and provide a type and path to the PLC and tag you want to read.  Note that with DOS/cmd.exe you will need to escape the command line (the & and = are special characters) and I always, always forget to do that :-(

Unfortunately, this still results in a crash.  I have to go get on an airplane in a short time, so I will try to continue working on this later.  I am sure that it is something pretty obvious, but I am out of time today.

If you could help debug, that would be great!

I am doing my testing with Windows 7 and the free version of Visual Studio 2013.

Donnie-github, 2015-10-07:
I managed to build the library and the sample your way, however I crated and a VS solution with two projects - libplctag and tag_rw. With VS project it's much easier for debugging. I committed the changes to your repository. Currently I don't have а PLC at hand to test but I'll be able to test it next week and will let you know the result.

kyle-github, 2015-10-07:
Were you able to get tag_rw to work to the point where it tried to contact the PLC?  I am running into problems with the nmake version, but it looks more and more like it is actually the cmd.exe command line parsing more than anything else.

Do you have a fork of the repository?  If so, can you send me a pull request with the patched data?

Donnie-github, 2015-10-07:
I've just sent you a Pull request.

kyle-github, 2015-10-07:
Thanks!

I'll go through it and see what you did.

kyle-github, 2015-10-11:
Hmm, so I was going through the tag_rw that you checked in.  It looks like there are a bunch of changes on the surface, but when I went through them, I really only saw that you had added a define to define sleep as Sleep (which is not correct I believe as Sleep has arguments in milliseconds).  Other than that, the rest looks like whitespace/formatting changes.  Am I correct?

I have not touched the VS project yet.  I will go through that and make sure that my nmake compiler flags etc. are all correct. 

Have you been able to get tag_rw to correctly parse the command line?  In my latest testing this weekend, I think that may be the core of the problem.  It looks like cmd.exe is doing savage and unnatural things to the -p argument.  How do I stop cmd.exe from interpreting the quoted string?  I tried single quotes, but that did nothing useful.  I ended up quoting most of the operator characters like = and & and that got me closer, but I am still not there.

I did not realize that cmd.exe was so different from my *nix shell experience.  It has been a very long time since I did much with any command line DOS tools.  Maybe there needs to be a different tool than tag_rw for DOS?

kyle-github, 2015-10-11:
OK, I doubled checked the MSDN docs and found that my core problem was not escaping all of the characters I needed to escape on the cmd.exe command line when testing tag_rw :-(

It turns out that you must escape everything.  '&', '=', ',' etc. all must be escaped.  I was not escaping the commas in my previous tests!  Sigh.

So, the code as checked into the 1.0 and master branches appears to work fine on Win7 32-bit with Visual Studio Express 2013, using nmake.  I tweaked tag_rw.c so that all calls to sleep went to my_sleep and then made that platform dependent.  I tested the same source with Linux (Ubuntu 15.04 64-bit) and Win7 as I noted above.

I will go over your pull request for the VS project.  While nmake requires fewer special cases (i.e. telling the compiler than all sources are in C not C++), it is more of a pain to use so maybe it was not a good detour.  Thanks, again, very much for setting that up.

Can I close this issue?

Donnie-github, 2015-10-12:
In my personal opinion the Windows developers would prefer a VS project than the nmake option because with VS they'll be able to debug. 
Yes, we can close the issue. 
Thank you.

closed 2015-10-12 by Donnie-github