libplctag/libplctag #19: reading time in CompactLogix L32E

reyanvaldes, 2015-12-02:
I recently noticed that to read a single tag takes around 11 ms in CompactLogix L32E. Is there any way we can speed up this time in the reading process? The challenge is for example if need to read 200 tags, that could takes around 2 200 ms as total scan cycle. Please do you have any recommendation. Thanks

thatcadguy, 2015-12-02:
I am interested to hear what Kyle says. I am unfamiliar with the functions
in libplctag, but in general you should be able to pull an array as one
operation, so the solution should be to have the PLC copy values into an
array and then grab the array(s).

kyle-github, 2015-12-02:
11ms is not bad.  For an AB PLC, about the fastest I have seen was 8ms.
There are two sources of the delay:

A. The PLC takes a while to respond.  This seems to be fairly large.

B. When you do IO, your thread/process can get rescheduled and the time
slice is often fairly large.

There are three solutions:

0.  Do what Jacob suggests and create a new tag in your PLC that contains
all the data from the other tags you want to read.  There are some tricks
here such as using sequence sentinels at the beginning and end of the
array.  You must do something like this to make sure that you don't get
partially updated data.

1. run the reads in parallel with threads.  I have tested this with several
hundred tags.  There is some code in the examples directory showing the use
of pthreads.    HOWEVER: you need to be careful not to overrun the PLC.
Their network resources are very, very limited.  Try to avoid having more
than 10-20 requests in flight at the same time.  Possibly the best way to
do this is to have a worker thread pool that grabs tags and reads them.
You can limit this pool to 15 threads (for example) to keep the number of
requests in flight down.  I have production code (in Java) using upwards of
30 threads that has been running for several years.

2.  Wait.  I am working on another version of libplctag that will support
the AB multi-operation request packet type.  With this, you will be able to
specify a "read group" for all the tags that should be read together.  They
will be requested in a single (or very small number) of request packets and
all the data will be streamed back from the PLC in one (set of) response.
If you need single-threaded fast reads of a lot of tags, then this is the
best bet.  Unfortunately, this is going slowly as I am completely
overloaded right now with other work and the amount of change required to
support this is quite significant.   There are a few emails on the email
list/board about this.

When developing the library, I found out that PLCs have extremely limited
resources so it is very, very easy to overrun them!  The failure modes were
not consistent.  Sometimes the connection dropped.   Sometimes the request
was just not handled.  Sometimes request responses were returned out of
order (I handle this case).  Etc.

If you have questions about #0 or #1, let me know and I will try to help.
Getting the locking right is not always trivial and without using locks in
the right way (it is part of the API) you will quickly get a SEGV :-(

Best,
Kyle

thatcadguy, 2015-12-02:
I agree with the statement that 11ms is actually pretty good. Especially on
a CompactLogix.

If possible, one more thing you can try: in the Advanced tab of Controller
Properties in RSLogix, you can increase the System Overhead time slice so
that communications get a larger amount of the processor time in the PLC.
Changing the way you pull tags is going to be the best option though I
think. Increasing the time slice is only a band-aid fix.

reyanvaldes, 2015-12-02:
Kyle
Thank you so much for your help and time with this. I will try option #1- using multi-threads, following your recommendation of limit to 15 threads and let you know. I will restructure the main reading loop thread in a multiple reading threads. The solution of reading tags as group is great option. 
Thanks

kyle-github, 2015-12-03:
@Jacob - I am not sure that this will do anything.  At least on a
ControlLogix system, the processor in the ENBT/EN2T card is the limiting
factor, not the main CPU.  Most or all of the work for network IO is done
via the network module processor.

As I was noting about solution #0, a key point is that any packet sent to
or from the PLC is handled completely asynchronously from the ladder logic
(or can be).  Thus, suppose you build a large DINT array with your data.
You read it from a remote PC via libplctag (or any other library).  You can
get data that is partially complete.  Because the read is not synchronized
with the ladder logic, the main CPU could be in the process of setting the
values in the tag when the network module reads the data out from
underneath it.

The way I've combated this (and it is true whether the reader is a PC or a
PLC), is to have what I call sentinel values in first position in the tag
and last position in the tag.  In your ladder that sets the values, first
calculate a new sequence ID.  I like to use DINTs and wrap at 1000000 or
so.  Set the first sentinel.  Set the rest of the data.  Now set the last
sentinel.

When you get a value from a PLC, check the sentinel values.  If they are
the same, you got complete data.  If they are not the same value, then you
got "lucky" and got partially completed data.

You might think that this isn't a big problem, but we found it was
happening far more often than we thought before implemented this.

I think, but am not sure, that you can put the sentinel values anywhere in
the tag, but they must be written in the order I described above.  However,
there was some reason why we didn't do this and I can't remember what it
was...

This can be a problem with any tag that is updated in parts, not just ones
that will not fit in a packet.

kyle-github, 2015-12-03:
@Reyan - I would start with five threads.  Try to use as few as possible to
get the performance you need.  Think carefully about how the rest of the
application works to make sure that you are not trying to use the same tag
from more than one thread at a time.  That is a quick way to a SEGV :-(

thatcadguy, 2015-12-07:
@kyle-github I haven't tested it so you may be right. Makes sense for the ControlLogix. The ethernet is integrated on the Compact L32E so I would think it would, but I also don't know about what's inside in terms of processors and I/O.

kyle-github, 2015-12-08:
@reyanvaldes - I did a few small tests.  I think you will get the lowest load on the PLC by doing Jacob's suggestion.  Create a new tag and copy all the other tags into it.  Doing things in parallel will help, but there is a ceiling of performance.

One thing that should be kept in mind is that it is almost identical in load on the PLC to get one DINT as to get an array of 120 DINTs.  The overhead of the network stack is the costly part, not the data size.  That said, when you go larger than that, then the response is broken up across at least two response packets and your load goes higher (but still not as much more as you would think).

reyanvaldes, 2015-12-08:
Kyle thanks for your recommendation.

reyanvaldes, 2015-12-09:
Kyle
I did some testing using multithreading but seems it actually depend on how many cores the microprocessor support, if not there is not big performance difference. Unfortunately I cannot use the array solution because all data to read are not same type, and for the other hand need to modify the program that somebody else did.

Kyle I would like to share with you other idea if you don't mind,  not sure how complicated it is for AB PLC allows working with buffer of bytes and read/write physical address not only by tag name.

You know I am working also with Snap7 library to read/write any data on Siemens PLC. In short words the library allows to read/write buffer of bytes from/to any address, if course Siemens works with plc address except latest version. First we are creating list of packages based on all tags to read, which include area source, db, begin address, end address, and counter of tags belongs to each package. Second read each package on a buffer and extract all tags data belongs to it. Then in one package we can read several tags data.

I knows AB works different way, by name instead of address,  but I wonder if possible to apply similar concept, kind of work at application level with buffer of bytes and physical address, as option for packages of bytes. and the application be responsible of parsing. I think will help to improve reading multiple tags in one shot.

Maybe need some functions to work at buffer level, such us:
- knows the plc address where tag is stored on PLC. This will give option to create package list based on start address, data size, and limit by some size. Not sure if it is possible on AB.
- allows read/write buffer of bytes, using physical PLC address. Not sure how difficult it is.
- do the parsing of buffer based on tag type. I believe same functions you provide should work. 

Thanks

kyle-github, 2015-12-09:
Perhaps you are already starting to hit the limits of the PLC.  By adding
additional threads, you are "pipelining" the requests to the PLC.  This can
increase latency for each individual request, but also increase the overall
read rate/bandwidth.  How often are you trying to read these tags?  Are you
overrunning the PLC's network processor?   At least with ENBT cards you can
point a web browser at the IP address of the module (port 80) and get
access to a minimal web-based status reporting system that can give you
information about network CPU load, packet loss, etc.

In one of our applications we read 16 tags of 125 DINTs each plus five or
so more of smaller size.  In that application we wanted to use read rates
of 250ms.  We found that this ran the network module CPU up to over 80%
(200ms caused problems with loads stead at 100% and very long response
latencies).  We finally settled on reads every 500ms.  That leaves enough
headroom in the network module to stay responsive to other requests.  We
also split the network load.  One ENBT handled all the PCL-to-PLC traffic
and the other handled PLC-to-PC traffic.  For this application we have not
created the single large tag.

We have done the method that Jacob described with different kinds of tags.
We created a User Defined Type (like a struct in C) that was the
concatenation of all the different tags we wanted to read.  Then we added a
single DINT at the beginning and a DINT at the end for the sequence
sentinel values.  We added a ladder to copy the data from the original tags
to the new one.  We then read that single new (very large) tag.  We
originally started doing this for explicit messaging between PLCs.

I am working on exactly the feature you describe (in AB's own way).  The
way this will work will be very similar to how it works now.  Here are the
changes that will be needed on the client side:

1) add a new attribute to the tag string "read_group=XZY" for all tags that
should be read at the same time.

2) set up read caching as well (i.e. for 100ms or something).  This is also
in the tag string.  Set the "read_cache_ms" attribute to 100 (for example).

3) do the reads as normal.

There will be no API changes to the C API.

What will happen is that triggering a read on any tag in a "tag group" will
result in all the reads for the tags in the group being done at the same
time.  The read requests will be packed into a small number of request
packets (request packets are size limited to about 540 bytes).  The results
will similarly be packed into a stream of result packets.  This will reduce
the number of requests in flight to a very low number.

The read caching is needed to avoid triggering too many reads of all the
tags.  In this manner, you can keep whatever original plc_tag_read() calls
you had at their original intervals, but the reads will not actually result
in a read to the PLC until the cache expires.

If you used the tag strings directly, there will be no code change
associated with this.

Best,
Kyle

reyanvaldes, 2015-12-09:
> @reyanvaldes Just a quick comment about your previous message - not a
> problem that the tags are different types, use the COP instruction in
> RSLogix to copy data bit by bit into DINTs or another data type of your
> choosing, convert back on the libplctag side. Same thing you have to do
> with Modbus basically. Just a note for the future, I saw you don't have
> access to the program.

Thanks for suggestion. Unfortunately I cannot modify the program. 
I think I will wait for the reading tags group to improve the reading time.

Thanks

reyanvaldes, 2015-12-09:
Thank you Kyle. 
For the time interval I am using 250 ms and 6 threads, on top of the current threads that the application is using. Unfortunately I have really only 2 cores available, so the "parallelism" performance is not as expected.

I think I will better wait for the new version that include read groups. Is there any limit for the amount if tags to read in a group? For example can I put 150 tags in same group? 

Thank you so much Kyle for your time and support. 

Thanks

thatcadguy, 2016-01-01:
@reyanvaldes Don't know how I didn't notice before but none of the Makefiles have optimization flags set -- might want to retry with -O2 if you haven't already.

reyanvaldes, 2016-01-02:
Ok, thank you

kyle-github, 2016-01-02:
Perhaps that will make a difference, but generally, the communications with the PLC will be IO-bound, not CPU-bound.  Thus, even a single processor can often show better throughput with multiple threads.

In my tests, the CPU usage of the library is usually very low.  Additional optimizations would not help in that case.

I recently had the opportunity to exchange an ENBT for an EN2T and got before and after measurements.  This was a application where the ENBT was hit fairly hard by the library.  We saw CPU levels of 55% or so.  With the EN2T, we saw 6%.  Nearly a 10x difference.  The CPU load on the PC was hard to measure because it also runs MySQL for data logging and a web server.  It was not very busy (but was doing a lot of disk IO).

I will continue to work toward supporting the AB multi-request packets as that may be the best bet to solve the problem.  However, 200 tags is a lot to read at once.    I am not sure what the limits will be in one request packet, but I will try to work around the limit in some way.

kyle-github, 2016-01-02:
Hmm, it was not clear here, the CPU measurements were from the CPU in the PLC network module (ENBT or EN2T), not the PC!

kyle-github, 2016-01-04:
One option I forgot about is to run the tag reads asynchronously:

plc_tag_read(tag, 0);

Do a bunch of those, wait a bit, check the status and keep going.

That will pipeline the requests about as effectively as using multiple threads, but might be easier for you to use depending on your application architecture.

There are some examples (look for async.c in the examples).

Best,
Kyle

reyanvaldes, 2016-01-04:
Ohh, it sounds good, I will check it.
When you say do a bunch of this,is there any limit? We have to wait some milliseconds? 

Thanks
Reyan

kyle-github, 2016-01-04:
When you fire a request asynchronously, the code doesn't wait around for a response.  You must call plc_tag_status() on the tags to see if the read has finished.  If everything is OK, but the read has not finished, you will get PLCTAG_STATUS_PENDING back.

When you pipeline a lot of requests, the delay can become large.  I've seen upwards of 400ms.  

Generally if I write something like this, I have a loop that reads the status and then waits.  The nice part is that you can trigger many reads (sending the request packets to the PLC) all at once and then wait for all of them to complete.

reyanvaldes, 2016-01-15:
Kyle
I implemented what you suggested and it worked great. It actually reduce the time drastically. In short words what I did is to work in a group of around 20 tags per reading and check after the status.

Thank you so much for support
Reyan

kyle-github, 2016-01-15:
Good to hear that this solved your problem.  Async IO is usually harder to do, but often has better performance and uses fewer resources.  Can I close this issue?

reyanvaldes, 2016-01-15:
Yes, thank you

kyle-github, 2016-01-15:
Solution found.  Closing issue.

closed 2016-01-15 by kyle-github