libplctag/libplctag #26: Android Support

nmnhero, 2016-02-10:
We badly need android support for this library. It can revolutionize the industry!

kyle-github, 2016-02-14:
The C runtime environment of Android has diverged from that of "stock" Linux so this might be a bit more challenging than a "pure" POSIX system.  I will look into it when I get a chance.  However, my current priority is working on version 2.0 which has a very different internal structure so I probably will not be able to get to this for a while.

I will keep this in mind as I work on 2.0 and hopefully I can make sure that the Android port will be fairly painless.

nmnhero, 2016-02-15:
Understood. But by "a while", do you mean months or years?

kyle-github, 2016-02-15:
Months.  Hopefully only a few.  I am really busy right now and the 2.0 code
is taking a lot longer than I had hoped.

kyle-github, 2016-08-02:
I am moving to a CMake-based build system.  I think this will help isolate platform dependencies, but I have only just managed to get Linux builds working.

thatcadguy, 2019-02-10:
@kyle-github I just compiled this with Termux, the only thing I had to change was to replace pthread_cancel() with pthread_kill() since Android's standard C library (Bionic) doesn't have pthread_cancel(). I can't test though, I can't get the dang OpenVPN app to work.

GitHubDragonFly, 2020-09-07:
To find all Android related deficiencies in any of the new plctag library releases, these would probably be the good steps to follow in Android Studio:

- Start a new Android Studio project
- Select "Native C++" template
- Choose Name / PackageName / Location, Java Language (or Kotlin) and Minimum SDK
- On the next screen "Customize C++ Support" select "Toolchain Default" (or if you know what you are doing one of C++11 / C++14 / C++17)

- In the newly created project, remove CMakeLists.txt and native-lib.cpp files from the "cpp" folder as well as all related entries in the MainActivity.java and activity_main.xml layout
- Copy all your libplctag project files and folders to the "cpp" folder
- Copy Tag.java file from the library project to the folder where the MainActivity.java file is (normally app/src/main/java/{PackageName folders})

- Download jna.aar file: https://github.com/java-native-access/jna/blob/master/dist/jna.aar
- On the menu click File/Project Structure, in the new window select Modules, click "+", select "Import .JAR/.AAR package" and browse to the download location to select the file
- Add to the app's build.gradle file "implementation project(':jna')" as a part of the "dependencies"

- On the menu click File/Save All and then File/Sync Project with Gradle Files
- Build/Rebuild Project (infrequent use of Build/Refresh Linked C++ Projects, File/Invalidate Caches / Restart might also help)

One of the first errors might be related to the deleted native-lib.cpp, so just do Build/Rebuild Project again.

Plctag related errors should be within CMakeLists.txt file, related to "examples" and "tests".

Otherwise the library should compile fine.

There might possibly be an error related to "-lpthread" entry.

kyle-github, 2024-06-09:
This was done some time ago.

closed 2024-06-09 by kyle-github