libplctag/libplctag #27: Many aborting request debug messages

dlbrown06, 2016-02-25:
I'm constantly reading a few specific tags about every 50ms. My console gets bloated with messages like below.

`request_handler_func:723 aborting request 000000000262CBB0`

Am I not setting this up correctly? These debug messages appear to have something to do with multiple sessions running that aren't being handled.

Once I call `plc_tag_create` I begin reading `plc_tag_read` that tag and I never destroy it until the program is finished running. From the documentation I would think this would be okay.

kyle-github, 2016-02-25:
Hi,

Unfortunately, I left a debug output locked on in the 1.0 version for a while.  It is harmless debugging output.

I think I fixed this up in the latest update to 1.0, but let me check.... looks like it was updated but not commented out completely.  The latest code on master does not have that output on that line, so it was modified.

Internally, requests are queued with the IO thread.  Once completed and the data is processed, I flag them for deletion by the IO thread and that is the semi-erroneous debug output.  The request is aborted (even though it is already complete) and removed and the memory freed.

You can ignore it or update to the latest to see if that fixes it.  I thought I got rid of that debug output completely, but it is still there, just not on by default.

Can you please check with an updated copy of the library and see if the problem persists for you?  Thanks for reporting this!

dlbrown06, 2016-02-26:
Okay I `git pull`ed the latest master. Unfortunately when I try to build this version I'm getting these errors.

```
C:\test\libplctag\vs>nmake -f vs.mk all

Microsoft (R) Program Maintenance Utility Version 11.00.50727.1
Copyright (C) Microsoft Corporation.  All rights reserved.

Could Not Find C:\test\libplctag\lib\libplctag_tag.obj
Could Not Find C:\test\libplctag\lib\ab\ab_common.obj
Could Not Find C:\test\libplctag\lib\util\attr.obj
Could Not Find C:\test\libplctag\lib\windows\platform.obj
Could Not Find C:\test\libplctag\vs\libplctag.dll
Could Not Find C:\test\libplctag\vs\libplctag.*
        cl /I..\lib /I..\lib\ab /I..\lib\util /I..\lib\windows /MDd /c /DLIBPLCTAGDLL_EXPORTS=1 /W3 /Fo..\lib\ /Tc ..\lib\libplctag_tag.c
Microsoft (R) C/C++ Optimizing Compiler Version 17.00.50727.1 for x86
Copyright (C) Microsoft Corporation.  All rights reserved.

libplctag_tag.c
..\lib\libplctag_tag.c(126) : error C2143: syntax error : missing ';' before 'type'
..\lib\libplctag_tag.c(128) : error C2065: 'debug' : undeclared identifier
..\lib\libplctag_tag.c(133) : error C2065: 'debug' : undeclared identifier
..\lib\libplctag_tag.c(153) : error C2143: syntax error : missing ';' before 'type'
..\lib\libplctag_tag.c(155) : error C2065: 'debug' : undeclared identifier
..\lib\libplctag_tag.c(160) : error C2065: 'debug' : undeclared identifier
..\lib\libplctag_tag.c(185) : error C2143: syntax error : missing ';' before 'type'
..\lib\libplctag_tag.c(187) : error C2065: 'debug' : undeclared identifier
..\lib\libplctag_tag.c(196) : error C2065: 'debug' : undeclared identifier
..\lib\libplctag_tag.c(227) : error C2143: syntax error : missing ';' before 'type'
..\lib\libplctag_tag.c(228) : error C2275: 'mutex_p' : illegal use of this type as an expression
        ..\lib\windows\platform.h(110) : see declaration of 'mutex_p'
..\lib\libplctag_tag.c(228) : error C2146: syntax error : missing ';' before identifier 'temp_mut'
..\lib\libplctag_tag.c(228) : error C2065: 'temp_mut' : undeclared identifier
..\lib\libplctag_tag.c(229) : error C2143: syntax error : missing ';' before 'type'
..\lib\libplctag_tag.c(231) : error C2065: 'debug' : undeclared identifier
..\lib\libplctag_tag.c(235) : error C2065: 'temp_mut' : undeclared identifier
..\lib\libplctag_tag.c(235) : warning C4047: '=' : 'int' differs in levels of indirection from 'mutex_p'
..\lib\libplctag_tag.c(238) : error C2143: syntax error : missing ';' before 'type'
..\lib\libplctag_tag.c(238) : error C2143: syntax error : missing ')' before 'type'
..\lib\libplctag_tag.c(238) : error C2065: '__sync_flag_nargle_238' : undeclared identifier
..\lib\libplctag_tag.c(238) : error C2065: 'temp_mut' : undeclared identifier
..\lib\libplctag_tag.c(238) : warning C4047: 'function' : 'mutex_p' differs in levels of indirection from 'int'
..\lib\libplctag_tag.c(238) : warning C4024: 'mutex_unlock' : different types for formal and actual parameter 1
..\lib\libplctag_tag.c(238) : error C2059: syntax error : ')'
..\lib\libplctag_tag.c(238) : error C2143: syntax error : missing ';' before 'for'
..\lib\libplctag_tag.c(238) : error C2065: '__sync_rc_nargle_238' : undeclared identifier
..\lib\libplctag_tag.c(238) : error C2143: syntax error : missing ';' before ')'
..\lib\libplctag_tag.c(250) : error C2065: 'tag' : undeclared identifier
..\lib\libplctag_tag.c(250) : error C2223: left of '->vtable' must point to struct/union
..\lib\libplctag_tag.c(251) : error C2059: syntax error : '}'
..\lib\libplctag_tag.c(254) : error C2059: syntax error : '}'
..\lib\libplctag_tag.c(257) : error C2059: syntax error : '}'
..\lib\libplctag_tag.c(412) : error C2143: syntax error : missing ';' before 'type'
..\lib\libplctag_tag.c(413) : error C2143: syntax error : missing ';' before 'type'
..\lib\libplctag_tag.c(415) : error C2065: 'debug' : undeclared identifier
..\lib\libplctag_tag.c(422) : error C2065: 'debug' : undeclared identifier
..\lib\libplctag_tag.c(468) : error C2065: 'debug' : undeclared identifier
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 11.0\VC\BIN\cl.EXE"' : return code '0x2'
Stop.
```

kyle-github, 2016-02-28:
I stood up a 32-bit Windows 7 VM in Virtual Box.  In that VM I have Visual Studio 2013 Express.  I am not sure what version that corresponds to.  Is it newer than version 11?  According to what I see online, version 11.0 corresponds to Visual Studio 2012.  I thought Express 2013 was pretty much the same (stripped down) as the full version 2012.  Maybe I have that wrong though? 

I went to the Start Menu. I selected show all programs.  I went to the Visual Studio folder and looked up the tools folder in that (all this is in the Start menu).

I then ran the entry to start a cmd.exe for native builds.

In the command line window, I went to the directory in which I have the code (in this case F:\projects\libplctag-1.0\vs\).

I ran:

nmake -f vs.mk libplctag

The library built.

I changed the directory to the examples directory.  I copied the DLL from the previous directory into the examples directory.

I ran:

 nmake -f vs.mk all

tag_rw.exe built.

I ran it against a PLC and it worked.

Can you please try the steps I went through?   I am starting to wonder if there is some sort of difference between VS 2012 and VS Express 2013 that is important?

I double checked that I was completely up to date with the latest code in GitHub and I am.  

Are you a VS developer?  If so, do you know how to make a .sln that will work no matter where the code is checked out?  I set up nmake, but apparently that is not a good choice for Windows developers.  I would love to get this set up so that it is a lot easier to use in Windows.

dlbrown06, 2016-03-16:
I am getting this same error I posted above when I use the `nmake -f vs.mk libplctag` cmd.

Unfortunately I am not a VS developer either. :/

kyle-github, 2016-03-16:
It is possible that VS 2012 does not work.  MS was slow to support the C99 standard.  I do not have access to VS 2012.  Can you try VS Express 2013 or newer?  It is free.

I will research the errors to see what is going on.  It looks like there is something that is causing the compiler to be unhappy.  It is happening in two places: one for all debugging calls (pdebug(...)) and it looks like my mutex wrapper macro is making it unhappy too.  That uses C99 constructs.

kyle-github, 2016-03-16:
The link here:

[VS Versions](https://en.wikipedia.org/wiki/Microsoft_Visual_Studio#History)

Shows that this seems to be VS 2012.  

From what I see in [this article](http://arstechnica.com/information-technology/2013/06/c99-acknowledged-at-last-as-microsoft-lays-out-its-path-to-c14/), it looks like C99 support is not really available in MS VS 2012 and only really starts to be (somewhat) supported in VS 2013.  Even then, not all C99 features are supported.  

So, as far as I can see, the library will not work with VS 2012.

dlbrown06, 2016-03-16:
Okay nice - I was able to compile it with 2013.

However, pulling the latest from master and compiling. I am still getting the `aborting request` debug message in the console.

dlbrown06, 2016-03-16:
I'm pulling from master though... Should I pull from another branch?

kyle-github, 2016-03-16:
Master or 1.0 should be the same right now. Let me take a look later today.  It should be commented out.

kyle-github, 2016-03-20:
It looks like I forgot to push to master or something.  I commented out the offending debug line.  Can you give it another try with the latest master?

dlbrown06, 2016-03-22:
That looks good. Thanks!

closed 2016-03-22 by dlbrown06