libplctag/libplctag #29: MainPogram tags logix5000

Felipeasg, 2016-06-12:
Hi there, recently I have tested the liblpctag with ControlLogix L35E e L35RE and this works petty well.
But by some reason I only can read/write tags created in the Controller Tags and I can't read the MainProgram tags (In the controller organizer).

I find the issue "[Addressing tags with : in the tag name](https://github.com/kyle-github/libplctag/issues/8)" where the problem with ":" was resolved to read tags in others scopes, but I try to read the MainProgramTags (MainPogram:PIC_7112.KP_SUP) and the lib always return the -5 error.

Someone had a similar error when trying access tags in the MainProgram scope on Logix5000?

[image]

When I put &debug=1 the log is:
[image]

Thanks in advance!

kyle-github, 2016-06-12:
Hmm, I haven't tried this.  You might want to join the mailing [list/forum](https://groups.google.com/forum/#!forum/libplctag) and see if anyone else has run into this.  My guess is that maybe the main program isn't named "MainProgram" but has some special name.

Felipeasg, 2016-06-13:
Kyle, thanks for the quick response...
I saw in the forum two topics about this problem (Read tag environment program and Program Scope Tags). In the secont topic I found something that could help me...
I will try acess the tag using something like this -> "PROGRAM:My_Program.myBOOLarray"
I will try tomorrow and back here to post the results..

Tanks again.

Felipeasg, 2016-06-14:
Good news, the trick works very well, to access program scope tag I used the following sintax:

**PROGRAM**:MainProgram.PIC7112.Kp_Sup

Thanks

kyle-github, 2016-06-21:
Can I close this?  I need to document it on the wiki too.

Felipeasg, 2016-07-12:
Sorry by the delay,.
Yes you can close.

Thanks.

kyle-github, 2016-07-13:
There is no publicly available documentation for CIP/EIP.  You have to buy it from the OVDA.  I do not have any copies.  I reverse engineered what is going on from the TuxEIP code.

Thanks for the link on CIP addressing!

closed 2016-07-13 by kyle-github