libplctag/libplctag #30: Destroying the tag inside plc_tag_destroy(plc_tag tag)

flyingvince, 2016-06-19:
Above the `plc_tag_destroy` function (in the FIXME comment) there is a mention that this function leaves a dangling pointer. As a suggestion, could this below be used to destroy the `tag` object and avoid this particular problem?

I am having issues with instabilities when using libplctag via the Python wrapper. The issues encountered are Python thread(s) becoming unresponsive and various kernel related problems as shown below.

```
Jun 13 04:48:05 SERVERNAME kernel:  [  233.667827] traps: python3[1579] general protection ip:7fb8c84f7542 sp:7fb8c84e4b50 error:0 in libplctag.so[7fb8c84e7000+15000]
Jun 15 15:39:44 SERVERNAME kernel:  [  794.702409] traps: python3[1668] general protection ip:7f8b9d25c227 sp:7f8b99b757c0 error:0 in libc-2.19.so[7f8b9d222000+1bb000]
Jun 17 15:53:24 SERVERNAME kernel:  [ 5442.060720] traps: python3[1635] general protection ip:7f8973b39c25 sp:7f8973b32c20 error:0 in libplctag.so[7f8973b35000+15000]
Jun 17 16:27:47 SERVERNAME kernel:  [ 1830.463503] traps: python3[1643] general protection ip:7ff0f7e93227 sp:7ff0f2f68cc0 error:0 in libc-2.19.so[7ff0f7e59000+1bb000]
Jun 17 16:40:21 SERVERNAME kernel:  [  238.774811] traps: python3[1641] general protection ip:7f947537cd6e sp:7f9471c109d0 error:0 in libc-2.19.so[7f94752fd000+1bb000]
Jun 17 20:51:12 SERVERNAME kernel:  [  456.961823] traps: python3[1685] general protection ip:7fab0910a0b7 sp:7fab090ffb90 error:0 in libplctag.so[7fab09102000+15000]
Jun 17 20:55:15 SERVERNAME kernel:  [   32.542667] traps: python3[1639] general protection ip:7f2c06b3d227 sp:7f2c03416540 error:0 in libc-2.19.so[7f2c06b03000+1bb000]
Jun 19 11:05:48 SERVERNAME kernel:  [ 2470.992905] traps: python3[1731] general protection ip:51858f sp:7f50094e80a0 error:0 in python3.4[400000+2fe000]
Jun 19 11:15:08 SERVERNAME kernel:  [  319.783533] traps: python3[1637] general protection ip:4d0a2b sp:7fe828e65040 error:0 in python3.4[400000+2fe000]
```

The frequency of instability issues increase when I read tags that have bigger sizes (50% of my tags are arrays, with size around 200-300 UINT16 elements, the rest are single UINT16 or UINT32 tags). 

Around the 13th June I was testing with reading unique tags (ex: N317[12], N317[2], ...). Later I added reading the entire arrays at once, then using ```libplctag.plc_tag_get_uint16(tag, index_offset)``` to extract the actual values and the frequency of kernel issues increased.

Here the updated code section, which may (?) fix the issue. My knowledge of C is rather limited and I'd welcome any feedback on whether this may help or even solve the dangling pointer issue referred to in the FIXME.

```
LIB_EXPORT int plc_tag_destroy(plc_tag tag)
{
        /* NEW : function to destroy the tag struct */
        void destroy_tag_struct(struct plc_tag_t **param)
        {
        *param = NULL;
        }

        if(!tag)
                return PLCTAG_STATUS_OK;

        int debug = tag->debug;
        mutex_p temp_mut;
        int rc = PLCTAG_STATUS_OK;

        pdebug(debug, "Starting.");

        /* clear the mutex */
        if(tag->mut) {
                temp_mut = tag->mut;
                tag->mut = NULL;

                critical_block(temp_mut) {
                        if(!tag->vtable || !tag->vtable->destroy) {
                                pdebug(debug, "tag destructor not defined!");
                                tag->status = PLCTAG_ERR_NOT_IMPLEMENTED;
                                return PLCTAG_ERR_NOT_IMPLEMENTED;
                        }

                        /*
                         * It is the responsibility of the destroy
                         * function to free all memory associated with
                         * the tag.
                         */
                        rc = tag->vtable->destroy(tag);
                }

                mutex_destroy(&temp_mut);

                /* NEW finally destroy the tag struct */
                destroy_tag_struct(&tag);

        }

        return rc;
}

```

Thank you.

kyle-github, 2016-06-20:
Thanks for the bug report.

The problem is that C only passes parameters by value.  So, when the tag parameter is passed into plc_tag_destroy(), it is copied.  When you take the address of it, you get the address on the stack of the local parameter (the compiler might even optimize this all away).  When you set that to null, it does not change the value of the variable in the calling function.

A second problem is that while GCC allows you to embed functions, the standard C specification does not.  That is not value C :-(

Can you run a test with debugging enabled?  You do this by adding &debug=1 to the tag string.  This will cause the library to dump out a lot of debugging output.  From that, hopefully we can tell if plc_tag_destroy is being called twice.  My guess is that it is.  It could be a use-after-free bug.  The fact that you are getting an error in libc often means that the malloc implementation noticed that you are freeing a block that is not allocated.

I am not a Python programmer, but I seen to remember someone telling me that using some frameworks in Python made shared C data not work right.  Each thread would need to open its own tag?  Hmm, can't remember this clearly...

I can't tell what the output is that you included in the issue report.  However, is has a pattern that makes me wonder if you have either a double free or a use-after-free problem.  Is it possible that the Python code ends up calling plc_tag_destroy() explicitly and then again implicitly due to GC?

flyingvince, 2016-06-21:
Thank you for your prompt response and for providing feedback, appreciated.

> Can you run a test with debugging enabled? You do this by adding &debug=1 to the tag string. This will cause the library to dump out a lot of debugging output. From that, hopefully we can tell if [...]

Definitely, will do that and submit the logs. `pdebug_impl()` logs to stderr, am I right?

>I am not a Python programmer, but I seen to remember someone telling me that using some frameworks in Python made shared C data not work right. Each thread would need to open its own tag? Hmm, can't remember this clearly...

I should specify that only one Python thread access PLC data via libplctag, the other Python threads are performing non-PLC related tasks. In my Python thread accessing PLC data via libplctag, I process one tag at a time, always with the same sequence: `plc_tag_create()`, `plc_tag_status()`, `plc_tag_read()` and finally `plc_tag_destroy()`. I loop through all tags in a `while True:` loop.

>However, is has a pattern that makes me wonder if you have either a double free or a use-after-free problem. Is it possible that the Python code ends up calling plc_tag_destroy() explicitly and then again implicitly due to GC?

Good point, let me investigate this as well. The logs from `pdebug()` should help confirm that hypothesis. Will get back when I have more info.

kyle-github, 2016-06-21:
Just a quick note on performance.  Looping through a set of tags creating, reading and then destroying them is going to be really slow.  Each tag will end up setting up a new TCP connection to the PLC.   Tags take up very little space.  You can create them all, do your reads in a loop, then destroy them.  This will use one TCP connection to the PLC and have one set up handshake.  

The fastest way (currently) to do this is to create all your tags in a loop.  Then start the reads all in a loop, but using async mode (timeout of zero).  Then check the status in a loop.  Then get the data in a loop.  There are limits to how many tags you can read like this.  I usually try to limit it to about 30 at a time.  However, you will find that your throughput goes up dramatically.

I am thinking about ways to modify the library so that double free and use after free cannot happen.  Or, at least are much, much harder to do.

flyingvince, 2016-06-21:
Log file is attached. I extracted 3'000 lines, which covers a few `plc_tag_create` to `plc_tag_destroy()` cycles. In the log I added some Python stdout logging (anything with `(UTC)	DEBUG` in the timestamp). 

From the look of it it seems that `plc_tag_destroy:231 Starting.` is called only once per cycle. Can you see anything abnormal from the log file?

[debug_v104.log.zip](https://github.com/kyle-github/libplctag/files/324980/debug_v104.log.zip)

kyle-github, 2016-06-24:
Sorry for the delay.  I am going on vacation and trying to get everything done before I leave has taken all my time lately.

I went through the log file and nothing is jumping out at me.  Was there any crash during this time?  I need to modify the plc_tag_destroy() function to print out the address of the tag object.    I am not sure I will have time to do that.  If you want, you can modify plc_tag_delete like this:

```c
LIB_EXPORT int plc_tag_destroy(plc_tag tag)
  {
        if(!tag)
                return PLCTAG_STATUS_OK;

        int debug = tag->debug;
        mutex_p temp_mut;
        int rc = PLCTAG_STATUS_OK;

        pdebug(debug, "Starting. (%p)", tag);   <- THIS LINE

        /* clear the mutex */
        if(tag->mut) {
                temp_mut = tag->mut;
                tag->mut = NULL;

                critical_block(temp_mut) {
                        if(!tag->vtable || !tag->vtable->destroy) {
                                pdebug(debug, "tag destructor not defined!");
                                tag->status = PLCTAG_ERR_NOT_IMPLEMENTED;
                                return PLCTAG_ERR_NOT_IMPLEMENTED;
                        }

                        /*
                         * It is the responsibility of the destroy
                         * function to free all memory associated with
                         * the tag.
                         */
                        rc = tag->vtable->destroy(tag);
                }

                mutex_destroy(&temp_mut);
        }

        return rc;
  }
```

The code to change is in libplctag_tag.c.  Change the first pdebug line that has "Starting." in the string in plc_tag_destroy().  This will make it print out the pointer of the tag data structure.  We can then look in the logs and see if any calls to plc_tag_destroy() are called twice with the same pointer.  It isn't perfect as it is possible that the memory for a new tag will be allocated at exactly the same location as the previous one.

Best,
Kyle

flyingvince, 2016-06-24:
Hi Kyle, same here: going on a holiday tomorrow ;-) Thanks for your input. I have been reviewing my Python code and I saw something that could have been improved. Doing tests at the moment and so far the application is stable. I will provide feedback if it ends up working well, if not I will follow your suggestion for the `pdebug` change. Stay tuned and thanks again for your support.

kyle-github, 2016-06-24:
A person on the list, Matt McPheeters, is also using Python.  He had some
problems due to how his library (Twisted I think) handled threading or
processes.  When we are both back, I can reach out to him to see what he
had to do.  Unfortunately, I have very little experience with Python :-(

Have a good vacation.

Best,
Kyle

kyle-github, 2016-07-13:
Did your changes help?  If so, can you write up a small synopsis of the problem so that others can avoid the problem in Python?

The next version of the library is going to use a different way to identify tags so that passing in a tag that is already closed will not cause a library crash.

flyingvince, 2016-07-14:
The changes in my Python code did not help, still experiencing instability issues as before. I haven't yet inserted the `pdebug` line (from your suggestion) into the code to gather more information, will do it early next week.

Can you please share a rough timeline for the release of the next version of libplctag?

kyle-github, 2016-07-14:
Unfortunately, I am very late with libplctag2.  I may see what I can hack
into the v1.0 library to remove the direct pointers from the API.  I don't
have a lot of time for the next week or two, but I will try to get
something done with the existing code.

Best,
Kyle

flyingvince, 2016-07-22:
I was able to follow your suggestion earlier and add a `pdebug` line inside the `plc_tag_destroy` function. 

The log files are attached. I experienced an issue that caused a kernel error:
```
kernel:  [ 4851.942219] traps: python3[2882]  
general protection ip:7fe87c0bc782 sp:7fe87b0a79e0
error:0 in libplctag.so[7fe87c0ac000+16000]
```

The sequence leading to that kernel error is as follows:
```
2016-06-22 05:11:11 plc_tag_abort:187 Starting.
2016-06-22 05:11:11 plc_tag_read:344 elapsed time 1000ms
2016-06-22 05:11:11 plc_tag_read:347 Done
2016-06-22 05:11:11 recv_eip_response_unsafe:144 Error reading socket! rc=-14
2016-06-22 05:11:11 request_destroy_unsafe:194 Starting.
2016-06-22 05:11:11 request_remove_unsafe:128 Starting.
2016-06-22 05:11:11 request_remove_unsafe:150 Done.
2016-06-22 05:11:11 request_destroy_unsafe:200 Done.
2016-07-22 05:11:11,852 (UTC)   ERROR   Could not read tag with tag_details : protocol=ab_eip&gateway=192.168.1.1&path=1,0&cpu=controllogix&elem_size=4&elem_count=2&name=NAM&debug=1, status = -11
2016-06-22 05:11:11 plc_tag_destroy:231 Starting. (0x7fe870050e20)
2016-06-22 05:11:11 ab_tag_destroy:376 Starting.
2016-06-22 05:11:11 plc_tag_abort:187 Starting.
2016-06-22 05:11:11 ab_tag_destroy:390 Removing tag from session.
2016-06-22 05:11:11 session_remove_tag:390 entering critical block 0x7fe874012ac0
2016-06-22 05:11:11 session_remove_tag_unsafe:356 Starting
```
This is where it ends: the call to the `session_remove_tag_unsafe` seems to provoke the `kernel general protection` error in the case of a `recv_eip_response_unsafe:144 Error reading socket! rc=-14`.

I added the application server log file as well (`server_log.txt`).

Do you see anything abnormal in the `pdebug_log.txt` file? Any insights?

Thank you.

[pdebug_log.txt](https://github.com/kyle-github/libplctag/files/377731/pdebug_log.txt)
[server_log.txt](https://github.com/kyle-github/libplctag/files/377732/server_log.txt)

kyle-github, 2016-07-23:
I am looking at this.  Something is odd.  Just before the snipped you have above, there are many lines where we get back an error of -14, which is a read error.  This implies that something is wrong with the socket itself.

I will need to add some debugging to get more information about what error it is, but in the mean time, are you using any Python threading or process creation in your code?

There is a problem with some Python libraries (particularly using threads and processes) where if you create something using an external C library and then fork off a thread/process, the object in question may not actually be allocated in the process that is trying to use it.

This may not be the case here.  But, it is worth looking at.  It is also possible that your code is overrunning the PLC by creating too many TCP connections in too short of a time.  The PLC cleans up things fairly fast, but I think it might be asynchronously.  So, even if you close your tag and the socket is cleaned up, it is possible that resources are still allocated on the PLC side.   

Is there a problem with leaving all of your tags allocated and then just calling plc_tag_read() when you need data?  The actual tag structure in memory does not take much space.  You can have thousands without using up a lot of memory (especially compared to a Python program).  These will use and share one single TCP socket that will last for the duration that your program is running.

What is the model of your PLC?

I will try to create a debug version of the library that has a lot more debugging so that we can see what is happening in the destroy function.  And, I want to see what error we see from the socket functions.   Something is failing there.  That said, it seems odd that the socket would fail at that point and not earlier when the socket is created.  I will need to go through the log carefully to see if I can find a pattern.

At this point my best guesses are that you are somehow creating the tag in one process and using it in another (could be threads), or something is overrunning the PLC and it is returning errors.  It isn't looking like a double free since the addresses are different each time but I haven't done a very detailed analysis.

Thanks for helping with the testing and the logs!

kyle-github, 2016-07-23:
I looked a little more at the logs and something is definitely wrong here:

2016-06-22 05:11:10 send_eip_request_unsafe:41 Starting.
2016-06-22 05:11:10 pdebug_dump_bytes_impl:1133 Dumping bytes:
00000 6f 00 2c 00 45 52 02 12 00 00
00010 00 00 20 1b 06 70 e8 7f 00 00
00020 00 00 00 00 00 00 00 00 01 00
00030 02 00 00 00 00 00 b2 00 1c 00
00040 52 02 20 06 24 01 0a 05 0e 00
00050 52 03 91 03 4e 41 4d 00 02 00
00060 00 00 00 00 01 00 01 00
2016-06-22 05:11:10 send_eip_request_unsafe:103 Done.

So, we have sent the read request bytes.  Now we need to wait for  the response.

2016-06-22 05:11:11 recv_eip_response_unsafe:165 request received all needed data.

OK.  Got the response.    But, now what is this error?

2016-06-22 05:11:11 recv_eip_response_unsafe:144 Error reading socket! rc=-14
2016-06-22 05:11:11 recv_eip_response_unsafe:144 Error reading socket! rc=-14

This occurs in the IO thread, not the thread that runs the tag object (that is the client thread).  I need to see what is happening here, but it looks like we are getting a socket error immediately after we read all the data.   After this, we see that there was a timeout (which seems a little odd).  

Are you calling plc_tag_status() at any point after the read or are you calling plc_tag_read() with a timeout?

flyingvince, 2016-07-25:
Hi Kyle,

```
I will need to add some debugging to get more information about what error it is, but in 
the mean time, are you using any Python threading or process creation in your code?
```
I am using Python threads, yes. But I am accessing the PLC only from one thread, and I import libplctag Python module directly from that specific thread.

```
It is also possible that your code is overrunning the PLC by creating too many TCP 
connections in too short of a time. The PLC cleans up things fairly fast, but I think it 
might be asynchronously. So, even if you close your tag and the socket is cleaned up, 
it is possible that resources are still allocated on the PLC side.
```
Very likely: I am querying data from the PLC in a loop, and essentially as fast as possible. I noticed that when I increment the number of tags queried from the PLC then think gets worse (issues are more frequent + I am getting the kernel error `general protection error in libplctag.so` described earlier).

```
What is the model of your PLC?
```
Allen-Bradley Logix 5573

```
Are you calling plc_tag_status() at any point after the read or are you calling 
plc_tag_read() with a timeout?
```
I am not calling the plc_tag_status() and I am calling the plc_tag_read() with a 1000 ms timeout.

```
Is there a problem with leaving all of your tags allocated and then just calling plc_tag_read() 
when you need data? 
```
This seems the next step for me to work on, I will implement that suggested approach.

kyle-github, 2016-07-31:
Can you please rebuild the library and try to do the same test again?  The code is:

```c

extern int socket_read(sock_p s, uint8_t *buf, int size)
{
    int rc;

    if(!s || !buf) {
        return PLCTAG_ERR_NULL_PTR;
    }

    /* The socket is non-blocking. */
    rc = read(s->fd,buf,size);

    if(rc < 0) {
        if(errno == EAGAIN || errno == EWOULDBLOCK) {
            return PLCTAG_ERR_NO_DATA;
        } else {
            pdebug(1,"Socket read error: rc=%d", rc);
            return PLCTAG_ERR_READ;
        }
    }

    return rc;
}

```

This replaces the routine of the same name in platform.c (the Linux one).

Something strange is going on with the socket itself.  It is possible that it is a red herring, but I want to make sure.  Thanks.

flyingvince, 2016-08-02:
After implementing the approach you suggested (`leaving all of your tags allocated and then just calling plc_tag_read() when you need data`) I did not experience any instabilities and/or issues related to kernel error.

At this stage the problem is solved with your suggested approach and it does not introduce any drawback. The tags are created **once**, and I call the `plc_tag_read()` with a timeout of 1000 ms on all my tags, in a loop. `plc_tag_destroy()` is never called, unless the application is stopped by the user.

Your guess about the code `overrunning the PLC by creating too many TCP 
connections in too short of a time` was probably correct.

At this stage I'd suggest we close the issue. Thanks for your help!

kyle-github, 2016-08-02:
I think that there is still a bug here.  My code should not be failing even if the socket is in a bad state.  If you still can, I would appreciate it if you could take your older code and use the socket_read() function above.  That may tell me what actual error the read() function is returning.  I need to understand why that is causing a failure.  At this point, it is the only thing I can see that is clearly going wrong.

I have implemented a level of indirection in a test version of the library that I hope to release in a few months.  That will, hopefully (!), prevent any accidental double-free or use-after-free problems.  It is not as portable as I would like, but I can't change the API yet.

flyingvince, 2016-08-04:
I did test again with the old code (using `pdebug(1,"Socket read error: rc=%d", rc);` and I experienced two kernel issues:

```
Aug 04 08:17:05 kernel:  [80289.489742] traps: python3[10480] general protection ip:7f72d1334186 sp:7f72cdf7f2a0 error:0 in libc-2.23.so[7f72d12fd000+1c0000]
Aug 04 08:17:05 kernel:  [80289.977279] veth4d18f4d: renamed from eth0
Aug 04 08:17:05 kernel:  [80289.990223] docker0: port 1(veth1388dd6) entered disabled state
Aug 04 08:17:05 kernel:  [80290.032604] docker0: port 1(veth1388dd6) entered disabled state
Aug 04 08:17:05 kernel:  [80290.034092] device veth1388dd6 left promiscuous mode
Aug 04 08:17:05 kernel:  [80290.034099] docker0: port 1(veth1388dd6) entered disabled state
```
-> see log file `export_8abbf85401eb.log`

and also 
```
Aug 04 08:31:25 kernel:  [  626.210556] traps: python3[1682] general protection ip:7ff4da347dee sp:7ff4d6f4ad60 error:0 in libc-2.23.so[7ff4da2c6000+1c0000]
Aug 04 08:31:25 kernel:  [  626.635642] docker0: port 1(vethe1da6eb) entered disabled state
Aug 04 08:31:25 kernel:  [  626.635780] veth5f2065a: renamed from eth0
Aug 04 08:31:25 kernel:  [  626.769057] docker0: port 1(vethe1da6eb) entered disabled state
Aug 04 08:31:25 kernel:  [  626.770479] device vethe1da6eb left promiscuous mode
Aug 04 08:31:25 kernel:  [  626.770487] docker0: port 1(vethe1da6eb) entered disabled state
```
-> see log file `export_4b0556d77646.log`

Does it help in any way? 

[Archive.zip](https://github.com/kyle-github/libplctag/files/400748/Archive.zip)

kyle-github, 2016-08-06:
Hi Vincent,

Are you sure that you ended up using the version of the library with the new debug code?  I am not seeing any of the errors that should be printed from the new pdebug line.  But I do see  the error that prints farther up the call chain.  So, it looks like the new code is not being used.

The errors you are seeing in the container seem odd.  Nothing that the code running in the container should cause the container to behave badly!  That's really weird.

flyingvince, 2016-08-11:
Hi Kyle,

Whoops... I made a mistake in the build process, will re-test again next week. Hopefully the pdebug will show up then ;-)

kyle-github, 2016-08-12:
Phwew.  OK.   Please let me know when you have a chance to run it.  I appreciate all your help.  Even if the other solution I proposed made the problem go away for you, the fact that you had it at all is a concern!

flyingvince, 2016-08-22:
Kyle,

Find attached the log file following the re-test. I did see quite of few of the `Socket read error:` I added in the C code, so hopefully this will help figure out what happened.

However I did not experience any `kernel / traps / libplctag.so` erorrs like I did when I first submitted the issue here, unfortunately.

[v121_19Aug2016.log.zip](https://github.com/kyle-github/libplctag/files/429710/v121_19Aug2016.log.zip)

kyle-github, 2016-08-22:
Thanks! I will try to look at this as soon as possible.  It is a little odd that the kernel errors are not showing up.  Was there an update of your container infrastructure in between tests?

kyle-github, 2016-08-24:
Ugh.  I forgot a part in the debug line.  Sigh.

I need to print the errno value, not the return code.

```c
extern int socket_read(sock_p s, uint8_t *buf, int size)
{
    int rc;

    if(!s || !buf) {
        return PLCTAG_ERR_NULL_PTR;
    }

    /* The socket is non-blocking. */
    rc = read(s->fd,buf,size);

    if(rc < 0) {
        if(errno == EAGAIN || errno == EWOULDBLOCK) {
            return PLCTAG_ERR_NO_DATA;
        } else {
            pdebug(1,"Socket read error: errno=%d", errno);
            return PLCTAG_ERR_READ;
        }
    }

    return rc;
}
```

I am sorry about that.  Can you run a short test until you see the error again.  It looks like it happens a lot.

flyingvince, 2016-08-24:
Ok I will test again (most likely next week), no problem.

kyle-github, 2016-08-24:
Thanks.  Sorry for the constant changes.

kyle-github, 2016-08-29:
This might not be related, but I am building a stress test program to test out the new 1.5 code and ran into a problem where the PLC is giving me errors after about 140 connections are created and dropped.  I am still trying to determine if it is due to PLC resources running out or my code reusing some ID (session or connection).  

The test is just creating a tag, reading and writing to it, destroying and repeating as fast as it can.  Again, it might be something completely different from what you are running into.

flyingvince, 2016-09-02:
I tried again but I could not generate any `Socket read error`, however I did experience another

 `
kernel:  [ 1591.441833] traps: python3[1978] general protection ip:7f50033c3177 sp:7f50003574e0 error:0 in libc-2.19.so[7f5003389000+1ba000]
`

which seems to originate from a `*** Error in python3: malloc(): memory corruption: 0x00007f4ff800b5f0 ***`. 

Can you see anything suspect in the log file attached around `04:55:28`?

On a side note, there seem to be an error in the month reported by pdebug() (showing `08` instead of `09`), just FYI.

kyle-github, 2016-09-05:
Thanks for the run.

I will look at the time stamp printing code.  Perhaps there is an error in it.

The error is at the very end of the log file . It is not clear what is happening, however, I do see some odd things in the log.  It looks like we should be seeing data being sent just before the Python malloc error, but while the calls show up in the log, I do not see the byte dumps of the packet.  So, I suspect that malloc is failing.

This limits it to a fairly small area of code.

I am not sure what is going on.  It is possible that memory is running out and malloc is failing.  If that is the case, then some of the error handing in the code should be catching that.  But, possibly there is a path where the failure does not get caught and null pointers are passed down.  If that is the case, then I could see a failure where memory gets overwritten.

I will need to try to recreate this.  I have run several tests that appear to do the same thing and they never seem to leak memory or have any problems.  

Thanks very much for the logs.

Best,
Kyle

kyle-github, 2016-09-05:
Here is the part:

2016-08-02 04:55:28 plc_tag_read:280 Starting.
    2016-08-02 04:55:28 eip_cip_tag_read_start:122 Starting
    2016-08-02 04:55:28 allocate_request_slot:275 Starting.
    2016-08-02 04:55:28 allocate_request_slot:280 setting max_requests = 10
    2016-08-02 04:55:28 eip_cip_tag_read_start:156 First read tag->num_read_requests=1, byte_offset=0.
    2016-08-02 04:55:28 build_read_request:374 Starting.
    2016-08-02 04:55:28 request_add:103 Starting. sess=0x7f4ff800ae90, req=0x7f4ff800b290
    2016-08-02 04:55:28 request_add_unsafe:69 Starting.
    2016-08-02 04:55:28 request_add_unsafe:89 Done.
    2016-08-02 04:55:28 request_add:109 Done.
    2016-08-02 04:55:28 build_read_request:489 Done
    2016-08-02 04:55:28 eip_cip_tag_read_start:188 Done.
    2016-08-02 04:55:28 check_read_status:871 calling ab_rag_read_cip_start_unsafe() to get the next chunk.
    2016-08-02 04:55:28 eip_cip_tag_read_start:122 Starting
    2016-08-02 04:55:28 eip_cip_tag_read_start:156 First read tag->num_read_requests=2, byte_offset=0.
    2016-08-02 04:55:28 build_read_request:374 Starting.
    *** Error in `python3': malloc(): memory corruption: 0x00007f4ff800b5f0 ***
(END)

There should be some packet dumps in there.  Here is one where there is:

    2016-08-02 04:55:28 ab_tag_create:250 Done.
    2016-08-02 04:55:28 plc_tag_read:280 Starting.
    2016-08-02 04:55:28 eip_cip_tag_read_start:122 Starting
    2016-08-02 04:55:28 allocate_request_slot:275 Starting.
    2016-08-02 04:55:28 allocate_request_slot:280 setting max_requests = 10
    2016-08-02 04:55:28 eip_cip_tag_read_start:156 First read tag->num_read_requests=1, byte_offset=0.
    2016-08-02 04:55:28 build_read_request:374 Starting.
    2016-08-02 04:55:28 request_add:103 Starting. sess=0x7f4ff80074c0, req=0x7f4ff8008d30
    2016-08-02 04:55:28 request_add_unsafe:69 Starting.
    2016-08-02 04:55:28 request_add_unsafe:89 Done.
    2016-08-02 04:55:28 request_add:109 Done.
    2016-08-02 04:55:28 build_read_request:489 Done
    2016-08-02 04:55:28 eip_cip_tag_read_start:188 Done.
    2016-08-02 04:55:28 send_eip_request_unsafe:41 Starting.
    2016-08-02 04:55:28 pdebug_dump_bytes_impl:1132 Dumping bytes:
    00000 6f 00 2e 00 79 f9 02 12 00 00
    00010 00 00 c0 74 00 f8 4f 7f 00 00
    00020 00 00 00 00 00 00 00 00 01 00
    00030 02 00 00 00 00 00 b2 00 1e 00
    00040 52 02 20 06 24 01 0a 05 10 00
    00050 52 04 91 04 42 33 30 30 28 06
    00060 01 00 00 00 00 00 01 00 01 00
    
    2016-08-02 04:55:28 send_eip_request_unsafe:103 Done.
    2016-08-02 04:55:28 recv_eip_response_unsafe:165 request received all needed data.
    2016-08-02 04:55:28 session_check_incoming_data_unsafe:606 got full packet of size 50
    2016-08-02 04:55:28 pdebug_dump_bytes_impl:1132 Dumping bytes:
    00000 6f 00 1a 00 79 f9 02 12 00 00
    00010 00 00 c0 74 00 f8 4f 7f 00 00
    00020 00 00 00 00 00 00 00 00 01 00
    00030 02 00 00 00 00 00 b2 00 0a 00
    00040 d2 00 00 00 c4 00 00 00 00 00
    
    2016-08-02 04:55:28 check_read_status:706 processing request 0
    2016-08-02 04:55:28 check_read_status:765 type byte = 196 (c4)

kyle-github, 2016-09-10:
I have been going through the code.  I have a theory:  I think that a memory allocation is failing but I am not catching the error.  In that case, a null pointer (or worse, a random one) is being handed off to the socket routines.  Those may throw an error, or not, based on a null or random pointer.  If they do not throw an error, then we may be reading a response packet right over the top of random memory.  That would corrupt the malloc structs (if not generally cause horrible problems).

Again, just a theory.  I need to find a good way to inject memory allocation failures.

I will be trying to run some static code analysers against the source to see if they show any potential problems.

Of course, all this seems to require that the memory allocation fails for some reason.  Is your container large enough?  Could it be running out of memory or leaking memory somehow?

flyingvince, 2016-09-11:
I do not specify any memory limits on the container when launching it, and the underlying OS has access to 4GB RAM, so it should be plenty enough. I haven't noticed any memory leak, no.

kyle-github, 2016-09-11:
Hmm, so much for that theory.  

I ran the code through a static code checker from Clang.  It found a few possible problems, but nothing really in the call chain that seems to be a problem.  Those are fixed.

I wish I could figure out how to duplicate this locally.  Even if the underlying problem is not in my code, the library should handle it more gracefully.  And, of course, if it is in my code, it needs to be fixed!

It looked like you were using Mac OS?  What kind of container are you using? I may need to try to duplicate the whole environment to see what is going on.

flyingvince, 2016-09-11:
When I experienced the issue reported above, I was using a Docker container which was made of:
```
Distributor ID:	Ubuntu Server
Description:	Ubuntu 14.04.4 LTS
Release:	14.04
Codename:	trusty

uname -r: 4.4.15-moby
Python 3.4.3 (default, Oct 14 2015, 20:28:29) 
```

MacOS was only used to clean up the log files, it was not used when connecting to the PLC system.

kyle-github, 2016-11-01:
Version 1.5 has been released.  This should stop any of these kinds of problems in the future.  I ran even more static checks and a very long burn-in period with Valgrind.  Nothing shows up.    I do not consider this bug closed at this point because I still cannot reproduce it.

kyle-github, 2016-11-22:
I am reconsidering.   I think that issue #38 may be related to this.   However #38 is also due to the changes in how I separated out parts of tag set up so that it would not block.  I will keep digging.

kyle-github, 2016-12-18:
I reworked the destructor code significantly in 1.5.1.  I will close this now.  I kind of want to leave this open because this is not a case that I seem to be able to cause on demand.  I cannot repeat it at all with the new code.

closed 2016-12-18 by kyle-github