libplctag/libplctag #31: Can anyone tell me how to browse AB controller tags?

KutapiAise, 2016-08-30:


kyle-github, 2016-08-30:
The library cannot browse for tags.  If you already know a tag's name, you can retrieve or write data.

If you need to introspect the tag names, I would suggest looking at the cpppo project:

https://github.com/pjkundert/cpppo

It is in Python, but seems to be well written and Perry is very quick to respond and fix bugs.  I will probably use it as a vehicle for writing tests for libplctag.  

Best,
Kyle

KutapiAise, 2016-09-01:
Thanks kyle for your insight!!. 
I will try to get in the python resource!!. 
Hope to see your implementation of "browsing" tags in libplctag asap!.

Best
Aish

kyle-github, 2016-09-06:
Libplctag will likely never browse tags.  There are a few rough edges to fix for Allen-Bradley PLCs, but most of the next development is going to be for increased performance and support for other kinds of PLCs.

kyle-github, 2017-03-28:
Actually....  A user gave me a link to a AB PLC Python library that does this.  I will try to support this in some future release.

flyingvince, 2018-02-09:
Kyle, do you have a timeframe in mind for implementing the ability to broswe AB tags? In the mean time can you share the link to the AB PLC Python library you referred to earlier?

flyingvince, 2018-02-12:
Hi Kyle,

I saw your response on Google Groups, I copied it here for the community. Thanks!

```
There are two libraries.   Perry's cccpo library is more of a toolkit
with which to build applications and supports more protocols 
than AB.  But it is more complicated to use.

There is also dmroeder's pylogix library.  This is just AB CIP/EIP but 
seems like it is pretty easy to use.  It also allows you to scan for 
PLCs on the local network and scan a Control/CompactLogix PLC for tags.   

Both libraries are pure Python I think.  I am not a Python programmer, so you 
should take my opinions with a grain of salt!

Best,
Kyle
```

kyle-github, 2018-02-14:
The time frame for supporting tag listing is probably a few weeks.   I have a prototype working, but I did it on a very unstable branch off of 2.0.   I am probably going to abandon that branch as it is requiring too much new code.   That said, I will try to retrofit the tag listing into the existing 1.5 branch.  Somehow.

kyle-github, 2018-12-26:
I am going to close this in favor of ER #46.   Now that 2.0 is out, I can focus on this work.

closed 2018-12-26 by kyle-github

afiya-g, 2019-05-06:
Hello Kyle,

Does current version supports now browsing of AB controller tags?
I'm using libplctag lib for my client project, and the requirement is to browse the AB controller tags.
If the current version is supporting, can you please let know how to do it.

Regards,
Syed.

kyle-github, 2019-05-06:
No, the library does not currently support browsing.

If you can use Python, you should look at D. M. Roeder's pylogix library on
GitHub.

I hope that helps!

Best,
Kyle