libplctag/libplctag #32: unable to pull the data after half an hour continuous read

KutapiAise, 2016-09-24:
Kyle, 

I have a list of 10  BOOL tags namely tag1, tag2...tag10[controller tags] in controllogix . I 'm continuously pulling the state of these tags and logging in to Database. However, after about  around half an hour, i fail to read the state of these tags.
Give me your insight in to  the issue.

Best 
Aishwarya

kyle-github, 2016-09-27:
Are you closing the tags when you are done with plc_tag_destroy()?  

If you have many tags, it is better to have logic that looks like this:

1. open all tags.
2. wait for plc_tag_create() calls to finish and all tags to go into PLCTAG_STATUS_OK state. 
3. read from all tags, async: plc_tag_read(tag[i],0). 
4. wait for reads to complete. 
5. get data from tags. 
6. repeat as long as you want to continue reading.
7. delete tags.

This is the most efficient way to do this.

The library does not change the value of the tag handle.  If it is becoming NULL, then something in your code is overwriting it or there is a memory overwrite problem somewhere.  Take a look at the data_logger example program in the examples.  Can you modify it to do something similar to what you want to do?

kyle-github, 2016-11-18:
Is this still a problem.  If so, please provide a test case.  If not, please let me know and I will close the issue.  Thanks.

kyle-github, 2016-11-29:
Please reopen this if you see it again.

closed 2016-11-29 by kyle-github