libplctag/libplctag #33: PLC 5, MicroLogix: need to support single bit reading

kyle-github, 2016-09-27:
The code currently will not translate N7:1/2 correctly.  It will handle special fields like .ACC.

kyle-github, 2016-10-07:
This might be an issue with 3-address forms.  Also, it looks like reading a bit is different from reading a word.  The commands may be different.

kyle-github, 2016-11-19:
This is going to require separating out the protocols into CIP and PCCC most likely.  It is also not clear that the existing API would easily allow for this as there is no inherent boolean type supported.  

Given the ubiquity of bool/coil, perhaps it makes sense to extend the API.

amacierzynski, 2017-06-21:
Is that mean that we will not be able to read data from Boolean data types from our MicroLogix 1400, or this is only for Integers as you put as an example in the post? 

Regards,
Adam

kyle-github, 2017-06-23:
Unfortunately in my tests, using the 0x0f/0x68/0x67 commands shows that the PLC/5, at least, returns an error if there is a third address element.  It will not send back the one bit.   I just did some more checking and found that this applies to both N and B data files at least.  

You can read a boolean file, B, but cannot give it a third address element: B3:4.2 (which does not make sense).   When you read a boolean element, you want to specify the data file and the element, but not a third element: B3:4.   The PLC will return a 16-bit word with all the bits.  

TL;DR you can read the data from integers and booleans but  you will have to mask the bits off yourself.

I have not made any changes to the library to support bit reads and writes yet.   I have to finish more of the refactoring in the 2.0 version of the library.

kyle-github, 2018-12-26:
Closing in favor of #82.   Boolean support is not good for most types of PLC.

closed 2018-12-26 by kyle-github