libplctag/libplctag #35: Multi-threaded Capabilities

kieranHay, 2016-10-26:
Thanks in advance for taking the time to read this.

I have multiple AB PLCs that I create, connect, read & write on different threads so that they run in parallel. However when a PLC is offline I try to reconnect/create the tags again until it succeeds. My question is, should this affect the other PLCs that are connected and are reading/writing?

At the moment I am having issues that it seems to be blocking the other PLC threads.

Thanks again.

kyle-github, 2016-11-01:
Hi,

Sorry for the delay.  I was trying to get version 1.5 out.

Are you using hostnames or IP addresses?  Hostnames require DNS look ups and those were fairly buggy in verison 1.0.  They are better in 1.5.  

The connect call is generally blocking, so if a PLC goes offline and you try to connect to it, a number of things could happen.  If the PLC just went offline, then it is possible that the network in between the application PC and the PLC has not realized this yet and you may hit timeouts.  If it has been a while (OS and setting dependent I believe), then it should fail quickly.  But, this should only effect one thread at a time.

However, if you are using different threads for every tag (I am not sure if that is what you were saying), then you should not have an overall hang.  I wonder if there is a mutex contention problem internally in the library?    

Can you provide more information about how your application uses the library so that I can try to duplicate this?

kyle-github, 2016-11-12:
I have been looking at this.  I think that you've stumbled on a mutex contention problem.

Here are the events that look like they will block in this case:

* create tags 1 and 2.
* do operations on both tags for a while.
* tag 2's PLC goes offline and tag 2 is closed.
* the application tries to reopen tag 2.
* the tag creation process continues fine until it gets to the point of opening the socket to PLC 2.  But PLC2 is not there.  In some circumstances, the network may not fail immediately to connect.  But, during this time, the tag creation has a central mutex locked.
* at the time this is happening, the other tag tries to read.  Part of that process involves queuing a request packet against the session object that is being created.  That involves locking the same mutex as in the step above.

This seems like a bug that needs to be fixed.  Unfortunately, it is not a simple one as the logic flow in question will need to be broken up into multiple parts.  Thanks for bringing this to my attention!

kieranHay, 2016-11-12:
Sorry for not getting back to you. Was going to create a simple test project showing the issue (and make sure I didn't mess up) but never got round to it. However you have found the issue. 
Thanks for investigating, I have gotten around this for now by using multiple programs for each plc. 

If you would like me test anything or help in anyway, I'm always willing to. Thanks again!

kyle-github, 2016-11-19:
This is turning into a bit of a mess.  It turns out that in some circumstances, the Allen-Bradley PLCs will drop packets!  I need to solve that urgently.

kyle-github, 2016-12-18:
This should be fixed in 1.5.1.

closed 2016-12-18 by kyle-github