libplctag/libplctag #36: ER: make PLC selection easier

kyle-github, 2016-11-19:
It is currently confusing for users to have to enter the protocol and CPU type.  

There seem to be a few options:

1. simplify the PLC selection to one value and create a look up table on the Wiki.
1. use make/family/model in the tag string.  This requires an internal lookup table in the library code though and search functionality.
1. use the existing protocol/cpu, but put a better look up table on the Wiki.
1. use the manufacturer part numbers directly.  This could be done either in the library or on the Wiki.
1. other?

I am leaning toward the first one.

xinthose, 2017-01-18:
I use an enumeration to simplify it:
```
enum PLC_PROT    // Protocols, see "check_cpu" in ab_common.c
{
     PROT_AB_PLC = 1,    // PLC, PLC5, SLC, SLC500
     PROT_AB_MLGX800 = 2,    // MicroLogix 800
     PROT_AB_MLGX = 3,    // MicroLogix 1000-1400
     PROT_AB_LGX = 4,        // CompactLogix, ControlLogix, FlexLogix
};
```

kyle-github, 2017-01-18:
If I rewrite the library in C++, I will think about heading this direction.  One of the problems is that I want to export a common interface for all types of PLC so that you do not have to have PLC-specific tag creation functions.

And I am thinking about using C++.  I found some interesting ways to keep the library size down.  Unfortunately, it is still many times larger than the C version.  I have not worked with uLibC++ yet.

kyle-github, 2017-03-28:
The first part of this is done.  Tag will be constructed based on make/family/model.   This is used with the new system special tags.   All the remaining work to support the AB families will be mostly splitting up existing code and refactoring.

closed 2017-03-28 by kyle-github