libplctag/libplctag #38: Memory Leak on Failed connection.

kieranHay, 2016-11-21:
It seems I have hit a memory leak. However you may already know about the issue as theres a comment above the destroy function:

 * FIXME - this leaves a dangling pointer.  Should we take the address
 * of the tag pointer as an arg and zero out the pointer?  That may not be
 * as portable.
 
It looks like the plc_tag variable isn't destroyed because tag->mut is NULL meaning  tag->vtable->destroy(tag); function will not get called. 

The tag->mut is NULL because it doesn't get past if(tag && tag->status == PLCTAG_STATUS_OK) within plc_tag_create because the connection is failing. The IP address I have passed is not valid so the API is giving a PLCTAG_ERR_BAD_GATEWAY. 

I was doing some stress tests and it doesn't take long for the PC to grind to a halt if you are looping around creating, connecting and attempting to destroy the tag because it failed.

I am using a controllogix as the test PLC but I don't think it makes much difference.

Thanks again!

kyle-github, 2016-11-21:
OK, I will add this to the list.  Thanks very much for doing this testing!    This sounds like it is not going to be too hard to fix.  Creating an indirection between the tag pointer and what the clients of the library use was a good first step so most of the infrastructure is there.  I will need to rethink the destruction process carefully because I think it can be simplified now.

kyle-github, 2016-12-18:
This should be fixed in release 1.5.1.

closed 2016-12-18 by kyle-github