libplctag/libplctag #39: Change debug after tag creation.

kieranHay, 2016-11-27:
Not a issue or rather important.

At the moment the only way to change the debug open is to recreate the tag. It would be nice if there was a method plc_tag_set_debug within the API.  It gives the ability to turn it on/off while the application using the API is running.

Thanks again!

kyle-github, 2016-11-27:
You can fake this by creating another tag while the first one is still around.  The debug level is now "global" and it is set by the last tag created.

kyle-github, 2017-01-18:
Changing milestone to 1.6 as that is the next API update.

kyle-github, 2017-03-28:
This is now supported in version 1.5.4.  It is not a new API function.  Instead it uses special tags and the existing tag API to read and write the value in the library.  This will be used in the future to allow for special tags for each protocol type and other uses.

closed 2017-03-28 by kyle-github