libplctag/libplctag #41: tag_rw error

QuinDennis, 2016-12-08:
// sorry, but I can find any docs on what the tag read error means.  this is trying to read a single
// float from Softlogix controller in slot 2.
//
tag_rw -t real32 -p 'protocol=ab_eip&gateway=192.168.0.111&path=1,2&cpu=LGX&elem_size=4&elem_count=1&name=AWARE_OPTDEMAND'
Arg[0]=./build/tag_rw
Arg[1]=-t
Arg[2]=real32
Arg[3]=-p
Arg[4]=protocol=ab_eip&gateway=192.168.0.111&path=1,2&cpu=LGX&elem_size=4&elem_count=1&name=AWARE_OPTDEMAND
INFO: Got tag 0x4001
INFO: tag status 0
INFO: reading tag 0x4001
ERROR: tag read error, tag status: PLCTAG_ERR_REMOTE_ERR

kyle-github, 2016-12-08:
Add the directive &debug=3 to the tag attribute string.  Then capture the output.  Sometimes this will help us debug it.  

Please post the question to the forum:

https://groups.google.com/forum/#!forum/libplctag

More people will look at the problem and help find solutions.  It also makes it easier for people to find the answer later!

I do not know if anyone has attempted to talk to a SoftLogix before.

There are a couple of things to check:

1. Is the tag in question a controller tag or a procedure (lower level) tag?  Try with a controller tag first.
1. The path looks a bit odd.  Does the SoftLogix really emulate the CPU module at the third slot in the chassis?  The hardware chassis slots are counted from left to right starting at zero.  I think the CompactLogix uses 0,0 or just ignore the path.  Try 1,0 or 1,1 as well.
1. As I mentioned above, please add the debug flag to the tag attribute string.  It should look like this:

`'protocol=ab_eip&gateway=192.168.0.111&path=1,2&cpu=LGX&elem_size=4&elem_count=1&name=AWARE_OPTDEMAND&debug=3'`

The path looks like the most likely problem.  Double check that the tag is at the controller level (this is not required but makes debugging easier).

I really hope we can get this to work!  I seem to remember reading somewhere that SoftLogix had some restrictions on networking.  I am trying to find where I read that.

There is a possibility that the SoftLogix ignores the path, but the error seems to indicate that it does not.  

Please let me know if any of these things works.   If not, the debug output would be helpful.  Thanks for reporting this!

QuinDennis, 2016-12-09:
[image]

Hi Kyle, thanks for your response!  I added &debug=3, the output didn't change, but tag_rw hung and never completed.

I tried the forum, but couldn't access it.

kyle-github, 2016-12-09:
So, a little searching on Google gives this:

> Error code 08: Service not supported. The requested service was not implemented or was not defined  or this Object.

Hmm, I haven't seen that one before.  That isn't a path problem.

The service in this case is reading the tag.  I cannot find the list of restrictions I once found about SoftLogix.  

Can you try the test again, but change the CPU type to micro800: cpu=micro800?  This changes the internal behavior of the library to use connected messaging instead of unconnected messaging.

Is the tag you are trying to read a controller tag?

kyle-github, 2016-12-09:
Hmm, I think maybe this is not going to work with your SoftLogix configuration.  Here is a manual:

[AB SoftLogix User Manual](http://literature.rockwellautomation.com/idc/groups/literature/documents/um/1789-um002_-en-p.pdf)

Note that chapter 3 talks about how to set up networking.  It looks like you need to add a network module to the SoftLogix chassis.  I do not see that in the screenshot above.  There is also a change you need to do to RSLinx.

Can you please add a network module as described in the above manual?  I think that might be necessary.  I am not sure, but it looks possible.

QuinDennis, 2016-12-12:
I'm also having issues just connecting a node-to-node or node-to-C# socket application.  I'm working with a VBox from a colleague.  Turns out to have Symantec end-point protection on it, which I think is the issue.  I'm spinning up a new W7 image and will try with that.  So, all may not be lost.

kyle-github, 2016-12-12:
Ah, that would do it too.  The port for AB's EIP protocol is 44818 (TCP).  That is where the PLC listens.

That said, please check the network module in the SoftLogix.  It does look to me like you need to add a virtual network module to the SoftLogix before it can talk to the outside world.  There is some passthrough that works via RSLinx (hence the "modules" that are installed), but _incoming_ connections appear to need a network module.  

There is emulation of hardware and there is over-emulation of hardware...

Is there a way to download SoftLogix and run  it in trial mode for a few days without a key?  I poked around a little bit this weekend, but did not find anything obvious.

QuinDennis, 2016-12-14:
Took forever to get everything installed, etc.  I've added an ENET module (slot 3, LGX in slot 2), and run the test again.
[image]

./tag_rw -t real32 -p 'protocol=ab_eip&gateway=192.168.0.119&path=3,2&cpu=LGX&elem_size=1&elem_count=1&name=AWARE_OPTDEMAND&debug=3'
Arg[0]=./tag_rw
Arg[1]=-t
Arg[2]=real32
Arg[3]=-p
Arg[4]=protocol=ab_eip&gateway=192.168.0.119&path=3,2&cpu=LGX&elem_size=1&elem_count=1&name=AWARE_OPTDEMAND&debug=3
thread(0001) 2016-11-14 11:03:31.940 ab_tag_create:90 Starting.
thread(0001) 2016-11-14 11:03:31.940 setup_session_mutex:513 Starting.
thread(0001) 2016-11-14 11:03:31.940 setup_session_mutex:526 Done.
thread(0001) 2016-11-14 11:03:31.940 ab_tag_create:156 entering critical block 0x13318e0
thread(0001) 2016-11-14 11:03:31.940 ab_tag_create:169 leaving critical block 0x13318e0
thread(0001) 2016-11-14 11:03:31.940 match_dhp_node:117 Bad syntax in DH+ route.  Expected : in route.
thread(0001) 2016-11-14 11:03:31.940 match_dhp_node:117 Bad syntax in DH+ route.  Expected : in route.
thread(0001) 2016-11-14 11:03:31.940 session_create_unsafe:428 Starting
thread(0001) 2016-11-14 11:03:31.940 session_connect:497 Starting.
thread(0001) 2016-11-14 11:03:31.940 session_connect:517 Done.
thread(0001) 2016-11-14 11:03:31.940 session_register:599 Starting.
thread(0001) 2016-11-14 11:03:31.940 session_register:634 sending data:
Dumping bytes:
00000 65 00 04 00 00 00 00 00 00 00
00010 00 00 00 00 00 00 00 00 00 00
00020 00 00 00 00 01 00 00 00 00
thread(0001) 2016-11-14 11:03:31.944 session_register:710 received response:
Dumping bytes:
00000 65 00 04 00 01 00 02 40 00 00
00010 00 00 00 00 00 00 00 00 00 00
00020 00 00 00 00 01 00 00 00 00
thread(0001) 2016-11-14 11:03:31.944 session_register:732 Done.
thread(0001) 2016-11-14 11:03:31.944 session_create_unsafe:481 Done.
thread(0001) 2016-11-14 11:03:31.944 ab_tag_create:259 Done.
thread(0001) 2016-11-14 11:03:31.944 plc_tag_create:212 Returning mapped tag 0x4001
INFO: Got tag 0x4001
INFO: tag status 0
INFO: reading tag 0x4001
thread(0001) 2016-11-14 11:03:31.944 plc_tag_read:436 Starting.
thread(0001) 2016-11-14 11:03:31.944 eip_cip_tag_read_start:138 Starting
thread(0001) 2016-11-14 11:03:31.944 build_read_request_unconnected:510 Starting.
thread(0001) 2016-11-14 11:03:31.944 build_read_request_unconnected:623 Done
thread(0001) 2016-11-14 11:03:31.944 eip_cip_tag_read_start:212 Done.
thread(0002) 2016-11-14 11:03:31.945 send_eip_request_unsafe:42 Starting.
thread(0002) 2016-11-14 11:03:31.945 send_eip_request_unsafe:104 Done.
thread(0002) 2016-11-14 11:03:31.947 session_check_incoming_data_unsafe:609 got full packet of size 48
Dumping bytes:
00000 6f 00 18 00 01 00 02 40 00 00
00010 00 00 a6 49 db 15 00 00 00 00
00020 00 00 00 00 00 00 00 00 01 00
00030 02 00 00 00 00 00 b2 00 08 00
00040 d2 00 01 01 11 03 01 00 00
thread(0001) 2016-11-14 11:03:31.952 check_read_status_unconnected:1233 processing request 0
thread(0001) 2016-11-14 11:03:31.952 check_read_status_unconnected:1272 CIP read failed with status: 1
thread(0001) 2016-11-14 11:03:31.952 check_read_status_unconnected:1273 Connection error!
thread(0001) 2016-11-14 11:03:31.952 check_read_status_unconnected:1425 Error received!
thread(0001) 2016-11-14 11:03:31.952 plc_tag_read:510 elapsed time 8ms
thread(0001) 2016-11-14 11:03:31.952 plc_tag_read:513 Done
ERROR: tag read error, tag status: PLCTAG_ERR_REMOTE_ERR
thread(0001) 2016-11-14 11:03:31.952 plc_tag_destroy:407 Starting.
thread(0001) 2016-11-14 11:03:31.952 plc_tag_destroy_mapped:361 Starting.
thread(0001) 2016-11-14 11:03:31.952 plc_tag_abort_mapped:298 Starting.
thread(0001) 2016-11-14 11:03:31.952 plc_tag_abort_mapped:315 Done.
thread(0001) 2016-11-14 11:03:31.952 ab_tag_destroy:379 Starting.
thread(0001) 2016-11-14 11:03:31.952 plc_tag_abort_mapped:298 Starting.
thread(0001) 2016-11-14 11:03:31.952 plc_tag_abort_mapped:315 Done.
thread(0001) 2016-11-14 11:03:31.952 session_destroy_unsafe:527 Starting.
thread(0001) 2016-11-14 11:03:31.952 session_destroy_unsafe:566 Done.
thread(0001) 2016-11-14 11:03:31.952 ab_tag_destroy:428 done
thread(0001) 2016-11-14 11:03:31.952 plc_tag_destroy_mapped:397 Done.
thread(0001) 2016-11-14 11:03:31.952 plc_tag_destroy:416 Done.
quin@quin-ubuntu:/media/quin/OS/Share/RA/libplctag-master/build$

kyle-github, 2016-12-14:
If I am reading the errors correctly, it looks like:

> This error occurs when a port number specified in a port segment is not available. Consult the documentation of the device to verify the correct port number.

So, the path looks suspect.  3,2 means "go out port 3 to slot #2".  I think that isn't correct.  Generally port "1" is the backplane port on an Ethernet module.  I think the path should be "1,3".  Please try with that.

The path syntax is not well documented (at least publically).  It seems to work in pairs of <port,target>  If you think of it as "route the message out port <port> to destination <target>" it is close.  I don't know what port 0 is, but port 1 is usually the backplane port (the backplane is considered a channel/bus like Ethernet or ControlNet).  Port 2 and 3 are usually the front-facing ports on a DHRIO module for DH+ channel 1 and 2.  I think that Ethernet modules use port 2 as well, but I have not managed to make multiple Ethernet hops work yet.

QuinDennis, 2016-12-14:
> message out port to destination " it is close. I don't know what port

Hi Kyle,

No luck, no change for 0,3 or 1,3.

kyle-github, 2016-12-15:
Hmm, this is annoying.   Can you try changing the CPU type to micro800?  Right now it is LGX.  Changing it to micro800 will cause it to use connected messaging.  There are some differences that might help debug this.

I see that somewhere I am not dumping out the outbound packet.  I need to do that.  I see the dump for the incoming packet.

QuinDennis, 2016-12-16:
I got my hardware back.  This is a DINT, ENET in slot0, L83 LGX in 
slot1.  This works fine!

Back on SoftLogix, changing to micro800 generates PLCTAG_ERR_TIMEOUT, so 
I'm thinking that's just not to be.

Thanks for your efforts!

My backup is a C# Sockets OPC app which I can reach from nodejs using 
sockets, a sort of OPC Tunnel to linux, which isn't too terrible for 
S.L., since Windows is involved anyway.  But your library works for 
hardware, which is great!  As time permits, I'll work with it more.  End 
game is to get data into a nodejs environment.  Node-ffi library looks 
like a good candidate to bring the c-to-node gap.

  ./tag_rw -t uint32 -p 
'protocol=ab_eip&gateway=192.168.0.81&path=1,1&cpu=LGX&elem_size=4&elem_count=1&name=AWARE_OPTRUN&debug=3'
Arg[0]=./tag_rw
Arg[1]=-t
Arg[2]=uint32
Arg[3]=-p
Arg[4]=protocol=ab_eip&gateway=192.168.0.81&path=1,1&cpu=LGX&elem_size=4&elem_count=1&name=AWARE_OPTRUN&debug=3
thread(0001) 2016-11-16 03:11:44.491 ab_tag_create:90 Starting.
thread(0001) 2016-11-16 03:11:44.491 setup_session_mutex:513 Starting.
thread(0001) 2016-11-16 03:11:44.491 setup_session_mutex:526 Done.
thread(0001) 2016-11-16 03:11:44.491 ab_tag_create:156 entering critical 
block 0x13b48e0
thread(0001) 2016-11-16 03:11:44.492 ab_tag_create:169 leaving critical 
block 0x13b48e0
thread(0001) 2016-11-16 03:11:44.492 match_dhp_node:112 Bad syntax in 
DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
thread(0001) 2016-11-16 03:11:44.492 match_dhp_node:112 Bad syntax in 
DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
thread(0001) 2016-11-16 03:11:44.492 session_create_unsafe:428 Starting
thread(0001) 2016-11-16 03:11:44.492 session_connect:497 Starting.
thread(0001) 2016-11-16 03:11:44.492 session_connect:517 Done.
thread(0001) 2016-11-16 03:11:44.492 session_register:599 Starting.
thread(0001) 2016-11-16 03:11:44.492 session_register:634 sending data:
Dumping bytes:
00000 65 00 04 00 00 00 00 00 00 00
00010 00 00 00 00 00 00 00 00 00 00
00020 00 00 00 00 01 00 00 00 00
thread(0001) 2016-11-16 03:11:44.495 session_register:710 received response:
Dumping bytes:
00000 65 00 04 00 00 01 00 0b 00 00
00010 00 00 00 00 00 00 00 00 00 00
00020 00 00 00 00 01 00 00 00 00
thread(0001) 2016-11-16 03:11:44.495 session_register:732 Done.
thread(0001) 2016-11-16 03:11:44.495 session_create_unsafe:481 Done.
thread(0001) 2016-11-16 03:11:44.495 ab_tag_create:259 Done.
thread(0001) 2016-11-16 03:11:44.495 plc_tag_create:212 Returning mapped 
tag 0x4001
INFO: Got tag 0x4001
INFO: tag status 0
INFO: reading tag 0x4001
thread(0001) 2016-11-16 03:11:44.495 plc_tag_read:436 Starting.
thread(0001) 2016-11-16 03:11:44.495 eip_cip_tag_read_start:138 Starting
thread(0001) 2016-11-16 03:11:44.495 build_read_request_unconnected:510 
Starting.
thread(0001) 2016-11-16 03:11:44.495 build_read_request_unconnected:623 Done
thread(0001) 2016-11-16 03:11:44.495 eip_cip_tag_read_start:212 Done.
thread(0002) 2016-11-16 03:11:44.496 send_eip_request_unsafe:42 Starting.
thread(0002) 2016-11-16 03:11:44.496 send_eip_request_unsafe:104 Done.
thread(0002) 2016-11-16 03:11:44.497 
session_check_incoming_data_unsafe:609 got full packet of size 50
Dumping bytes:
00000 6f 00 1a 00 00 01 00 0b 00 00
00010 00 00 bf 56 f1 03 00 00 00 00
00020 00 00 00 00 00 00 00 00 00 00
00030 02 00 00 00 00 00 b2 00 0a 00
00040 d2 00 00 00 c4 00 01 00 00 00
thread(0001) 2016-11-16 03:11:44.500 check_read_status_unconnected:1233 
processing request 0
thread(0001) 2016-11-16 03:11:44.500 check_read_status_unconnected:1361 
Got 4 bytes of data
thread(0001) 2016-11-16 03:11:44.500 plc_tag_read:510 elapsed time 5ms
thread(0001) 2016-11-16 03:11:44.500 plc_tag_read:513 Done
data[0]=1 (1)
thread(0001) 2016-11-16 03:11:44.500 plc_tag_destroy:407 Starting.
thread(0001) 2016-11-16 03:11:44.500 plc_tag_destroy_mapped:361 Starting.
thread(0001) 2016-11-16 03:11:44.500 plc_tag_abort_mapped:298 Starting.
thread(0001) 2016-11-16 03:11:44.500 plc_tag_abort_mapped:315 Done.
thread(0001) 2016-11-16 03:11:44.500 ab_tag_destroy:379 Starting.
thread(0001) 2016-11-16 03:11:44.500 plc_tag_abort_mapped:298 Starting.
thread(0001) 2016-11-16 03:11:44.500 plc_tag_abort_mapped:315 Done.
thread(0001) 2016-11-16 03:11:44.500 session_destroy_unsafe:527 Starting.
thread(0001) 2016-11-16 03:11:44.500 session_destroy_unsafe:566 Done.
thread(0001) 2016-11-16 03:11:44.500 ab_tag_destroy:428 done
thread(0001) 2016-11-16 03:11:44.500 plc_tag_destroy_mapped:397 Done.
thread(0001) 2016-11-16 03:11:44.500 plc_tag_destroy:416 Done.
Done

kyle-github, 2016-12-16:
Hmm, timeout?  That's... not what I was expecting.

Unfortunately, the OPC app is almost certainly hitting RSLinx.   What I understand about SoftLogix is that it can use RSLinx as a sort of bridge to do all the network work.  But, RSLinx only speaks OPC to external clients.   OPC is not a challenge I want to take on right now.   It is Microsoft's COM (well, DCOM I think), but with a bunch of semi-documented APIs.

I will see if there is a way I can get access to a SoftLogix set up.  

The library is used by many to talk directly to real hardware.   We've used it for PLC/5 and Control/CompactLogix.   It does talk to Micro830 and 850 and SLC 500 and several other kinds of MicroLogix (1100 and 1400 were tested).  AB is very inconsistent with their protocol on their own hardware.   

I want to leave this bug open because the SoftLogix system should be supported.

kyle-github, 2016-12-16:
I ran across this note (for Kepware, but it might be important):

https://www.kepware.com/support/knowledge-base/default.asp?solution=/_ui/selfservice/pkb/PublicKnowledgeSolution/d?&id=50140000000RvKL

Note that they talk about basically turning off RSLinx.   The comments in the SoftLogix manual indicate that you need to configure the drivers in RSLinx correctly otherwise it will block the SoftLogix.   I wonder if that is what is happening here?

kyle-github, 2016-12-16:
BTW, if you wanted to build a Node.js library to wrap libplctag, a contribution would be very welcome :-)

And if you do so, please let me know what could be changed with the API.  My wrapping was mostly Java.   I will probably work on some C++ and Python next or maybe .Net.

QuinDennis, 2016-12-16:
Yes, most certainly!  Would be fun.  Here's an image of a demo I 
recently built for an Automation Fair in ATL.  This is mostly simulated 
data (in node), but the next step is moving the simulations to LGX, 
which I've done.  Our library project would connect this to LGX.

I-Campbell, 2018-01-11:
The last test of this showed the path as 0,3 or 1,3.
The image shows the processor in slot 2, so the path should have been 1,2 (It is indeed 1 to get to the softlogix backplane).

I do not have access to SoftLogix to retest this bug using the correct path.

> Hi Kyle,
>
>No luck, no change for 0,3 or 1,3.

kyle-github, 2018-12-17:
Has either of you had any luck with this?

kyle-github, 2019-07-02:
Closing for now.   Please reopen if you get a chance to try again.  I would like to get this working!

closed 2019-07-02 by kyle-github

kodaman2, 2019-08-16:
I use this lib in java for L73 and L84 contrologix controllers https://github.com/EPICSTools/etherip and this one in python https://github.com/dmroeder/pylogix to make utilities for the company I work for, usually I test everything on softlogix if I am not at the office. Never had any issues with neither etherip or pylogix. Now I am using the c# wrapper for libplctag https://github.com/mesta1/libplctag-csharp. I compiled libplctag for win64. And tested in my unity3d project with no issues. I never done anything special with rslinx or anything else that I can remember. Just launch softlogix, download my program, and go to town.

In my sample project I tested for a boolean tag "Start", and both true and false were read fine with no issues.

I am not entirely sure, but on @QuinDennis pic of the SL backplane there is rslinx on both slot 0 and 1, I believe this happens when you have classic and enterprise running. If I was you I would uninstall enterprise or make sure is not running that could be your problem I do remember maybe having an issue when using pylogix and I think I had both classic and enterprise installed I could be wrong but worth a try.

using the csharp wrapper in unity3d PLCTest.cs:
```csharp
using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using UnityEngine.UI;
using System.Runtime.InteropServices;
using System;
using LibplctagWrapper;
using System.Threading;

public class PLC_Test : MonoBehaviour
{
    public Text myText = null;
    
    // Start is called before the first frame update
    void Start()
    {

        myText.text = "Test";

        var tag1 = new Tag("192.168.0.26", "1, 1", CpuType.LGX, "Start",
        DataType.Int8, 1);

        var client = new Libplctag();

        // add the tag
        client.AddTag(tag1);

        // check that the tag has been added, if it returns pending we have to retry
        while (client.GetStatus(tag1) == Libplctag.PLCTAG_STATUS_PENDING)
        {
            Thread.Sleep(100);
        }

        // if the status is not ok, we have to handle the error
        if (client.GetStatus(tag1) != Libplctag.PLCTAG_STATUS_OK)
        {
            Debug.LogFormat($"Error setting up tag internal state. Error { client.DecodeError(client.GetStatus(tag1))}\n");
            return;
        }

        // Log to console
        Debug.Log(tag1.ToString());

        Boolean a_bool = client.ReadBool(tag1, 0, 5000);

        Debug.Log(a_bool.ToString());
        myText.text = a_bool.ToString();

        // Execute the read
        var result = client.ReadTag(tag1, 5000);

        // Check the read operation result
        if (result != Libplctag.PLCTAG_STATUS_OK)
        {
            Debug.LogFormat($"ERROR: Unable to read the data! Got error code {result}: {client.DecodeError(result)}\n");
            return;
        }

    }

    // Update is called once per frame
    void Update()
    {
        
    }
}

```

[image]

[image]

[image]

[image]

[image]

kyle-github, 2019-08-17:
Thanks, @kodaman2!   

Hopefully this will help others solve this problem.