libplctag/libplctag #42: plc_tag_destroy never returns

xinthose, 2016-12-16:
Hello.  My application uses multi-threading with C++ as the wrapper in Linux.  Occasionally, not all of the time, when a read / write to a tag fails (because I unplugged the Ethernet cable) I go into error recovery mode.  When I can ping the PLC again, I unlock then destroy each tag before recreating them.  I can unlock the tag fine, but destroying it sometimes never returns.  

`plc_tag_unlock(tag);`
`int status = plc_tag_destroy(tag);`

Is it correct to say that you should destroy all tags from a PLC if you lose connection with it, regain it, and want to recreate your tags?

kyle-github, 2016-12-16:
Yes.  You should close down all open tags.  The state of the whole connection chain is questionable and the PLC usually (ever?) cannot recover without a full reconnection from scratch.

What version is this?  1.0 or 1.5?

So the test case is to open some tags, run read/write for a while, unplug the cable and watch it hang?

Can you provide some of the tag attribute strings?   Particularly which CPU and path.

Depending on where this is happening, it could be a TCP-level issue.  If you have outstanding data and you try to close the connection, I can see that it might hang.   I will need to make sure that the sockets are not being opened with data flushing enabled.  I think that is not the default, but I will make sure.

xinthose, 2016-12-16:
OK.  It's version 1.5.  
Yes, I would let it hang and reconnect when it can ping it again (I'm attaching my ping class if it helps you at all).
I think your assessment sounds right.  The Ethernet communication could be causing it to hang.  
I have to account for these situations and be flexible in my code, per my engineer/boss's instructions.  
A long hang time is fine with me (like how TuxEip was), so long as the function does not block indefinitely.  
[ping.zip](https://github.com/kyle-github/libplctag/files/657977/ping.zip)

kyle-github, 2016-12-17:
How long do you see this hang?   The TCP problem I was thinking of is actually somewhat difficult to recreate on demand.

kyle-github, 2016-12-18:
I still cannot recreate it.  It looks like it really has to happen at just the right moment.  However, I did add code to set the SO_LINGER option.  I set it to 2 seconds.  So, if there is outstanding data on the TCP connection when close() is called, it should wait no more than the timeout of the upper stack plus 2 seconds.  

This will be in version 1.5.1 which should be out in the next week or so.

xinthose, 2016-12-18:
OK.  Thank you.  My scenario is that a Linux PC creates tags with three different ML-1400's.  I have a function that reads one tag once a minute with each PLC to keep the connection alive.  I unplug one, try to write some tags to the PLC, I get a time-out error, then try to destroy the tags and recreate them.  

I control sports equipment with a tablet, so safety is paramount (lawyers, etc.).  This video gives the general idea: https://www.youtube.com/watch?v=AUeinXX5P1o

kyle-github, 2016-12-18:
Would it be better to have a zero wait?  I can set that up too. 

The more I think about it, the more it seems like that might be the best setting.  If you are closing the socket, you are done.  Any data that has not been sent probably should not be (or the state of the PLC might not be what you think it is when you restart).  

Is the library running on the tablet itself or on some server somewhere?

kyle-github, 2016-12-18:
This should be closed in 1.5.1.  Please try it out!

closed 2016-12-18 by kyle-github

xinthose, 2016-12-19:
OK.  Will do.  Thank you.  It will have to wait until the next project because this one shipped.  I host the HTML5/JavaScript/CSS3 app on Apache from the Linux PC running the libplctag code.  I use REST as the intermediary.  
I wouldn't mind a zero wait so long as it works properly.

kyle-github, 2016-12-19:
Note that TCP will hang for several second on open if the host in question is not around.  I will look into making that shorter too.

xinthose, 2016-12-20:
OK.  Thank you.