libplctag/libplctag #43: extra param/size issue with Python wrapper on Windows/RPi

kyle-github, 2016-12-20:
Hi everyone, Im getting a plc tag size error.
I am expecting to read from tag name "scada" with 100 integer values.
I am currently testing the python module.

This is the command i passed:
 python tag_rw.py -t sint32 -p "protocol=ab_eip&gateway=192.168.1.23&path=1,0&cpu=LGX&elem_size=4&elem_count=100&debug=1&name=scada"

I have using -t uint32 , -t uint8, -t sint32 with element size of 4, but the library always responds with the below :

Traceback (most recent call last):
  File "tag_rw.py", line 162, in <module>
    main()
  File "tag_rw.py", line 95, in main
    tag_size = libplctag.plc_tag_get_size(tag)
  File "/home/xeon/libplctag/contrib/Python/plctag/libplctag.py", line 266, in plc_tag_get_size
    return plcTagGetSize(tag)
TypeError: this function takes at least 2 arguments (1 given)

kyle-github, 2016-12-20:
I did try the same implementation though on windows ( python using anaconda environment) + on a raspberry pi jessie. For those systems the same python module still had errors.

kyle-github, 2016-12-20:
(above from the email thread on the forum)

kyle-github, 2016-12-23:
It works on RPi 2B, with the latest 2016-11-25 Raspian image.

kyle-github, 2017-03-20:
I have to rewrite the wrapper anyway to use the 2.0 API.  I will attempt to fix this at that time.  As noted, it works on the RPi for me.

kyle-github, 2018-12-17:
I will be testing on Windows soon.

kyle-github, 2018-12-17:
Hmm, Windows 10 with Python 3.7.1 does not run.  I got it to compile correctly, but I get a similar error to the one above.

`
return plcTagCreate(attributeString, timeout)
ctypes.ArgumentError: argument 1 <class TypeError>: wrong type
`

Does anyone have experience with ctypes and Python on Windows 10 x64?

stej2084, 2021-09-15:
Seem to be getting the same issue that you noted with plcTagCreate returning argument 1:<class 'TypeError'>: wrong type.

This is on windows 10 and python 3.8.10.

Command passed:
python .\tag_rw.py -t uint8 -p 'protocol=ab_eip&gateway=192.168.45.92&path=1,0&cpu=LGX&elem_size=1&name=MySINT'

Tried all the -t variables and always the same traceback:

Traceback (most recent call last):
  File ".\tag_rw.py", line 170, in <module>
    main()
  File ".\tag_rw.py", line 76, in main
    tag = libplctag.plc_tag_create(plc_path, CREATE_TIMEOUT)
  File "C:\Users\jonathons\Documents\comms_py\plctag\libplctag.py", line 275, in plc_tag_create
    return plcTagCreate(attributeString, timeout)
ctypes.ArgumentError: argument 1: <class 'TypeError'>: wrong type

commit 429b4671 (2020-04-26): fixes #43