libplctag/libplctag #44: plc_tag_get_uint16 big endian

kieranHay, 2016-12-20:
I could be wrong but it looks like the code in plc_tag_get_uint16 for big endian is wrong. Currently its: 
`res = ((uint16_t)(t->data[offset+2]) << 8) +
		      ((uint16_t)(t->data[offset+3]));`

I believe it should be this:
`res = (int16_t)(((uint16_t)(data[offset]) << 8) +
			((uint16_t)(data[offset + 1])));`

kyle-github, 2016-12-20:
I think you are correct.   I've never found a good test system that was big-endian.  I will check through these carefully.  Thanks for this!

As part of the work to support Modbus, I will be refactoring all the endian-specific code.   There are Modbus systems that do not conform to either typical "big" or "little" endian models.

I will try to fix this in the next release of libplctag.  Maybe over the holidays I can figure out how to set up a big-endian system on Qemu to test this out.

kyle-github, 2017-01-20:
Fixing this in 1.5.3.

kyle-github, 2017-01-25:
It looks like there was a cut-and-paste error in several of the big-endian blocks.

I may just remove the big-endian blocks since AB PLCs use only little-endian data.  All the code here will be rewritten anyway when I support ModBus since there are many ModBus devices with wildly varying byte and word order.

kyle-github, 2017-03-18:
I am closing this, not necessarily because it is fixed but because the fix is going to be a lot more invasive and this is not actually a bug.   It is not a bug for the wrong reasons, so this was a very good report!

The core of why this is not a bug at this time is that the big/little endian change cannot happen in the current library because AB PLCs are the only type supported and they are all little endian.   So, the big endian code cannot be called.

As I move to support other kinds of PLCs, all this code is going to be rewritten in a different way.

Thanks again for the info!

commit 22cee34c (2020-04-26): fixes #44
closed 2017-03-18 by kyle-github