libplctag/libplctag #46: Add accessory/helper function to get tag list from Logix PLCs

thatcadguy, 2017-01-25:
Kyle, been checking out pylogix lately (https://github.com/dmroeder/pylogix), a small Python library for dealing with Logix PLCs. That library includes a function aptly named GetTagList(). In future this would be very helpful to add to the library, I think. When I have some more free time I might be able to port the Python code from that project to C and submit a PR. It's Apache licensed so we would just give attribution and link or something.

kyle-github, 2017-01-25:
Interesting!  I had looked at cpppo too (also Python).

I will queue this for a feature drop.   I will be out of the country for a couple of weeks on business, so I won't be able to look at this until mid to late February.

This might be a good test of some new internal architecture changes I want to try.

kyle-github, 2017-11-17:
This is in progress.

Thanks to a new user, I got more documentation on how this is supposed to work.  AB did publish some docs on this publicly.

afiya-g, 2019-05-06:
Hello Kyle,

Does current version supports now browsing of AB controller tags?
I'm using libplctag lib for my client project, and the requirement is to browse the AB controller tags.
If the current version is supporting, can you please let know how to do it.

Regards,
Syed.

vlcvboyer, 2019-05-25:
Hi Kyle,
Do you need some help on this feature Implementation ?

kyle-github, 2019-05-25:
I have some old code that does not work with the current state of the library.    

I should really dust that off and port it to the current version.  This feature is requested often!

If you want to take a swing at it, I would love the help!

kyle-github, 2019-05-25:
The way I prototyped this was to have a special tag name, "@tags".

There is some very annoying logic you need to implement with the internal tag IDs.

vlcvboyer, 2019-05-25:
can please indicate which branch contains this preliminary code?
Thanks

kyle-github, 2019-05-27:
This has been reopened as ticket #97.

closed 2019-05-27 by kyle-github

kyle-github, 2019-05-27:
Hi @vlcvboyer, there is an almost working version in the branch "tag_list" on GitHub.   I ran out of time today to get more done.

kyle-github, 2019-06-02:
This has been implemented in version 2.0.17.

vlcvboyer, 2019-06-03:
Thks !
I'll test it in a short time.

commit 2a30aa23 (2020-04-26): fixes #46