libplctag/libplctag #47: Add accessory/helper function to discover Logix PLCs

thatcadguy, 2017-01-25:
Similar to #9, pylogix also has a Discover() function that discovers all Logix PLCs in the computer's network. Would also be an excellent addition to libplctag IMO.

kyle-github, 2017-01-25:
Also an interesting idea.   I think this can be done via a UDP broadcast IIRC.   It seems like it would be interesting to combine this with the previous request.

The more I think about this, I am not sure I want it as part of the library, but it should be in related tools.  It is difficult to make functionality like this generic for multiple kinds of PLCs.

thatcadguy, 2017-08-12:
I would say it would be generic enough if you tried to find A-B PLCs and then pinged (Modbus TCP function code 3) every IP in the current subnet with both standard Modbus TCP ports (502, 1502). As I recall, you wanted to add Modbus, right?

kyle-github, 2017-11-17:
Still have not done this.   I need the whole new structure of the library working to handle UDP as easily.   It might be something I can do with each type of protocol.  I have a structure now that will let me separate that out for the types of protocols that can support it.   There might need to be some user input to get this information.

Folcogh, 2020-04-07:
Hi,

Is there some plan to do that ?
This library cannot be used to work with real PLC programs if it's not possible to get the tag list, or at least an identifier for an already existing tag...

Thanks in advance for your answer!

kyle-github, 2020-04-08:
Hi Folco,

You can get the tag list.  See the list_tags.c example.

As to using it with real PLC programs, well, the library has been in use in
production since 2011 and publicly available since 2012 :-)

I do plan to have PLC discovery at some point, but it is not the highest
priority.   What the users have asked for was S7 support.   The current
priority is to get a good test harness so that I can add regression testing
to the CI pipeline in Azure.  Then I can start changing the library more
quickly.

Best,
Kyle

xinthose, 2020-04-08:
@Folcogh In my case, my program reads/writes to a PLC whose program we wrote, so we know tag names ahead of time.  I have used this library in production for years now, no issues.

Folcogh, 2020-07-11:
Thank you very much for the feedback. I finally understand the API. Really good lib, I could connect and do some things I needed to do.
But I definitely need to be able to discover all tags to fulfill my needs.

kyle-github, 2020-07-11:
There is an example of how to discover tags in an Allen-Bradley CIP-based PLC (Control Logix, Compact Logix).   The example program is `list_tags.c` in the examples.

commit 607baa1c (2020-04-26): fixes #47