libplctag/libplctag #48: Compiling Error

dpernia, 2017-03-19:
Hello Guys! Thanks for this great library.
I have downloaded it but when I try to compile the simple.c program, it tells me that there are undefined references to some tags:
I have already checked the routes to the #include <libplctag.h> and <utils.h>, and i placed them in the /usr/include folder but I can't find the cause of the errors, I have placed them on different folder using "" in case of a different folder or < > in case that i place them in /usr/include.
This is the message the compiler sends:

**It's in spanish:**
simplep.c:(.text+0x16): referencia a `plc_tag_create' sin definir
simplep.c:(.text+0x53): referencia a `sleep_ms' sin definir
simplep.c:(.text+0x5f): referencia a `plc_tag_status' sin definir
simplep.c:(.text+0x70): referencia a `plc_tag_status' sin definir
simplep.c:(.text+0x80): referencia a `plc_tag_status' sin definir
simplep.c:(.text+0x87): referencia a `plc_tag_decode_error' sin definir
simplep.c:(.text+0xbe): referencia a `plc_tag_read' sin definir
simplep.c:(.text+0xd1): referencia a `plc_tag_decode_error' sin definir
simplep.c:(.text+0x11b): referencia a `plc_tag_get_int32' sin definir
simplep.c:(.text+0x164): referencia a `plc_tag_get_int32' sin definir
simplep.c:(.text+0x1a5): referencia a `plc_tag_set_int32' sin definir
simplep.c:(.text+0x1c0): referencia a `plc_tag_write' sin definir
simplep.c:(.text+0x1d3): referencia a `plc_tag_decode_error' sin definir
simplep.c:(.text+0x20d): referencia a `plc_tag_read' sin definir
simplep.c:(.text+0x220): referencia a `plc_tag_decode_error' sin definir
simplep.c:(.text+0x267): referencia a `plc_tag_get_int32' sin definir
simplep.c:(.text+0x29b): referencia a `plc_tag_destroy' sin definir

Can you please help me to find and correct the problem?
Thanks in advance!

kyle-github, 2017-03-20:
Hi, it is not possible to help you without knowing a few more things.   Can you please tell me:

- What OS (and whether it is 32 or 64-bit)?
- What compiler (and version)
- how you are building your code?  The file simplep.c is not one of the examples, so you must be doing some sort of build process on your own.
- is the code in C or C++?
- what version of the library are you using?

Thanks!

Also, consider joining the forum/mailing list here:

https://groups.google.com/forum/#!forum/libplctag

There are other users who may be able to help you in Spanish.

Thanks for using the library!

dpernia, 2017-03-21:
HI Kile: Thank you for your prompt reply

This is the information:
- O.S.: Ubuntu 16.04 LTS
- Compiler: gcc version 5.4.0
- The code is in C. The file simplep.c is the same as simple.c (I just changed the name that day to preserve the original simple.c, _however i am using now the original simple.c_)
- I am using the latest version of the library (2015-02-21 / I downloaded it from this site last week)

I am compiling with this command line:   _gcc simple.cpp -o simple_
I have also tried with these command lines:  
- _gcc simple.cpp -lm -o simple_
- _gcc -o simple simple.c -fno-strict-aliasing_
I have also tried with g++: 
- _g++ simple.cpp -lm -o simple_
and a lot of other options that i've found on different forums to compile C and C++ programs

This is the code I am using:

`
#include <stdio.h>
#include "../lib/libplctag.h"
#include "utils.h"

#define TAG_PATH "protocol=ab_eip&gateway=167.175.150.26&path=1,0&cpu=LGX&elem_size=4&elem_count=10&name=PZ_PIRITAL_REAL[0]&debug=1"
#define ELEM_COUNT 10
#define ELEM_SIZE 4
#define DATA_TIMEOUT 5000

int main()
{
    plc_tag tag = PLC_TAG_NULL;
    int rc;
    int i;

    /* create the tag */
    tag = plc_tag_create(TAG_PATH);

    /* everything OK? */
    if(!tag) {
        fprintf(stderr,"ERROR: Could not create tag!\n");

        return 0;
    }

    /* let the connect succeed we hope */
    while(plc_tag_status(tag) == PLCTAG_STATUS_PENDING) {
        sleep_ms(100);
    }

    if(plc_tag_status(tag) != PLCTAG_STATUS_OK) {
        fprintf(stderr,"Error setting up tag internal state. Error %s\n", plc_tag_decode_error(plc_tag_status(tag)));
        return 0;
    }

    /* get the data */
    rc = plc_tag_read(tag, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d: %s\n",rc, plc_tag_decode_error(rc));
        return 0;
    }

    /* print out the data */
    for(i=0; i < ELEM_COUNT; i++) {
        fprintf(stderr,"data[%d]=%d\n",i,plc_tag_get_int32(tag,(i*ELEM_SIZE)));
    }

    /* now test a write */
    for(i=0; i < ELEM_COUNT; i++) {
        int32_t val = plc_tag_get_int32(tag,(i*ELEM_SIZE));

        val = val+1;

        fprintf(stderr,"Setting element %d to %d\n",i,val);

        plc_tag_set_int32(tag,(i*ELEM_SIZE),val);
    }

    rc = plc_tag_write(tag, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d: %s\n",rc, plc_tag_decode_error(rc));
        return 0;
    }

    /* get the data again*/
    rc = plc_tag_read(tag, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d: %s\n",rc, plc_tag_decode_error(rc));
        return 0;
    }

    /* print out the data */
    for(i=0; i < ELEM_COUNT; i++) {
        fprintf(stderr,"data[%d]=%d\n",i,plc_tag_get_int32(tag,(i*ELEM_SIZE)));
    }

    /* we are done */
    plc_tag_destroy(tag);

    return 0;
}`

and the answer is still the same:

> simple.cpp:(.text+0x16): referencia a `plc_tag_create' sin definir
> simple.cpp:(.text+0x55): referencia a `plc_tag_status' sin definir
> simple.cpp:(.text+0x69): referencia a `sleep_ms' sin definir
> simple.cpp:(.text+0x77): referencia a `plc_tag_status' sin definir
> simple.cpp:(.text+0x8c): referencia a `plc_tag_status' sin definir
> simple.cpp:(.text+0x93): referencia a `plc_tag_decode_error' sin definir
> simple.cpp:(.text+0xca): referencia a `plc_tag_read' sin definir
> simple.cpp:(.text+0xdd): referencia a `plc_tag_decode_error' sin definir
> simple.cpp:(.text+0x12b): referencia a `plc_tag_get_int32' sin definir
> simple.cpp:(.text+0x174): referencia a `plc_tag_get_int32' sin definir
> simple.cpp:(.text+0x1b5): referencia a `plc_tag_set_int32' sin definir
> simple.cpp:(.text+0x1cc): referencia a `plc_tag_write' sin definir
> simple.cpp:(.text+0x1df): referencia a `plc_tag_decode_error' sin definir
> simple.cpp:(.text+0x219): referencia a `plc_tag_read' sin definir
> simple.cpp:(.text+0x22c): referencia a `plc_tag_decode_error' sin definir
> simple.cpp:(.text+0x277): referencia a `plc_tag_get_int32' sin definir
> simple.cpp:(.text+0x2a7): referencia a `plc_tag_destroy' sin definir

I hope you can help me to find the solution. :-)

kyle-github, 2017-03-26:
The response on the forum should have covered this problem.  Are you still having problems compiling and linking your code?  Please let me know...

dpernia, 2017-04-07:
Hello Kyle: Thank you... Yes, I'm still having the problem when I try to compile the program. :-(

kyle-github, 2017-04-10:
Please include the exact command line you use to compile.  It looks like you are not including the library in the command line.  Thanks.

kyle-github, 2017-04-15:
I see now that your command line does not include the library.   You need to include both libpthread and libplctag:

gcc -o simple simple.c utils_posix.c -L . -lplctag -lpthread -I .. -I .

You may need to add -I options to make sure that you can import the libplctag.h include header for where you have placed it.  The above compiles simple.c if you run it in the examples directory and the library libplctag.so is either in the same directory or it is known to ld.so.

I hope this helps!

kyle-github, 2017-05-28:
Have you been able to make any progress on this?  Please let me know.   Thanks!

kyle-github, 2017-09-03:
I think you might be gone.  Closing.

commit c62b184f (2020-04-26): fixes #48
closed 2017-09-03 by kyle-github