libplctag/libplctag #49: Windows build comments

historicbruno, 2017-04-14:
First, thanks for your work on this library!

I have a few answers to questions posed in your BUILD.md instructions.
* It seems **all** versions of Visual Studio 2015 and newer don't include C++ compilers unless you choose the option, it's not only an Express edition or download limitation. I had to modify the installation, since I originally missed that.
* cmake-gui definitely works and I prefer it to the command line on Windows as well :) You only need to do the following after installing CMake:
    * Open cmake-gui
    * For "Where is the source code" choose the project root directory, i.e. where CMakeLists.txt is located
    * For "Where to build the binaries" choose a build folder inside of that directory, or anywhere else
    * Press **Configure**
    * Choose your desired version of Visual Studio from the list, and press **Finish**
    * You can set CMAKE_INSTALL_PREFIX to a different folder, recommended because the default isn't writable (only needed if you build the Install project in VS, which is nice for organization)
    * Press **Generate**
    * There is now an **Open Project** button, so you can open the project automatically in the correct version of Visual Studio
* It was **not** necessary to run the developer command prompt, in my testing. This is with a default installation of the latest CMake, and VS 2015 Community Edition on Windows 10. I think this simplifies the instructions a bit.

kyle-github, 2017-04-15:
Thanks for the updates.  I will try to put these updates into the next release.   I do not have Windows 10, so I did not know that some of these things get so much cleaner!

historicbruno, 2017-04-17:
Oh, it shouldn't be specific to Windows 10. My dev machine is now Windows 10, but I used CMake and its gui on Windows 7 and XP many times over the years for various open source projects :)

kyle-github, 2017-05-28:
Oops, I forgot to include these in 1.5.5.   Scheduled for 1.5.6.

kyle-github, 2017-09-03:
I just pushed these into the head of the 2.0 branch.  Soon to be a real release.  I hope :-/

Thanks again for the hints!

commit 9ff4d1de (2020-04-26): fixes #49
closed 2017-09-03 by kyle-github