libplctag/libplctag #50: PLCTAG_ERR_TIMEOUT

xinthose, 2017-05-05:
Hello.  I am seeing an issue where I occasionally get the error status of "PLCTAG_ERR_TIMEOUT" when writing a single value to a PLC.  I reduced the frequency I write to the PLC (every 5 sec instead of 3 sec) and increased the timeout to 10,000 ms instead of 5,000 ms, but I still get this occasional error.  I'm worried this might happen in the middle of a customer transaction and end it early.  I see the error after hours of successful writes (this is a PLC heartbeat so it knows it's connected to the PC) and reads of system_ready from the PLC.  Should I increase the timeout?  I am attaching some log files for your reference (search for text "[error]"). 
[Logs.zip](https://github.com/kyle-github/libplctag/files/979339/Logs.zip)

Library Version: 1.5.3
OS: Ubuntu 16.04 64-bit
PLC: MircroLogix 1400

Thank you.

kyle-github, 2017-05-05:
Thanks for the bug report and the log files!

Is there any chance of getting this rerun with the library debugging turned to level 3?   Append '&debug=3' to the string used to create the tag.  It does generate a lot of "noise" but then I can see if there is anything odd happening.

Here is where I see the errors in the first log file:

17-05-04 22:40:45.019273 [error] src/plctag.cpp(read_tag:316) >> tag[1] >> plc_tag_read >> status = PLCTAG_ERR_TIMEOUT
17-05-04 22:40:45.022085 [error] src/events.cpp(micrologix_read_system_ready:2717) >> _plctag.read_tag >> error = -2
17-05-04 22:40:45.142358 [error] src/main.cpp(event_dispatcher:1592) >> MICROLOGIX PLC ERROR
17-05-04 22:40:45.142951 [info] src/events_error.cpp(micrologix_error:380) >> started
17-05-04 22:40:45.143310 [info] src/main.cpp(reset_controller_threads:370) >> controller threads reset
17-05-04 22:40:47.247107 [error] src/plctag.cpp(write_tag:461) >> tag[4] >> plc_tag_write >> status = PLCTAG_ERR_TIMEOUT
17-05-04 22:40:47.247655 [error] src/events.cpp(micrologix_write_heart_beat:2747) >> _plctag.write_tag >> error = -3

Note that these errors are on different tags (tag[1] and tag[4]).  Are they against the same PLC?   The fact that they are against different tags suggests that the timeout may be real.  Are there any other operations you are doing against the PLC at the same time that worked during that interval?

Also, if the timeouts are set for 5 or 10 seconds, then it looks like these two operations are in flight at the same time.

The second log shows the same pattern, but the read and write operations are reversed:

17-05-05 08:18:50.533507 [info] src/events.cpp(micrologix_read_system_ready:2684) >> system_ready = true
17-05-05 08:19:05.510315 [error] src/plctag.cpp(write_tag:461) >> tag[4] >> plc_tag_write >> status = PLCTAG_ERR_TIMEOUT
17-05-05 08:19:05.510870 [error] src/events.cpp(micrologix_write_heart_beat:2747) >> _plctag.write_tag >> error = -3
17-05-05 08:19:05.747810 [error] src/main.cpp(event_dispatcher:1592) >> MICROLOGIX PLC ERROR
17-05-05 08:19:05.748331 [info] src/events_error.cpp(micrologix_error:380) >> started
17-05-05 08:19:05.748776 [info] src/main.cpp(reset_controller_threads:370) >> controller threads reset
17-05-05 08:19:10.538495 [error] src/plctag.cpp(read_tag:316) >> tag[1] >> plc_tag_read >> status = PLCTAG_ERR_TIMEOUT
17-05-05 08:19:10.539053 [error] src/events.cpp(micrologix_read_system_ready:2717) >> _plctag.read_tag >> error = -2
17-05-05 08:19:10.971512 [warning] src/events_other.cpp(error_handling:194) >> SYSTEM FAULT
17-05-05 08:19:10.980342 [info] src/events_other.cpp(error_handling:226) >> error handled

Again, the use of apparently different tags suggests that the timeout is real.  Here, though, the difference in time is 5 seconds almost exactly.  What was the timeout here?

As another test, would it be possible to start a ping test (using the command line ping command) that runs and dumps into a log showing the ping loss and latency during the run of the C++ program?  If there is a real network problem, we might see that in the ping responses.

In order to more easily recreate this, I would suggest dropping the timeout much lower.  Perhaps 250ms.

Is the MicroLogix doing any kind of remote IO?  Is there possibly anything else that is happening on that network?  If you have any way of monitoring the network traffic to and from the MicroLogix, that may help to determine if there is something else happening at those times.

Does your program try to reconnect immediately after the fault?

xinthose, 2017-05-05:
Thanks for the quick response.  I suppose it could be an issue related to the network and not the library.  I don't know if I can enable debugging because it's currently in use by a customer.  

- tag[1] is a float of length 1 and is read every 10 sec with a timeout of 10s
- tag[4] is a float of length 1 and is written every 5 sec with a timeout of 10s

Both tags go to the same PLC.  I did not think about staggering these read / writes so that they do not overlap as much.  

I will look into logging ping times.  I may make that standard now.  

I do reconnect after pinging the PLC IP address.  It does recover fine and has no errors again for hours afterward.  I'm not too sure about network traffic, its the customer's site.  The MicroLogix does not use any remote I/O that I know of.

Felipeasg, 2017-05-25:
Hi there, I usually also receive the -11 error (PLCTAG_ERR_TIMEOUT). 
I think that is a network problem, because sometimes you have others HMI requesting a log of tags read/write. 

Another point is that the "unconnected" CIP messages (i believe that this type of message is used in libplctag) is used in the connection establishment process and for infrequent, low-priority message.

I saw that you suggest a reconection, the reconection procedure is destroy the tag and recreate the tag again? 

Thanks in advance.

Felipe

kyle-github, 2017-05-27:
Yes, when you know that the problem is transient, then destroying the tag and recreating will help.  Note that the underlying TCP connection is automatically shared between tags in the same PLC, so it is important to close all the tags to that PLC, then reopen them.  I am going to see if I can change the internal logic to make this so that it does not require that much coordination.

I also want to be able to reconnect automatically.  That is for the version 2.0 code though.

Felipeasg, 2017-05-28:
Ok. So Is not recommended recreate just the tag that receive the error? 
Because I create the tags in different moments in the program using the **share_session=1**.
Is it recommended create all tags Together? 
Can creating them at separeted times and in different c++ classes cause problems?

**Obs:**
I conclude that is a transient because I have several tests using [cpppo](https://github.com/pjkundert/cpppo) where is possible emulate controllogix tags and simulate some readings and writes using libplctag. Using o cpppo I don't get the -11 error.

Thanks for  the quick response.

Felipe

kyle-github, 2017-05-28:
Yes, if the session (CIP-speak for the TCP connection) is shared, then closing and reopening one tag may not be sufficient.   The logic around fully closing a TCP session when it has gone bad is not very smart right now.   It is something I want to tweak in the near future.  You actually do not need to put in share_session=1 as that is the default.

If you look at the Java wrapper, you'll see that there is a thread that does tag reconnection.   The thread waits five seconds before trying again to give all the tags a chance to be closed.   This is not the best way to do things necessarily for all apps, but was good enough for our use.

Felipeasg, 2017-05-30:
Really is a nice example. Since **java** sintax is similar to **c** I could understand the logic.
But I'm fail to use the wrapper, due my inexperience with java.
Do you have some example? and how to compile?

I am tried to do:

```
$ javac -classpath Tag.java Main.java
``

But I can't use the Tag class in my Main.java.

Felipe

Felipeasg, 2017-05-30:
I can use the wrapper with eclipse. I don’t now if is the way is more correct.

```java
package testlibplctag;

import testlibplctag.Tag;

import com.sun.jna.Native;

public class Main {
    private static Tag tag1;
    private static Tag tag2;

	public static void main(String[] args) {                                                                  
        
    	/**
    	 * Using enip_server (cpppo) to emulate the tag
    	 * 
    	 * $ enip_server FLOAT=REAL FLOAT2=REAL
    	 */
    	
    	tag1 = Tag.create("protocol=ab_eip&gateway=localhost&path=1,0&cpu=compactlogix&elem_size=4&element_count=1&name=FLOAT");
    	tag2 = Tag.create("protocol=ab_eip&gateway=localhost&path=1,0&cpu=compactlogix&elem_size=4&element_count=1&name=FLOAT2");
    	
    	int rc1 = tag1.read(1000);
    	int rc2 = tag2.read(1000);
    	
    	
    	System.out.println("Result read tag1: " + rc1);
    	System.out.println("Result read tag2: " + rc2);

    	
    	
    	
//      Tag tag = (Tag) Native.loadLibrary("plctag", Tag.class);                                              
                                                                                                              
        System.out.println("Hello, World");                                                                   
    }         
}
```

kyle-github, 2017-06-01:
The library will search for a matching session internally when you recreate the tag.   That is why it is important to destroy all the tags that share a session.   That is the logic that I want to change.   Timeouts are not that obvious though.   It could be very saturated link, not a bad link.   That is partly why the logic for this is outside the library.

Felipeasg, 2017-06-04:
Ok.

I tryed replicate the java wrapper using c++ with retryTags ([retryTag c++](https://github.com/Felipeasg/retryTags), authough the SIGPIPE signal was breaking may application, I was able to solve this handling this signal.

I tested the program ([retryTag c++](https://github.com/Felipeasg/retryTags) stopping the server while the client try to read the tags and everything works well.

I believe that the retry is the solution, but something crazy is happening in my real application with more threads, the program simply hangs when I try to execute some libplctag API function when the server stops. I try to manage SIGPIPE in my threads but did not work yet.

Anyway thanks for the retryTags tip. 

OBS: I saw you release the 2.0.1 version I will test my application with it.

kyle-github, 2017-06-04:
Adam, Felipe,

During my testing for the MicroLogix string bug today I noticed something that may be at the root of many of the timeout problems.

I saw that the VPN I was using to access the MicroLogix (on a different continent than I am) had some strange behavior.  I would see no response packets for very long periods of time (>600ms at time) followed by many packets all at once.   On average, the response time was within the allowed resend windows.   But, these packets did not behave as the average.   This very bursty behavior is not something I've seen before on an industrial network.  But, if there are many HMI hits and other traffic, then it is possible that nothing gets through for hundreds of milliseconds.

I am going to rethink the base internal retry delays.   I think I will need to make it much, much longer.   Currently it defaults to 50ms, but in the cases I saw, 800ms would be a lot better!   This is one of those cases where the application has the information to make a decision about whether to wait or not and the library can only guess.

Felipe, thanks for doing that work with C++!  Do you have any debug logs you can share?   Heavy threading is usually the place where things break :-/

Felipeasg, 2017-06-05:
Well I debug a little more using &debug=4 in the tag string attribute, I could understand what is happening..

In the [retry thread](https://github.com/Felipeasg/retryTags/blob/master/tag.cpp#L297) I am using a mutex to lock the acess to tag and retryTags vector.
In the [read() function](https://github.com/Felipeasg/retryTags/blob/master/tag.cpp#L112) (method similar to your [read()](https://github.com/kyle-github/libplctag/blob/master/wrappers/java/Tag.java#L164) in the java wrapper) also have a mutex to block the access to the opaque pointer to tag.

When I try to read forcing the tag server stop, the **plc_tag_read()** go into an infinite loop (and don't  return a PLCTAG_ERROR), as I used a mutex to ensure that the tag will not be used in the retryThread, all threads stops, the thread that call the method read() and the retryTag thread.

Bellow is the log when I try to read the tag (with the [read() ](https://github.com/Felipeasg/retryTags/blob/master/tag.cpp#L112) method that call plc_tag_read()).
When the program call **plc_tag_read()** it dump the log bellow and stay there in a infinite loop. 

```
thread(0001) 2017-05-04 19:11:07.943 INFO plc_tag_read:526 Starting.
thread(0001) 2017-05-04 19:11:07.943 DETAIL plc_tag_read:528 Reading tag id ptr 0x4001.
thread(0001) 2017-05-04 19:11:07.943 DETAIL plc_tag_read:529 Reading tag id 16385
thread(0001) 2017-05-04 19:11:07.943 DETAIL map_id_to_tag:1526 Starting
thread(0001) 2017-05-04 19:11:07.943 DETAIL map_id_to_tag:1543 Correct mapping for id 16385 found with tag 0x7fffec001120
thread(0001) 2017-05-04 19:11:07.943 INFO eip_cip_tag_read_start:174 Starting
thread(0001) 2017-05-04 19:11:07.943 INFO build_read_request_unconnected:532 Starting.
thread(0001) 2017-05-04 19:11:07.943 INFO request_create:51 Allocating mem.
thread(0001) 2017-05-04 19:11:07.943 INFO request_create:62 Finish.
thread(0001) 2017-05-04 19:11:07.943 DETAIL session_add_request:1013 Starting. sess=0x7fffec001590, req=0x7fffec002e90
``` 

**obs**: I am using the libplctag-1.5.4 in this program.

Thanks in advance.

Felipeasg, 2017-06-05:
News...

I am using now the libplctag-2.0.1 and when I create the tag in my real application the following error is reported, a lot of times:

```
thread(0002) 2017-05-04 21:58:42.779 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.777 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.778 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.779 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.780 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.781 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.782 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.783 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.784 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.786 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.787 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.788 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.789 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.790 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.791 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.792 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.793 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.795 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.796 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.797 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.798 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.799 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.800 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.801 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.802 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.803 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.804 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001590
thread(0002) 2017-05-04 21:58:42.806 DETAIL process_session_tasks_unsafe:1037 Checking for things to do with session 0x7fcc88001
```

It seems tat when the libplctag is used in my application the thread is not created in the better way...

kyle-github, 2017-06-05:
Thanks for the pointers to the code.  I am trying to determine what is going on here.   

One point in the code that is confusing is that it looks like you have m_oSync as a static (from tag.h):
`static CSyncObject m_oSync;`

So, when anything in the Tag class calls lock, no other thread, even if it is on another tag, can run.   However, I think your code is closer than Tag.java.  I will need to think about this.  I think you want to have m_oSync be an instance variable, not a class variable.

For the 2.0.1 version the background IO thread has a debug output that is quite... "chatty."   This means that the background thread is running.  So, the whole library is not blocked, just the front end API part.

kyle-github, 2017-06-05:
I will see what I can do to rewrite Tag.java for the new API (I realized that I forgot to update it in 2.0!).

Felipeasg, 2017-06-05:
Really is a little bit strange, but I make this to emulate something similar to [synchronized()](https://github.com/kyle-github/libplctag/blob/master/wrappers/java/Tag.java#L73) functionality. 

Rrereading your code now, I saw you didn't use synchronized to protect tags, but to protect retryTag ArrayList. 

Despite having used the libplctag-2.0.1 my the plc_tag_read() still entering in an infinite loop when I try to read and the server stoped, and my [retryTags](https://github.com/Felipeasg/retryTags)  is not working anymore.

Felipeasg, 2017-06-05:
I believe that libplctag read is lock forever in the following [part of code](https://github.com/kyle-github/libplctag/blob/1.5.1/src/protocols/ab/session.c#L1015).

I comment the line cited above and the code entering in a infinity loop with following log:

```
...
thread(0003) 2017-05-05 02:00:44.516 INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
thread(0003) 2017-05-05 02:00:44.516 DETAIL cip_encode_path:212 Found regular routing. Conn path length=1
thread(0003) 2017-05-05 02:00:44.516 INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
thread(0003) 2017-05-05 02:00:44.516 DETAIL cip_encode_path:212 Found regular routing. Conn path length=2
thread(0003) 2017-05-05 02:00:44.516 DETAIL cip_encode_path:300 ioi_size before 2
thread(0003) 2017-05-05 02:00:44.516 DETAIL session_find_or_create:244 Starting
```

In other words the program enter in infinite loop [here](https://github.com/kyle-github/libplctag/blob/1.5.1/src/protocols/ab/session.c#L244).

Felipeasg, 2017-06-05:
I enable pdebug in this [line](https://github.com/kyle-github/libplctag/blob/1.5.1/src/platform/linux/platform.c#L369) and [here](https://github.com/kyle-github/libplctag/blob/1.5.1/src/platform/linux/platform.c#L385) and confimate that the mutex lock forever.

The log when I try to read the tag with plc_tag_read after to stop the server is:

```
thread(0001) 2017-05-05 14:33:34.904 INFO plc_tag_read:526 Starting.
thread(0001) 2017-05-05 14:33:34.904 DETAIL plc_tag_read:528 Reading tag id ptr 0x4001.
thread(0001) 2017-05-05 14:33:34.904 DETAIL plc_tag_read:529 Reading tag id 16385
thread(0001) 2017-05-05 14:33:34.904 DETAIL map_id_to_tag:1526 Starting
thread(0001) 2017-05-05 14:33:34.904 DETAIL mutex_lock:369 locking mutex 0x7ff4e4000b70
thread(0001) 2017-05-05 14:33:34.904 DETAIL mutex_lock:385 Done.
thread(0001) 2017-05-05 14:33:34.904 DETAIL map_id_to_tag:1543 Correct mapping for id 16385 found with tag 0x7ff4e4001120
thread(0001) 2017-05-05 14:33:34.904 INFO eip_cip_tag_read_start:174 Starting
thread(0001) 2017-05-05 14:33:34.904 INFO build_read_request_unconnected:530 Starting.
thread(0001) 2017-05-05 14:33:34.904 DETAIL session_add_request:1013 Starting. sess=0x7ff4e4001590, req=0x7ff4e4002e90
thread(0001) 2017-05-05 14:33:34.904 DETAIL mutex_lock:369 locking mutex 0x7ff4e4000bb0

```

kyle-github, 2017-06-05:
The critical_block macro is just a wrapper allowing me to use { } to surround a block that is protected by a mutex. 

The output you see is from the background IO thread that the library creates.  This thread is how the library allows both synchronous and asynchronous behavior.   It is completely normal to see it output something every couple of milliseconds.   It just runs in a look scanning over all the open TCP channels and looking for data to send or receive.  

The latest log you showed just shows session_find_or_create starting.  Note that the code to add a request to a session (queuing the request for the IO thread) is deterministic.  It just chains to the end of a list and then adds a new element.  If the list is corrupt, that could be a problem, but then you would have a crash almost immediately.

Because they use the same mutex, the two critical_block sections prevent more than one thread from entering into the protected sections of code at a time.

4DIAC looks nice!

I will try to work on the Java code more in the near future.  I have a big project this week.

Felipeasg, 2017-06-05:
The session_find_or_create was called only when I remove the critical_block from the [**session_add_request**](https://github.com/kyle-github/libplctag/blob/1.5.1/src/protocols/ab/session.c#L1015). I remove to confirm that somehow happens when I put libplctag in the code of 4DIAC Run Time Enviromnent (FORTE) that make the libplctag thread enter in a infinite deadlock.

What happens when the plc_tag_read() is called and the tag server crashes? This is specific situation is stopping the code in the plc_tag_read().

kyle-github, 2017-06-06:
An important thing to remember is that libplctag is NOT threadsafe.

I am still trying to figure out if what you are seeing is due to multiple threads entering the mutex at once (and if so, how that results in deadlock) or some different problem.   That there is a problem is certain, but which one is it?

One thing is clear: I did not plan for integration into frameworks like 4DIAC where you do not control the threading as clearly as your own application.   

It looks like I need to bit the bullet and make the libplctag API fully threadsafe.  

While the result is the same as that of this issue (#50), I think the problems with a non-threadsafe API are a different issue.  I will file another bug.   Can you please track your tests against that new bug?

kyle-github, 2017-06-06:
To bring this back to the original bug, I saw something in my tests over a VPN that created a problem similar to the timeout errors described.

The core of the problem was that the response packets got bunched up together.   Then, the time between the last request and the first response would exceed the base retry time.  At that point a failure cascade happens.   Each retry makes the previous response invalid due to the new sequence ID.   As long as the responses never catch up to the retries, the failure will continue and eventually a timeout will occur.

Pushing retry configuration up to the application level via special settings is going to take a while.   As a work around, make the default retry time and the minimum retry interval much larger than they currently are.

xinthose, 2017-06-06:
OK, thank you.  How long of a timeout do you recommend?  I currently use 5,000 ms in every read/write call.  I will try to avoid threads reading/writing simultaneously to the same PLC.

kyle-github, 2017-06-06:
I am going to look at what I can do with internal timeouts.   The problem
is that when using connected messages, you get packet loss.   So, retries
must be a part of the library implementation.   Controlling that in a way
that does not cause a fatal cascade is tricky.   The best luck I have had
so far was with making the internal timeout 800ms or more.  But that was
just hacking in a fixed value and did not allow the user to set anything.

I think what I will do is create a new tag attribute that allows you to set
the internal retry interval.   This should be less than the timeout value.
     Having it as part of the tag attributes is annoying in that you cannot
change it at runtime as you can change the end timeout.   Hmm...   I'll
have to think about that.

For threading, the key thing is to have no more than one thread per tag.
I test with that all the time (and with multiple threads per tag, but not
with enough scenarios!).   You can have more than one tag per thread but
not the other way around.

kyle-github, 2017-06-06:
I made a prerelease for v1.5.6 with some tweaks to help with timeouts.   I added two new attributes:

num_retries - the maximum number of times the system will attempt to resend a packet.
default_retry_interval - the number of millseconds to wait before trying to send a packet again.

I would use these carefully.  In your case, I would try setting num_retries to something fairly small like 6 and the default_retry_interval fairly large as well like 600 or 900.  I would keep your timeouts at 5 seconds.

xinthose, 2017-06-07:
OK.  Thank you.  For the record, I see this error rarely in C++.  I will use this new version if I see the error crop up again.  Self recovering code is the best.

xinthose, 2017-06-13:
Hello.  I tried your code and the problem is fixed.  I was still seeing `PLCTAG_ERR_TIMEOUT ` with 1.5.7 (I'm assuming it includes the changes from 1.5.6) but after the customer shifted some Ethernet cables around, I have not seen it since.  My problem could have been caused by faulty cables and/or Ethernet switch.  
Thank you for your help on this issue.

commit 7d9bf7d7 (2020-04-26): fixes #50
closed 2017-06-13 by xinthose