libplctag/libplctag #54: Reading array of strings in Micrologix returns only first string

kyle-github, 2017-05-31:
thread(0001) 2386-04-30 23:21:00.732 INFO ab_tag_create:171 Starting.
thread(0001) 2386-04-30 23:21:00.732 INFO session_create_unsafe:524 Starting
thread(0001) 2386-04-30 23:21:00.733 INFO session_create_unsafe:578 Done
thread(0001) 2386-04-30 23:21:00.733 INFO session_init:594 Starting.
thread(0001) 2386-04-30 23:21:00.733 INFO session_connect:624 Starting.
thread(0001) 2386-04-30 23:21:00.737 INFO session_connect:644 Done.
thread(0001) 2386-04-30 23:21:00.737 INFO session_register:750 Starting.
thread(0001) 2386-04-30 23:21:00.737 INFO session_register:785 sending data:
thread(0001) 2386-04-30 23:21:00.737 INFO session_register:786 00000 65 00 04 00 00 00 00 00 00 00
thread(0001) 2386-04-30 23:21:00.737 INFO session_register:786 00010 00 00 00 00 00 00 00 00 00 00
thread(0001) 2386-04-30 23:21:00.737 INFO session_register:786 00020 00 00 00 00 01 00 00 00 00
thread(0001) 2386-04-30 23:21:00.749 INFO session_register:861 received response:
thread(0001) 2386-04-30 23:21:00.749 INFO session_register:862 00000 65 00 04 00 26 b4 85 db 00 00
thread(0001) 2386-04-30 23:21:00.749 INFO session_register:862 00010 00 00 00 00 00 00 00 00 00 00
thread(0001) 2386-04-30 23:21:00.749 INFO session_register:862 00020 00 00 00 00 01 00 00 00 00
thread(0001) 2386-04-30 23:21:00.749 INFO session_register:885 Done.
thread(0001) 2386-04-30 23:21:00.749 INFO session_init:609 Done.
thread(0001) 2386-04-30 23:21:00.749 INFO ab_tag_create:354 Done.
thread(0001) 2386-04-30 23:21:00.749 INFO plc_tag_create:266 Returning mapped tag 00004001
thread(0001) 2386-04-30 23:21:00.750 INFO plc_tag_read:517 Starting.
thread(0001) 2386-04-30 23:21:00.750 INFO eip_pccc_tag_read_start:122 Starting
thread(0001) 2386-04-30 23:21:00.750 INFO session_add_request_unsafe:968 Starting.
thread(0001) 2386-04-30 23:21:00.750 INFO request_acquire:71 Acquiring request.
thread(0001) 2386-04-30 23:21:00.750 INFO refcount_acquire:61 Ref count is now 2
thread(0001) 2386-04-30 23:21:00.750 INFO session_add_request_unsafe:997 Total requests in the queue: 0
thread(0001) 2386-04-30 23:21:00.750 INFO session_add_request_unsafe:999 Done.
thread(0001) 2386-04-30 23:21:00.750 INFO eip_pccc_tag_read_start:236 Done.
thread(0002) 2386-04-30 23:21:00.751 INFO session_check_outgoing_data_unsafe:1078 Readying unconnected packet to send.
thread(0002) 2386-04-30 23:21:00.751 INFO request_acquire:71 Acquiring request.
thread(0002) 2386-04-30 23:21:00.751 INFO refcount_acquire:61 Ref count is now 3
thread(0002) 2386-04-30 23:21:00.751 INFO session_check_outgoing_data_unsafe:1086 sending packet, so 1 unconnected requests in flight.
thread(0002) 2386-04-30 23:21:00.752 INFO send_eip_request_unsafe:65 Sending unconnected packet with session sequence ID 6eac
thread(0002) 2386-04-30 23:21:00.752 INFO send_eip_request_unsafe:93 Sending packet of size 67
thread(0002) 2386-04-30 23:21:00.752 INFO send_eip_request_unsafe:94 00000 6f 00 2b 00 26 b4 85 db 00 00
thread(0002) 2386-04-30 23:21:00.752 INFO send_eip_request_unsafe:94 00010 00 00 ac 6e 00 00 00 00 00 00
thread(0002) 2386-04-30 23:21:00.752 INFO send_eip_request_unsafe:94 00020 00 00 00 00 00 00 00 00 01 00
thread(0002) 2386-04-30 23:21:00.752 INFO send_eip_request_unsafe:94 00030 02 00 00 00 00 00 b2 00 1b 00
thread(0002) 2386-04-30 23:21:00.752 INFO send_eip_request_unsafe:94 00040 4b 02 20 67 24 01 07 3d f3 45
thread(0002) 2386-04-30 23:21:00.752 INFO send_eip_request_unsafe:94 00050 43 50 21 0f 00 aa 6e 68 00 00
thread(0002) 2386-04-30 23:21:00.752 INFO send_eip_request_unsafe:94 00060 02 00 06 30 00 02 00 00
thread(0002) 2386-04-30 23:21:00.755 INFO request_release:83 Releasing request.
thread(0002) 2386-04-30 23:21:00.755 INFO refcount_release:91 Refcount is now 2
thread(0002) 2386-04-30 23:21:00.757 INFO session_check_outgoing_data_unsafe:1057 1 unconnected requests in flight.
thread(0002) 2386-04-30 23:21:00.759 INFO session_check_outgoing_data_unsafe:1057 1 unconnected requests in flight.
thread(0002) 2386-04-30 23:21:00.760 INFO session_check_outgoing_data_unsafe:1057 1 unconnected requests in flight.
thread(0002) 2386-04-30 23:21:00.763 INFO session_check_outgoing_data_unsafe:1057 1 unconnected requests in flight.
thread(0002) 2386-04-30 23:21:00.764 INFO session_check_outgoing_data_unsafe:1057 1 unconnected requests in flight.
thread(0002) 2386-04-30 23:21:00.766 INFO session_check_outgoing_data_unsafe:1057 1 unconnected requests in flight.
thread(0002) 2386-04-30 23:21:00.767 INFO recv_eip_response_unsafe:202 Received unconnected packet with session sequence ID 6eac
thread(0002) 2386-04-30 23:21:00.767 INFO receive_response:710 Packet sent initially 12ms ago and was sent 1 times
thread(0002) 2386-04-30 23:21:00.768 INFO update_resend_samples:690 Packet round trip time 13, running average round trip time is 129
thread(0002) 2386-04-30 23:21:00.768 INFO receive_response:715 got full packet of size 65
thread(0002) 2386-04-30 23:21:00.768 INFO receive_response:716 00000 6f 00 29 00 26 b4 85 db 00 00
thread(0002) 2386-04-30 23:21:00.768 INFO receive_response:716 00010 00 00 ac 6e 00 00 00 00 00 00
thread(0002) 2386-04-30 23:21:00.768 INFO receive_response:716 00020 00 00 00 00 00 00 00 00 00 04
thread(0002) 2386-04-30 23:21:00.768 INFO receive_response:716 00030 02 00 00 00 00 00 b2 00 19 00
thread(0002) 2386-04-30 23:21:00.768 INFO receive_response:716 00040 cb 00 00 00 07 3d f3 45 43 50
thread(0002) 2386-04-30 23:21:00.768 INFO receive_response:716 00050 21 4f 00 aa 6e 39 08 53 74 72
thread(0002) 2386-04-30 23:21:00.768 INFO receive_response:716 00060 69 6e 67 20 31 00
thread(0001) 2386-04-30 23:21:00.777 INFO request_release:83 Releasing request.
thread(0001) 2386-04-30 23:21:00.777 INFO refcount_release:91 Refcount is now 1
thread(0001) 2386-04-30 23:21:00.777 INFO request_release:83 Releasing request.
thread(0001) 2386-04-30 23:21:00.778 INFO refcount_release:91 Refcount is now 0
thread(0001) 2386-04-30 23:21:00.778 INFO refcount_release:94 Calling destructor function.
thread(0001) 2386-04-30 23:21:00.778 INFO check_read_status:350 Done.
thread(0001) 2386-04-30 23:21:00.779 INFO plc_tag_read:588 elapsed time 29ms
thread(0001) 2386-04-30 23:21:00.779 INFO plc_tag_read:591 Done
string 0 (82 chars) = 'String 1'
string 1 (82 chars) = ''

There should be two strings.   Only one is returned in the response packet.  It looks like the wrong read command might be issued.

kyle-github, 2017-06-11:
After some experimentation I have some results that are annoying.   While reading multiple strings works with a MicroLogix, you can only read a maximum of two strings because of the response packet size limitations.   For some DH+ connected SLC models (5/01 and 5/02 apparently), the limit is so small (41 words) that you can only get one string.   

I will modify the library in the following ways:

- Add in logic to handle multiple request packets for embedded PCCC calls.
- Add in logic to parse the PLC/5-style data file names so that I can determine when I need to request one thing at a time (strings) and when I do not (INT).   Experimentation will be needed to determine what I can read in groups and what I need to read one at a time. 
- Refactor some of the DH+ and PCCC embedded handling so that there is not so much duplication of code.

I am not sure how long this will take.   I am pretty busy right now.   But it is one of my top priorities.

kyle-github, 2017-09-09:
Marking this as an enhancement.   Apparently this is exactly how AB decided to make the protocol respond.   This seems highly annoying.

I-Campbell, 2018-01-11:
As the maximum response size is unknown without experimentation, is it feasible to do:
1. Ask for everything
2. Processor will respond with as much it can, eg two strings.
3. Examine response for completeness of data.
4. If only a portion of the data is responded, ask for the rest.
5. Repeat until all data retrieved or error.

A further enhancement on this would be to then store the expected number of elements returned for a particular processor type / data type / tag, and get it right first time the next time a read is requested.

kyle-github, 2018-01-12:
Possible?  Yes.   Feasible?   Not with the 1.x version of the library.  The problem there is that there is nothing to drive the above sequence of steps.  There isn't a thread of control for it.  I could build a state machine that sequenced when you called plc_tag_status() on a tag, but that is... not ideal.

The version 2.x (which exists, really!) is shaping up to be able to do this.  Not trivial by any means, but a lot more feasible than in the 1.x version.

kyle-github, 2018-12-14:
Version 2.0 does allow for shorter-than expected responses, but only on Logix-class tags so far.

I think the real fix will be to implement multiple-request support in PCCC.   It looks feasible, but it probably will not be a 2.0 feature.

kyle-github, 2018-12-14:
It may be easier to implement real CSP/PCCC/DF1 support first and then base the solution on the target PLC type.   Different CSP PLCs prefer different DF1 dialects.

kyle-github, 2018-12-18:
I am marking this as post 2.0.

kyle-github, 2019-06-02:
Now that SLC/MicroLogix support is broken out into two parts, it may be possible to look at this again.  I will take this up after I get the Debian packaging and some internal clean ups done.

kyle-github, 2019-10-07:
Closing this in favor of #114.   That will be the general issue for this support.

commit e9dbf86b (2020-04-26): fixes #54
closed 2019-10-07 by kyle-github