libplctag/libplctag #58: any issue with compiling for x64 in visual studio?

kyranf, 2017-10-17:
If I wanted to use this API as part of a C++ program compiled as a 64-bit program (and running on windows 7 pro x64) will we encounter any functionality issues due to compiling and running as a x64 executable? 

Will we have issues with memory alignment or any hard-coded bit-fiddling within this library? Thanks for any clarifications you can give. 

We are stuck with having to compile for 64 bit target due to another 3rd party library.

kyle-github, 2017-10-17:
I usually compile with 64-bit Windows when I use Windows.  I also use exclusively 64-bit code when I use Linux, so the code is 64-bit clean.

kyranf, 2017-10-19:
@kyle-github   thanks for that info, that's great news. 

An extra question I have is: this library/stack allows for read/write of tags in a host PLC, is the protocol handled as implicit messaging (constant read/write to tag names) or is it via explicit messaging (direct-messages) to named tags? 

from the setup required, it looks (at first glance) to support the implicit messaging way of doing things, which is ideal for real-time control/feedback and is better than doing single direct messages of explicit table contents.. 

my company has a lightweight working stack for direct messaging, but not the implicit continuous updating-tag-style communications

kyle-github, 2017-10-19:
Everything is explicit messaging still.   The horribly delayed 2.0 series is being reworked to support completely different models such as implicit messaging, automatic reconnection, better retry behavior and better handling of some older PLCs that do not speak CIP well.  It will also speak ModBus TCP after I finally come up for air and get a chance to hack that in.   After CIP, ModBus looks like a walk in the park.

kyranf, 2017-10-19:
Thankyou, that's all my questions for now. Closing my issue!

commit 6506bd88 (2020-04-26): fixes #58
closed 2017-10-19 by kyranf