libplctag/libplctag #60: SCL5/04 DH+ over serial

dhoney, 2017-11-17:
Is it possible to read DH+ over serial?

The examples seem to use a gateway are there other options to do this natively?

kyle-github, 2017-11-18:
At this point there is not.   Earlier versions of the library had some experimental serial code but I never had enough hardware to try it out.   

I may revive this at some point simply because I would like to have native comms with PLC 5 and SLC machines over TCP and that means DF1 encapsulated on TCP in what is sometimes called PCCC.   There are few docs on what that is that I can find though.

dhoney, 2017-11-18:
@kyle-github I have a SLC5/04, SLC4/03, and a Micrologix.

This gives access to DH+, DH485 and DF1.

Starting withDH+, If you can point me towards the branch or commit hash I'd be happy to do some testing and validation for you as this is a need I have.

kyle-github, 2017-11-18:
There is a lot of code to write.   The original 1.0 series had serial ports covered in the platform wrapper code (linux/platform.c and windows/platform.c).  I stripped it out in 2.0 for sure and maybe in 1.5 because I never used it.   There is not any code for DF1 packets, just the BCC/CRC coding and a few other things.  

There used to be a library floating around called something like libabcomms (not sure about that).   It was from the late 1990s I think.   It supported serial access.

dhoney, 2017-11-18:
This is the closest I found regarding DF1:
https://sourceforge.net/projects/libpccc/

kyle-github, 2017-11-18:
Oh, intersting!  I did not remember that one.  That is an older library too but it looks like the author is keeping it up to date.  I will have to investigate when I get some time.

My highest priority right now are getting a working 2.0 out the door.  This will have support for multi-tag requests and tag enumeration (get all the tags from a PLC) for ControlLogix-class systems.  Since this is a such a big change internally, and I do not have much free time right now, this is going to take a while.

I will mark this library to check when I have a moment.

kyle-github, 2017-11-18:
Thinking a bit more about the structure of how DF1 operates, it seems like patching this onto 2.0 is a better bet.   The 1.x versions really do not have much in the way of background threads handling things.   2.0 is a lot easier to work with in that regard.

This could be interesting to test someday.  I do not have any computers with serial ports anymore!

dhoney, 2017-11-19:
Are DF1 and DH+ fundamentally the same?  

Can a DF1 PLC talk DH+?

kyle-github, 2017-11-19:
The underlying packets of data are the same regardless of whether they are transmitted on serial (DF1), DH+ or TCP.  The wrapper around each differs.  IIRC, there is not CRC/BCC byte when using TCP.

I have very limited time and very limited access to PLCs now.   I do not have any serial ports on any of my computers.  I used to have a USB serial adapter, but I think I gave that away.

If you want to look at the protocols, they (at least DH+ and partial DF1 IIRC) are documented on line.   AB at least made those public.   AB's specs are always partially incomplete and usually either leave out important chunks or make things less clear than they could be :-(   Just being able to enumerate the tags in a ControlLogix-class PLC is something that clearly could be done, but it was not until recently that I could find any documentation (thanks to someone who was using the library).

I would love to get patches :-)

kyle-github, 2017-11-20:
Here is where I found the DF1 protocol docs:

http://literature.rockwellautomation.com/idc/groups/literature/documents/rm/1770-rm516_-en-p.pdf

dhoney, 2017-11-20:
Excellent thank you!

I'll admit I'm a bit excited I get to bust out sigrok this week and sniff some serial packets again :-)

Should I come up with anything meaningful I'd be delighted to contribute some patches.

kyle-github, 2019-09-07:
I am going to close this for now.   We have no serial connected systems any more and even the last of the DH+ has been phased out.   There are not many PLC/5 or SLC 500 systems left.  Only a few MicroLogix and most of those are being replaced too.

closed 2019-09-07 by kyle-github

dhoney, 2019-09-07:
@kyle-github Thank you for keeping up on this and I understand your decision.
However I will say that despite PLC/5 and SLC systems being officially "phased out", I do assure you they are still prevalent in the field.

Ebay these days is the defacto rockwell distributor and there are many OEM systems we've run into out in the rust belt area that will keep using older PLCs until either the entire machine is replaced or it is upgraded fully, both are almost equally as costly and so the systems usually stay in place.

Should you decide to pick this up again there are many cheap systems available online via ebay and my lab could potentially provide you remote access to some of these systems too.

Cheers

Seabee1971, 2019-10-09:
@dhoney I agree my facility has Rockwell PLCs going back to SLCs. I am trying to figure out how to integrate our old SLCs, micrologix, and controllogix PLCs into a scada type system using Python. The Controllogix has some rock solid modules for Python using Ethernet/IP but I am struggling to find and thing that is not Ethernet/IP based for Python. If you come up with any solutions, I would love to hear about them.

kyle-github, 2019-10-09:
I added a poll in issue #116.  Please vote!  I added this issue specifically to the poll.

dhoney, 2019-10-09:
Thank you for taking the time to set that up Kyle, I really appreciate it.

Would you be open to me polling other communities independently and linking
those results as well such as /r/plc?
Reason being I think a large part of users who benefit may not have github
accounts.

kyle-github, 2019-10-10:
Sure.  If they'll use (and improve) the library some day!

If someone does not have a GitHub account...  Hmm, hadn't thought of that.  I guess they can still download.  Maybe I should look for another way to do a poll?

I sometime look at /r/plc, but not that often.

kyle-github, 2019-10-10:
Hmm, I just noticed that the poll does not show the last option which was DF1/PCCC PLCs!

Seabee1971, 2019-10-10:
I was wondering about the DF1 option :)

kyle-github, 2019-10-11:
I killed the old poll and started one at SurveyMonkey.  Please see issue #116 for the link.  

There is also the possibility of checking "other" and entering specific requests.

kyle-github, 2019-10-13:
Here is the SurveyMonkey link.   Please vote!

kyle-github, 2019-10-13:
And now with the link: https://www.surveymonkey.com/r/D27XSZY!

Seabee1971, 2020-04-22:
Kyle, It has been about 6 months since the poll when up. Any updates? Now that I am working from home I have pulled out my SLC 05/04 and my microLogix 1500 in hopes of begging you to help me get them communicating with Python :)

commit 4f971849 (2020-04-26): fixes #60