libplctag/libplctag #62: Wrapper for FreePascal

fluisgirardi, 2018-02-01:
Hi!

I have done a wrapper for Freepascal. Soon as possible I'll write a small demo. Attached is the FPC wrapper.

[libplctag.pas.zip](https://github.com/kyle-github/libplctag/files/1686767/libplctag.pas.zip)

kyle-github, 2018-02-01:
Thanks!  I will pull this and try to get it into the distribution in the next few days (I am travelling right now).

kyle-github, 2018-12-26:
Hi, I have just released 2.0!   This has some API changes.   Can you check it out (check out branch 2.0) and let me know if you can make changes to the Free Pascal wrapper or you need help!

Witenite, 2019-02-02:
Hi guys,
firstly, great work. I have been looking for a suitable open source or at least freely available library to use with either Delphi or FPC (I use Code Typhon) and this really does appear to be the most suitable solution. I work for a large sawmill in New Zealand and we have a number of ControlLogix and PLC5 PLC's around the place. We extract a lot of data using some proprietary software, and the data is captured in a database. This proprietary software is not very good (it's also about 10 years old) and in addition to the database requirements, I have written a couple of Delphi/FPC programs to facilitate operations, but really need a library such as the one you have produced, to provide some much needed features. As a proof of principle, getting PLC communication working like this will really help my case in terms of convincing management that we need to invest in better tools, or (preferably) even contribute to open source efforts such as yours.
By chance, do you have some instructions or details in terms of how to implement the library and the associated wrapper for Lazarus? I could figure it out myself, but it's been quite some time since I did so (and then only in Delphi) and would save me a fair amount of fiddling time. If not, then not to worry, I'll figure it out.

If/when successful with the library/wrapper, I'll be happy to give some feedback in terms of my experience with the software on site, and how it performs etc. I prefer Linux, but we also have Windows servers at work etc. so I will probably have an opportunity to test it in different programs and OS over time.

kyle-github, 2019-02-02:
Hi, I am afraid that the last time I did anything in Pascal was about 30 years ago.  Ah, Turbo Pascal.   

I have not done anything with the library above.  I will try to get this pulled over the weekend and put into the Git repo.   Hopefully I can figure out what needs to happen to port this to 2.0!

I am not sure what Lazarus is, so I doubt I can help you there.  Perhaps the Pascal wrapper author can help?

Witenite, 2019-02-04:
Hello Kyle,
sorry, my comment was more directed at fluisgirardi, as apparently he has created a wrapper for Free Pascal (Lazarus). Delphi is a commercial "flavour" of Pascal. The Turbo Pascal you remember is the "grand-daddy" of modern day Object oriented Pascal. A very enjoyable language in my opinion, but I'm completely off-topic here.

fluisgirardi, 2019-02-04:
Hi Kyle, witenite!

As you can see, the wrapper was created to a previous libplctag version. So, I or someone else have to update it to handle the new libplctag.

If I'm right, I have a small demo, where I used to compare the scan time of on variable, in different OSs. I'll try find there, it was based on some Kyle example.

To me,  the most complicated step was built the library with the correct visual studio version (for Windows). On Linux it was easy. After that, install and use was easy.

@kyle, I'll try find some time to update the wrapper. I'm very busy last months.

kyle-github, 2019-02-05:
Thanks!

I hope to release the 2.0 branch as the main (master) branch over the next weekend.

I really liked Turbo Pascal's IDE.  The compiler was so fast!   Even on old hardware.  I would love to find an IDE like it for C.

Witenite, 2019-02-05:
Kyle,
if you ever feel up to it, try/look up Code Typhon which is an excellent  (also free/open source) IDE for Lazarus, a very good Object oriented Pascal compiler. In terms of C, I use Visual Studio these days which I believe is also good as a C compiler (or so I have been told). Eclipse is another very good option, and I use that to develop C code for Arduino boards. Far more powerful than the Arduino native IDE. See http://eclipse.baeyens.it/index.shtml
There are other C compiler options obviously, for Eclipse as well.

Witenite, 2019-02-05:
Hi fluisgirardi,
if you are able to update your wrapper to suit version 2.0 at some point, that would be really appreciated. I am looking forward to trying this out. Also if you have some brief instructions in terms of building the whole package for windows (Visual Studio as you mentioned) that would be excellent too. Sorry, I don't mean to sound as if I'm loading you up with work to do here, but as you have already done all the "hair pulling" and swearing, you now know the right procedure and it would save me a heap of time etc.

For what it's worth (as I've stated before) I am happy to provide feedback on how well everything works for me in a real-life application, and report any bugs I may find along the way (if any).
Thanks again.

kyle-github, 2019-02-06:
@Witenite thanks for the list.   I use Codelite mostly.   It is relatively easy to use and gets out of my way.  It also has reasonable support for CMake.   I have ended up investing a fair amount of effort into the CMake build system with the library, so having something that supports it fairly well out of the box is very nice.

@fluisgirardi I can pull the Zip file you sent into the source if you want.  I still have not had a chance to look at it :-(   I am busy trying to get ready for a vacation.

kyle-github, 2019-07-04:
Er, where did we leave this?   I would like to get the new version pulled into the repo.    Where is the repo for this wrapper?   I see the attachment, but not the repo??

fluisgirardi, 2019-07-04:
@kyle-github sorry, I have no time to do a make a complete revision o Pascal wrapper of libplctag, to assure that it's working well as the previous wrapper. The unique thing that I have done the wrapper, but it's UNTESTED!!
[libplctagv2.pas.zip](https://github.com/kyle-github/libplctag/files/3359276/libplctagv2.pas.zip)

kyle-github, 2019-07-04:
Thanks!!

I will try to get to this in the next week or two.  It is vacation time!

thevinge, 2019-08-15:
Hi @fluisgirardi 
Have you had time to test your new wrapper?

I have tried to use it with Delphi on Windows. I have no problem loading the dll into my application using your wrapper, however, when I try to create a PLC tag, I always get the value -4. I even get the value (-4) when the Tag_path is incorrect. I have tested the tag path with @kyle-github simple Cpp example, - and it works perfectly. I think it has something to do with the pointer to the tag_path in Delphi, but I am not sure.

Could you provide an example using your wrapper?

Thank you, @kyle-github  for writing an awesome library, and thank you, @fluisgirardi for making a wrapper for FP and Delphi.

fluisgirardi, 2019-08-15:
@thevinge yes I have time, but don't have a PLC...

thevinge, 2019-08-15:
@fluisgirardi , It works now, I found my mistake. I had to convert my string to an array of ansichar.

kyle-github, 2019-10-06:
Sorry I dropped the ball on this!   I am adding this into the repo now.

kyle-github, 2019-10-06:
I have this in a release version of 2.0.24.   Very sorry about the delay!

@thevinge can you please test that release version to see if it works?

kyle-github, 2020-03-30:
I will close this for now.   Hopefully it is working!

closed 2020-03-30 by kyle-github

fluisgirardi, 2020-03-30:
For me it's ok

commit 369f82c4 (2020-04-26): fixes #62