libplctag/libplctag #63: Micro800 Read Issues

patrickjmcd, 2018-02-23:
I'm having issues reading from a Micro800 PLC. I'm getting
```
Error setting up tag internal state. Error PLCTAG_ERR_REMOTE_ERR
ERROR: Unable to read the data! Got error code -11: PLCTAG_ERR_TIMEOUT
```

when using the tag path 
```
protocol=ab_eip&gateway=10.20.4.40&path=1,0&cpu=micro800&elem_count=1&elem_size=4&name=val_FlowTotalizerB
```

When working with another package I've been using, I needed to alter the forward open command to reference a Direct Connection vs a Backplane connection. I'm having trouble finding sample code that references a Micro800. Does anyone have a good example of this? 

If it helps, here's my trace from using debug=3
```
thread(0001) 2018-01-23 15:51:52.837 INFO find_tag_create_func:88 Matched protocol=ab_eip
thread(0001) 2018-01-23 15:51:52.838 INFO ab_tag_create:174 Starting.
thread(0001) 2018-01-23 15:51:52.838 INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
thread(0001) 2018-01-23 15:51:52.838 INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
thread(0001) 2018-01-23 15:51:52.838 INFO session_create_unsafe:429 Starting
thread(0001) 2018-01-23 15:51:52.838 INFO refcount_init:33 Initializing refcount struct with count=1
thread(0001) 2018-01-23 15:51:52.838 INFO session_create_unsafe:486 Done
thread(0001) 2018-01-23 15:51:52.838 INFO session_init:502 Starting.
thread(0001) 2018-01-23 15:51:52.838 INFO session_connect:532 Starting.
thread(0001) 2018-01-23 15:51:52.899 INFO session_connect:552 Done.
thread(0001) 2018-01-23 15:51:52.900 INFO session_register:636 Starting.
thread(0001) 2018-01-23 15:51:52.900 INFO session_register:671 sending data:
thread(0001) 2018-01-23 15:51:52.900 INFO session_register:672 00000 65 00 04 00 00 00 00 00 00 00
thread(0001) 2018-01-23 15:51:52.900 INFO session_register:672 00010 00 00 00 00 00 00 00 00 00 00
thread(0001) 2018-01-23 15:51:52.900 INFO session_register:672 00020 00 00 00 00 01 00 00 00 00
thread(0001) 2018-01-23 15:51:52.962 INFO session_register:747 received response:
thread(0001) 2018-01-23 15:51:52.962 INFO session_register:748 00000 65 00 04 00 01 86 c7 a1 00 00
thread(0001) 2018-01-23 15:51:52.962 INFO session_register:748 00010 00 00 00 00 00 00 00 00 00 00
thread(0001) 2018-01-23 15:51:52.962 INFO session_register:748 00020 00 00 00 00 01 00 00 00 00
thread(0001) 2018-01-23 15:51:52.962 INFO session_register:771 Done.
thread(0001) 2018-01-23 15:51:52.962 INFO session_init:517 Done.
thread(0001) 2018-01-23 15:51:52.962 INFO connection_find_or_create:88 Starting.
thread(0001) 2018-01-23 15:51:52.962 INFO connection_create_unsafe:177 Starting.
thread(0001) 2018-01-23 15:51:52.962 INFO refcount_init:33 Initializing refcount struct with count=1
thread(0001) 2018-01-23 15:51:52.962 INFO session_acquire:934 Acquire session=0x7fdaa7804800
thread(0001) 2018-01-23 15:51:52.962 INFO refcount_acquire:63 Ref count is now 2
thread(0001) 2018-01-23 15:51:52.962 INFO connection_create_unsafe:232 Done.
thread(0001) 2018-01-23 15:51:52.962 INFO connection_find_or_create:108 Creating new connection.
thread(0001) 2018-01-23 15:51:52.962 INFO connection_perform_forward_open:245 Starting.
thread(0001) 2018-01-23 15:51:52.962 INFO refcount_init:33 Initializing refcount struct with count=1
thread(0001) 2018-01-23 15:51:52.962 INFO send_forward_open_req:308 Starting
thread(0001) 2018-01-23 15:51:52.962 INFO session_add_request_unsafe:802 Starting.
thread(0001) 2018-01-23 15:51:52.962 INFO request_acquire:75 Acquiring request.
thread(0001) 2018-01-23 15:51:52.962 INFO refcount_acquire:63 Ref count is now 2
thread(0001) 2018-01-23 15:51:52.962 INFO session_add_request_unsafe:831 Total requests in the queue: 0
thread(0001) 2018-01-23 15:51:52.962 INFO session_add_request_unsafe:833 Done.
thread(0001) 2018-01-23 15:51:52.962 INFO send_forward_open_req:375 Done
thread(0002) 2018-01-23 15:51:52.963 INFO session_check_outgoing_data_unsafe:970 Readying connected packet to send.
thread(0002) 2018-01-23 15:51:52.963 INFO request_acquire:75 Acquiring request.
thread(0002) 2018-01-23 15:51:52.963 INFO refcount_acquire:63 Ref count is now 3
thread(0002) 2018-01-23 15:51:52.963 INFO session_check_outgoing_data_unsafe:978 sending packet, so 1 connected requests in flight.
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:65 Sending unconnected packet with session sequence ID 2ac3445d
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:93 Sending packet of size 88
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:94 00000 6f 00 40 00 01 86 c7 a1 00 00
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:94 00010 00 00 5d 44 c3 2a 00 00 00 00
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:94 00020 00 00 00 00 00 00 00 00 01 00
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:94 00030 02 00 00 00 00 00 b2 00 30 00
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:94 00040 54 02 20 06 24 01 0a 05 00 00
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:94 00050 00 00 c9 22 68 1a 30 80 3d f3
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:94 00060 45 43 50 21 01 00 00 00 40 42
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:94 00070 0f 00 f8 43 40 42 0f 00 f8 43
thread(0002) 2018-01-23 15:51:52.963 INFO send_eip_request_unsafe:94 00080 a3 03 01 00 20 02 24 01 00
thread(0002) 2018-01-23 15:51:52.963 INFO request_release:87 Releasing request.
thread(0002) 2018-01-23 15:51:52.963 INFO refcount_release:93 Refcount is now 2
thread(0002) 2018-01-23 15:51:53.029 INFO recv_eip_response_unsafe:202 Received unconnected packet with session sequence ID 2ac3445d
[... 192 lines cut ...]
thread(0001) 2018-01-23 15:51:58.102 WARN connection_close:546 Unable to use ForwardClose response!
thread(0001) 2018-01-23 15:51:58.102 INFO request_release:87 Releasing request.
thread(0001) 2018-01-23 15:51:58.102 INFO refcount_release:93 Refcount is now 0
thread(0001) 2018-01-23 15:51:58.102 INFO refcount_release:96 Calling destructor function.
thread(0001) 2018-01-23 15:51:58.102 INFO connection_close:560 Done.
thread(0001) 2018-01-23 15:51:58.102 INFO session_release:946 Release session=0x7fdaa7804800
thread(0001) 2018-01-23 15:51:58.102 INFO refcount_release:93 Refcount is now 1
thread(0001) 2018-01-23 15:51:58.102 INFO connection_destroy:483 Done.
thread(0001) 2018-01-23 15:51:58.102 INFO session_release:946 Release session=0x7fdaa7804800
thread(0001) 2018-01-23 15:51:58.102 INFO refcount_release:93 Refcount is now 0
thread(0001) 2018-01-23 15:51:58.102 INFO refcount_release:96 Calling destructor function.
thread(0001) 2018-01-23 15:51:58.102 INFO session_destroy:563 Starting.
thread(0001) 2018-01-23 15:51:58.102 INFO session_destroy:622 Done.
thread(0001) 2018-01-23 15:51:58.102 INFO ab_tag_destroy:537 Finished releasing all tag resources.
thread(0001) 2018-01-23 15:51:58.102 INFO ab_tag_destroy:539 done
thread(0001) 2018-01-23 15:51:58.102 INFO plc_tag_destroy_mapped:501 Done.
thread(0001) 2018-01-23 15:51:58.102 INFO plc_tag_destroy:524 Done.
thread(0001) 2018-01-23 15:51:58.102 INFO ab_teardown:148 Releasing global AB protocol resources.
thread(0001) 2018-01-23 15:51:58.102 INFO ab_teardown:150 Terminating IO thread.
thread(0001) 2018-01-23 15:51:58.104 INFO ab_teardown:158 Freeing global session mutex.
thread(0001) 2018-01-23 15:51:58.104 INFO ab_teardown:162 Done.
thread(0001) 2018-01-23 15:51:58.104 INFO lib_teardown:114 Tearing down library.
thread(0001) 2018-01-23 15:51:58.105 INFO lib_teardown:124 Destroying global library mutex.
thread(0001) 2018-01-23 15:51:58.105 INFO lib_teardown:130 Done.
```

kyle-github, 2018-02-24:
Hi Patrick,

Try leaving out the "path=1,0" part.   It has been a while since I tested a
Micro800 but I think that there is no routing path.

They are a little weird.   They only implement part of the protocol.
 Devices can vary in what they support and do not support.   Some DH+
gateways, for instance, will accept any path element, but will ignore it.
Others will fail to connect if you provide a path :-(

Best,
Kyle

patrickjmcd, 2018-02-24:
Bingo. Nailed it.

Thanks for the help.

Would it be possible to add that under the Micro800 section of README.md for others?

commit bd556027 (2020-04-26): fixes #63
closed 2018-02-28 by patrickjmcd