libplctag/libplctag #65: Support automatic reconnection

kyle-github, 2018-03-17:
It is too annoying to have to do simple reconnections in every app using the library.   And, the rules for when reconnection works are not as easy to understand as they should be.

Implement reconnection within the library.   TBD on how this should look.

xinthose, 2018-03-17:
An auto-reconnect feature would be great, like setting the reconnect option to MySQL.

kyle-github, 2018-12-14:
Some of this is working in 2.0 so far, but I still need a way to pass along the reconnection timeout time.  There are multiple aspects to this:

- how long should the library wait before giving up?  It cannot be a fixed value as connections vary widely in the amount of time, and variance, it takes to get a response.
- once we decide to terminate a session, how long should we wait until trying it again?
- how many times should we retry.  Alternatively, how long should we go without a successful reconnection?

kyle-github, 2018-12-26:
A version of this is in 2.0.  I will open additional bugs against the library for specific changes.

commit 7b140710 (2020-04-26): fixes #65
closed 2018-12-26 by kyle-github