libplctag/libplctag #66: Hi,

truong32ct, 2018-04-13:


truong32ct, 2018-04-13:
and when i try with this code, it cannot read the data:
#include <stdio.h>
#include "../lib/libplctag.h"
#include "utils.h"

#define TAG_PATH "protocol=ab_eip&gateway=192.168.1.101&path=1,0&cpu=LGX&elem_size=2&elem_count=1&name=Temp_Engineer&debug=1"
#define ELEM_COUNT 1
#define ELEM_SIZE 2
#define DATA_TIMEOUT 5000

int main(int argc, char **argv )
{
	plc_tag tag = PLC_TAG_NULL;
	int rc;
	int i;
	tag=plc_tag_create(TAG_PATH);
	rc=plc_tag_read(tag,DATA_TIMEOUT);
	if(rc!=PLCTAG_STATUS_OK)
  {
	fprintf(stderr, "tag fault, fault code: %d\n", rc);
	return 0;
  }
	for (i=0; i<ELEM_COUNT;i++)
  {
	int val= plc_tag_get_int16(tag, (i*ELEM_SIZE));
	for (i=0; i<ELEM_COUNT;i++)
   {
	fprintf(stderr,"temp[%d]=%d\n",i,val);
   }
  }
	plc_tag_destroy(tag);
	
	return 0;
}

[image]

kyle-github, 2018-04-14:
Hi,
Please update the tag string as follows:
`protocol=ab_eip&gateway=192.168.1.101&path=1,0&cpu=LGX&elem_size=2&elem_count=1&name=Temp_Engineer&debug=3`

This `debug=3` provides more debugging output.  

I suggest that you try to use the tag_rw tool in examples:

`./tag_rw -t sint16 -p 'protocol=ab_eip&gateway=192.168.1.101&path=1,0&cpu=LGX&elem_size=2&elem_count=1&name=Temp_Engineer&debug=3'`

I use this until I get the tag string correct and then work it into a program.

Have you checked the basics:
- can you ping the PLC from the RPi?
- Is there any kind of firewall between the RPi and the PLC?
- Is the tag name correct?
- Is the tag type really INT (16-bit signed integer)?

You mentioned FLOAT above, but your example is looking for a 16-bit integer.

truong32ct, 2018-04-14:
Thank you Kyle for fast reponse. 
I already check these things:
- can you ping the PLC from the RPi? I pinged and it is ok 
- Is there any kind of firewall between the RPi and the PLC? there is no firewall
- Is the tag name correct? the name is correct
- Is the tag type really INT (16-bit signed integer)? yes
Yesterday, I tried different type of data to test and I think the problem is the tag name, when I change Temp_Engineer into PROGRAM:MainProgram.Temp_Engineer. It read correctly. Thank you very much Kyle. 
[image]
[image]

truong32ct, 2018-04-14:
Furthermore, there are some more questions:
- How could I read 2 tag at the same time?
- Can I continuously read the data in real time?
- How can I move the data into mySQL? I tried to include mySQL library into the code but when I build, I saw it cannot understand the library. 
Can you help me out?
[image]
[image]

truong32ct, 2018-04-14:
Up to now, I cannot read the tag without the prefix. There are other program tags in my PLC code and I want to read many of them in real time, can you help me to do that?

kyle-github, 2018-05-04:
Tags cannot be read as a stream.  You will need to do a call to plc_tag_read() each time you want to get new state from the PLC.  

As for logging it into a database, I suggest creating some tests where you only work on the database and getting it to read and write.   The MySQL community is quite large and there are many examples online.

commit aa059ebf (2020-04-27): fixes #66
closed 2018-05-04 by kyle-github