libplctag/libplctag #69: provides a pkg-config file

scontini76, 2018-05-04:
Many libraries provides a pkg-config (.pc) file that helps to detect the library and provides the correct compiler and linker parameters to the build system. It would be nice if libplctag follows this practice. 

see: https://www.freedesktop.org/wiki/Software/pkg-config/

If you agree I could help you with a pull request on this issue.

kyle-github, 2018-05-10:
Oops!  I missed this one. 

Yes, please!   I would love to have some idea how to support this.

scontini76, 2018-05-10:
Please have a look at my pull request #70

kyle-github, 2018-06-02:
Sorry for the delay.   This was merged.   Thank you!

Do you know of a way to take the CMake version information and use that to change the version in the source code?

kyle-github, 2018-06-02:
Thanks, again!

I have merged this and done another release.

closed 2018-06-02 by kyle-github

scontini76, 2018-06-02:
Glad that you like the changes. I'm not very skilled with CMake, I mostly use automate and recently meson, but I can have a look to pass define to compiler in CMake or how to prepare a config.h file, I'll work on it if you want, fill an issue so I'll remember to work on it in my spare time!

kyle-github, 2018-06-09:
Thanks, I will take a look and see if I can figure it out.   It should be possible.