libplctag/libplctag #73: A crush

oravnat, 2018-07-27:
We have a large system which reads a lot of tags. The system works great for about 2 hours, but then it crushed.
We tried to find the cause, and it seems that it is in the libplctag library. More specific:
In the function send_eip_request_unsafe() in file eip.c we have
            req->session->session_seq_id++;
and req->session is NULL.
We use windows 10 and most of the code of the program is in c#. The program uses a lot of threads, but we use a lock (the c# lock keyword) to protect the plc_tag_read() and plc_tag_get_xxx () calls.
However, we use a different lock for each tag, as we understand that different tags may be read in parallel. 
Please help us to solve problem.
Thanks!

kyle-github, 2018-07-28:
This does not sound good!

Can you run the system in debug mode?

At the end of each tag attribute string, add "&debug=3".   Capture the output into a file and attach it here.

Thanks!

kyle-github, 2018-07-28:
I am looking through the code to see how we could get to send_eip_request_unsafe() and have session be null...  I will put a check in there, but this almost sounds like memory corruption.   Again, still a bug that it can even happen!

oravnat, 2018-07-29:
Thank you very much for the quick response!
The log file was huge, so we attach only the last minute before the crush.
We see now that we didn't use the last version. We use version v1.5.10 of the library (10-Jun-18).
We will try the last version soon.

The stack where the problem occurs is following:

>	plctag.dll!send_eip_request_unsafe(ab_request_t * req) Line 57	C
 	plctag.dll!session_send_current_request(ab_session_t * session) Line 899	C
 	plctag.dll!session_check_outgoing_data_unsafe(ab_session_t * session) Line 999	C
 	plctag.dll!process_session_tasks_unsafe(ab_session_t * session) Line 1026	C
 	plctag.dll!request_handler_func(void * not_used) Line 1072	C
 	kernel32.dll!@BaseThreadInitThunk@12()	Unknown

[s-log.txt](https://github.com/kyle-github/libplctag/files/2238854/s-log.txt)

kyle-github, 2018-07-30:
Version 1.5.11 is a preview but it might be worth a try.   The session pointer should never be null at the point where this is crashing.   I am trying to trace down how that could happen.

kyle-github, 2018-07-31:
I was able to go through the log snippet.  Here is the overview of what I see:

1. the PLC is being flooded with requests.
2. at some point, the PLC reboots or resets the connection.
3. as the tags error out due to that, it hits a bug in my code and calls through a null pointer.

First, it looks like you are talking to a PLC5, SLC 500 or MicroLogix.   That is extremely slow!   So, if you have a lot of threads reading a lot of data files, you are going to overrun the PLC.  There are retries happening due to the very slow response times.  These are flooding the PLC system.

It looks like the PLC itself is resetting or rebooting.   I see the packets getting through (with a lot of retries and large delays), then suddenly I see errors show up from the send function.  The error is:

> Winsock Error 10054. Winsock error 10054 is "Connection reset by peer", or "An existing connection was forcibly closed by the remote host". The error means exactly what it says - the remote host closed the connection unexpectedly.

Are you trying to read tags from one PLC or many?   There appear to be multiple TCP connections open.  However, once a failure is seen no further packets get through.   It really looks like the core of the problem is that the PLC is being flooded and resets itself or at least the connection.

It can often take at least 50ms (and sometimes longer) to read a single data file from a PLC/5.   That is what I have to test on and it is _slow_.   So, consider that you might only be able to read 10-20 data files per second.  At best.  Probably fewer.

Once the failures start to happen, the tag threads start to time out and reset.   Eventually that seems to trigger the bug that causes the crash.

Here are some suggestions in order of how useful they will be (in my opinion!):

1. I can see that you read a single value from a lot of N and F files.   Can you add some code in the PLC to copy that data into a single N file and a single F file?   Then, use a tag read that gets the single N file and another for the single F file.  This will reduce the number of threads needed and reduce the number of packets sent to the PLC.
2. space out your reads.  Put a delay of at least one second between each read of a tag in each thread.  That still might be too much.  Maybe two seconds would be a better delay?
3. use fewer threads.   Read all the tags in one thread, sequentially.

I will try to figure out how to solve the crash problem.  At least I have an idea of how to cause it now!

oravnat, 2018-08-02:
The tags are from one PLC in this case. I saw in the debugger we had only one session.
After your response, we changed the reads to be in blocks instead of single tags (this was planned anyway). This improves the performance, but the crash still happens. We also read all the tags from one thread now. We use the timeout in the read method, so the next read takes place only after the previous one is completed.
The connection between the computer and the PLC may be unstable in our system (through a cellular modem). Do you think this may be the core of the problem?

kyle-github, 2018-08-04:
Ah, a cellular connection will be even worse and will have a lot of packet loss and high latency.   It is possible that the entire connection is going down.  Perhaps it is not the PLC that is resetting the connection, but the cellular modem itself?

The crash is still my problem, but it sounds like you are going to have a difficult time getting much data from that PLC without reading a lot less frequently.   How often are you reading these tags?

Is the crash happening about as often?

Try to the 1.5.11 version of the library I made available as a pre-release.   I added a bit more checking to that code and it might help the crash problem (not really the underlying problem, but it should prevent it from being as bad).

kyle-github, 2018-09-03:
I think this is related to issue #75.  
The failure mode here is probably causing or being caused by the same flaw.   At least I think so.

kyle-github, 2018-10-11:
Can you please try the latest 1.6 branch?   There are some changes that may help this problem.  I am not sure if it will fix it in your case, but it should make things better.

oravnat, 2018-10-12:
Thank you,
We will try the 1.6 branch soon.

oravnat, 2018-10-15:
We tried the latest 1.6 branch and it didn't work for us.
The program didn't crash now, but we also couldn't read the tags. Apparently, the session timeout is continuously passed when reading a tag and the worker thread exited the loop.

2387-09-15 10:24:14.400 thread(0001) WARN recv_eip_response:1273 Timed out waiting for data to read!
2387-09-15 10:24:14.400 thread(0001) WARN process_requests:1048 Error receiving packet response PLCTAG_ERR_TIMEOUT!
2387-09-15 10:24:14.400 thread(0001) WARN session_handler:915 Error while processing requests PLCTAG_ERR_TIMEOUT!

kyle-github, 2018-10-16:
Wow!   The timeout is 2 seconds.  

Can you easily build the library?   If so, can you change SESSION_DEFAULT_TIMEOUT in src/protocols/ab/session.h?   Currently it is set to 2000 which is 2 seconds (the value is in milliseconds).

Can you set it to something higher and see if that helps?  Perhaps 5 seconds?

If not, I may back out the timeout code completely or make it extremely long.  I want to have some sort of error catching so I will need to keep some form of timeout probably.   Or perhaps I need to figure out how to turn off the timeout completely.

Do you have any idea how long the timeout would need to be to read the tags over the cell modem?   It sounds like the connection is quite bad.

oravnat, 2018-10-16:
It was my mistake. I used the path attribute where I shouldn't.
We test the latest 1.6 branch now.
We can easily build the library, but it now works with your code without changes. It looks like the 2 seconds timeout is sufficient for the cell modem.

oravnat, 2018-10-19:
OK, it doesn't crash now. Each time the connection is lost, all the relevant tags stop working, but we detect when it happens and re-create the tags.
I understand that automatic re-connection is not implemented yet and scheduled to a later version, but is there a simple way to manually request a re-connection?
By looking at the library code, it looks like creating a new tag with the specific gateway would initialize the re-connection - but is there a better way?
Thank you.

kyle-github, 2018-10-22:
Right now the best way is to destroy the tag and create a new one.   However, you can run into a race condition (which I am trying to fix with version 1.6) where you need to destroy all tags first before creating any new ones.  

Version 1.6 should behave better and be a lot more flexible in this regard.  Please give that a try.   You should be able to simply destroy each tag and recreate it to reconnect.

kyle-github, 2018-12-14:
The entire code path that this crash was on is now changed in 2.0.  Once that is released, I would appreciate it if you could try it once it is out.   Thanks for your help!

kyle-github, 2018-12-26:
2.0 has been released.  I am keeping it on a separate branch for a few weeks to make sure that all the wrapper owners get a chance to port their code.

oravnat, 2018-12-27:
I will try it soon.

kyle-github, 2019-01-19:
If you can test on Linux, please test with 2.0.7.  I will need to work to make sure that Windows does not crash when the remote side closes the connection.   Unfortunately, this is difficult for me to recreate on demand.

kyle-github, 2019-01-19:
BTW, the new code should reconnect automatically in about 10 seconds.   I will be working on additions later to make this a configurable parameter, but only once reconnection is stable.

franklupo, 2019-01-21:
Hi,
I'm not sure it's the same problem, but with 2.0.7 I remain in wait for 60 sec then it gives me error and crashes.

```
protocol=ab_eip&gateway=10.155.128.192&path=1, 0&cpu=LGX&elem_size=4&elem_count=1&name=TKP_PLC_D_P1[10]&debug=3
2019-00-21 14:40:48.863 thread(1) tag(0) INFO find_tag_create_func:80 Matched protocol=ab_eip
2019-00-21 14:40:48.863 thread(1) tag(0) INFO ab_tag_create:156 Starting.
2019-00-21 14:40:48.864 thread(1) tag(0) INFO rc_alloc_impl:99 Starting, called from ab_tag_create:163
2019-00-21 14:40:48.864 thread(1) tag(0) INFO rc_alloc_impl:118 Done
2019-00-21 14:40:48.864 thread(1) tag(0) INFO session_create_unsafe:482 Starting
2019-00-21 14:40:48.864 thread(1) tag(0) INFO rc_alloc_impl:99 Starting, called from session_create_unsafe:486
2019-00-21 14:40:48.864 thread(1) tag(0) INFO rc_alloc_impl:118 Done
2019-00-21 14:40:48.864 thread(1) tag(0) INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
2019-00-21 14:40:48.864 thread(1) tag(0) INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
2019-00-21 14:40:48.864 thread(1) tag(0) INFO cip_encode_path:308 Done.
2019-00-21 14:40:48.864 thread(1) tag(0) INFO session_create_unsafe:577 Done
2019-00-21 14:40:48.864 thread(1) tag(0) INFO session_init:593 Starting.
2019-00-21 14:40:48.864 thread(1) tag(0) INFO session_init:607 Done.
2019-00-21 14:40:48.864 thread(1) tag(0) INFO ab_tag_create:323 Done.
2019-00-21 14:40:48.864 thread(2) tag(0) INFO session_handler:949 Starting thread for session 0xa51ec0
2019-00-21 14:40:48.864 thread(2) tag(0) INFO session_open_socket:622 Starting.
2019-00-21 14:40:48.864 thread(2) tag(0) INFO session_open_socket:639 Done.
2019-00-21 14:40:48.864 thread(2) tag(0) INFO session_register:652 Starting.
2019-00-21 14:40:48.870 thread(2) tag(0) INFO session_register:719 Done.
2019-00-21 14:40:48.870 thread(2) tag(0) INFO perform_forward_open:1772 Starting.
2019-00-21 14:40:48.870 thread(2) tag(0) INFO try_forward_open_ex:1858 Starting.
2019-00-21 14:40:48.870 thread(2) tag(0) INFO rc_alloc_impl:99 Starting, called from session_create_request:2283
2019-00-21 14:40:48.870 thread(2) tag(0) INFO rc_alloc_impl:118 Done
2019-00-21 14:40:48.870 thread(2) tag(0) INFO send_forward_open_req_ex:2023 Starting
2019-00-21 14:40:48.870 thread(2) tag(0) INFO send_forward_open_req_ex:2082 Done
2019-00-21 14:40:48.870 thread(2) tag(0) INFO recv_forward_open_resp:2095 Starting
2019-00-21 14:40:48.874 thread(2) tag(0) INFO recv_forward_open_resp:2151 ForwardOpen succeeded with our connection ID 5849fdd0 and the PLC connection ID ffff5268
2019-00-21 14:40:48.874 thread(2) tag(0) INFO recv_forward_open_resp:2160 Done.
2019-00-21 14:40:48.874 thread(2) tag(0) INFO refcount_cleanup:244 Starting
2019-00-21 14:40:48.874 thread(2) tag(0) INFO refcount_cleanup:256 Done.
2019-00-21 14:40:48.874 thread(2) tag(0) INFO try_forward_open_ex:1898 Done.
2019-00-21 14:40:48.874 thread(2) tag(0) INFO perform_forward_open:1819 Done.
2019-00-21 14:40:48.875 thread(1) tag(0) INFO plc_tag_create:458 tag set up elapsed time 11ms
2019-00-21 14:40:48.875 thread(1) tag(11) INFO plc_tag_create:476 Returning mapped tag ID 11
2019-00-21 14:40:48.875 thread(1) tag(11) INFO plc_tag_create:478 Done.
protocol=ab_eip&gateway=10.155.128.192&path=1, 0&cpu=LGX&elem_size=4&elem_count=1&name=TKP_PLC_B_P1&debug=3
2019-00-21 14:40:48.875 thread(1) tag(11) WARN lookup_tag:1762 Tag with ID 0 not found.
2019-00-21 14:40:48.875 thread(1) tag(0) WARN plc_tag_set_int32:1214 Tag not found.
2019-00-21 14:40:48.875 thread(1) tag(0) INFO plc_tag_create:328 Starting
2019-00-21 14:40:48.875 thread(1) tag(0) INFO find_tag_create_func:80 Matched protocol=ab_eip
2019-00-21 14:40:48.875 thread(1) tag(0) INFO ab_tag_create:156 Starting.
2019-00-21 14:40:48.875 thread(1) tag(0) INFO rc_alloc_impl:99 Starting, called from ab_tag_create:163
2019-00-21 14:40:48.875 thread(1) tag(0) INFO rc_alloc_impl:118 Done
2019-00-21 14:40:48.875 thread(1) tag(0) INFO ab_tag_create:323 Done.
2019-00-21 14:40:48.875 thread(1) tag(0) INFO plc_tag_create:458 tag set up elapsed time 0ms
2019-00-21 14:40:48.875 thread(1) tag(12) INFO plc_tag_create:476 Returning mapped tag ID 12
2019-00-21 14:40:48.875 thread(1) tag(12) INFO plc_tag_create:478 Done.
2019-00-21 14:40:48.896 thread(1) tag(12) INFO plc_tag_read:662 Starting.
2019-00-21 14:40:48.896 thread(1) tag(12) INFO tag_read_start:185 Starting
2019-00-21 14:40:48.896 thread(1) tag(12) INFO build_read_request_connected:277 Starting.
2019-00-21 14:40:48.896 thread(1) tag(12) INFO rc_alloc_impl:99 Starting, called from session_create_request:2283
2019-00-21 14:40:48.896 thread(1) tag(12) INFO rc_alloc_impl:118 Done
2019-00-21 14:40:48.896 thread(1) tag(12) INFO session_add_request_unsafe:853 Starting.
2019-00-21 14:40:48.896 thread(1) tag(12) INFO session_add_request_unsafe:873 Total requests in the queue: 1
2019-00-21 14:40:48.896 thread(1) tag(12) INFO session_add_request_unsafe:875 Done.
2019-00-21 14:40:48.896 thread(1) tag(12) INFO build_read_request_connected:354 Done
2019-00-21 14:40:48.896 thread(1) tag(12) INFO tag_read_start:205 Done.
2019-00-21 14:40:48.897 thread(2) tag(0) INFO pack_requests:1455 Starting.
2019-00-21 14:40:48.897 thread(2) tag(12) INFO pack_requests:1465 Only one request, so done.
2019-00-21 14:40:48.897 thread(2) tag(0) INFO prepare_request:1596 Preparing connected packet with connection ID 5849fdd0 andsequence ID 1(1)
[... cut ...]
2019-00-21 14:41:48.914 thread(2) tag(0) INFO session_unregister:733 Done.
2019-00-21 14:41:48.914 thread(2) tag(0) INFO session_close_socket:742 Starting.
2019-00-21 14:41:48.914 thread(2) tag(0) INFO session_close_socket:750 Done.
2019-00-21 14:41:48.914 thread(1) tag(13) WARN check_read_status_connected:845 Session reported failure of request: PLCTAG_ERR_WRITE.
2019-00-21 14:41:48.914 thread(1) tag(13) INFO refcount_cleanup:244 Starting
2019-00-21 14:41:48.914 thread(1) tag(13) INFO refcount_cleanup:256 Done.

```

kyle-github, 2019-01-21:
Thanks for the bug report!   

I think this is similar to what user user GBT is seeing on the forum.   I managed to recreate it locally with a longer wait between requests, so I should be able to fix this.

This is definitely due to new code in 2.0.

kyle-github, 2019-01-21:
I have released version 2.0.9 for this problem.   I was able to reliably recreate it simply by waiting for 3 minutes.   I would really appreciate it if you can test whether this fixes your crash bug!

I noticed something odd in your log above:

> 2019-00-21 14:40:48.904 thread(1) tag(12) WARN lookup_tag:1762 **Tag with ID 0 not found.**
> 2019-00-21 14:40:48.904 thread(1) tag(0) WARN plc_tag_set_int32:1214 **Tag not found.**
> 2019-00-21 14:40:48.904 thread(1) tag(0) INFO plc_tag_create:328 Starting

This is all from the thread that was creating the tags.   In this case, plc_tag_set_int32() is being called with a tag ID of 0.   That is not a valid tag ID.   No values less than 10 are created.   0 (zero) sounds more like a status that is being passed as a tag ID.   I see this in the log several times.   All times, it is followed by a tag creation as above.   You might want to check some of your tag handling logic.

Again, thank you for the bug report!

franklupo, 2019-01-21:
HI,
I fix ID equal 0. I test 2.0.9.

Best regards

kyle-github, 2019-01-21:
I released more updates.  I finished up my code to do automatic disconnect and reconnect.   That should eliminate a large number of the errors you are seeing.   Waiting 60 seconds between reads would often cause the PLC to time out.   The 2.0.10 code will automatically disconnect from the PLC after 5 seconds of inactivity.  It will reconnect automatically when there are more requests to perform.

franklupo, 2019-01-22:
Hi,
I tested the new version on linux with L82 eth and there are no problems. I try with windows in the day.

Best regards.

franklupo, 2019-01-22:
Windows ok.

Thanks!

kyle-github, 2019-02-10:
Please reopen this if it is still seen.   Thanks for all the help reporting and debugging this!

closed 2019-02-10 by kyle-github