libplctag/libplctag #74: Compatibility with .NET 4.0 / VS2010

ouelletsimon, 2018-07-31:
Hi!

Is there any chance that you could publish a build for .NET 4.0 / Visual Studio 2010?

Thanks

kyle-github, 2018-08-01:
I do not have that version of Visual Studio.  I will see if I can find a copy of the free version floating around.  

Have you tried to build the code with it?  If so, was there any particular that that was failing?

I do use some C99-standard features.   You would think by 2010 those would be supported, but unfortunately, Microsoft decided not to support much of C99 until sometime around Visual Studio 2012 (and even then the support is spotty).   I am not sure that they support all that much of C11 yet.

I was able to make the code compile under VS 2012 by forcing compilation as if it was C++.  The CMake build system should do this automatically.

What OS are you targeting with a compiler that old?   As far as I know VS 2012 will still build code for Windows XP.

kyle-github, 2018-08-26:
Any luck?

If you are using and targeting Windows 7, then you can use one of the Visual Studio community editions.   They tend to be about a year behind the "full fat" commercial Visual Studio editions.  They are what I use.  The versions are free.

[Visual Studio Community Edition](https://visualstudio.microsoft.com/vs/community/)

Visual Studio 2012 has support for a few features that would be difficult to remove (some of C99).  The C support is not great in Visual Studio before 2013 or so, but I was able to make the code compile as C++ with 2012.

kyle-github, 2018-09-26:
Mesta automation wrote up a nice little post about it:

https://www.mesta-automation.com/how-to-communicate-to-an-allen-bradley-plc-with-c-and-libplctag-ethernet-ip-library/

kyle-github, 2019-10-05:
I found Microsoft's download page [here](https://visualstudio.microsoft.com/vs/older-downloads/), but I do not see VS Express or Community edition.  I do not own a VS subscription license and rely on the free versions.  Unfortunately, it will not let me download VS 2010.

I am going to close this for now, but I will keep trying to find a way to get it legally without spending a lot of $$$.   If you have 2010, I can try to create some changes in CMake to see if it works.

I already do some hacks in CMake to cause the library to compile as C++ in VS 2013.  That version does not support C99 (only 14 years after the spec was finalized!) fully.   So perhaps that would work for VS 2010?

closed 2019-10-05 by kyle-github