libplctag/libplctag #75: -6 (PLCTAG_ERR_CREATE)

FabioBentoLuiz, 2018-08-05:
Hi, 
I have a Windows service that connects to 3 PLCs to read and write tags (about 10 tags on each PLC).

The service has 3 threads (one for each PLC) and after some time running, I start to get status -6 (PLCTAG_ERR_CREATE) to create new tags. It is aleatory between the threads/PLCs and also between the tags (reading a tag sometimes succeeds sometimes not).

Which are the conditions to get this error?

I have to read all the tags constantly (1 or 2 seconds) to identify changes and here is how I'm reading a boolean based on the simple.c (same happens with int and dint):

```
plc_tag tag = plc_tag_create(tagAddress);
int rc;

   /* everything OK? */
   if (!tag) {
      protokoll(_T("Could not create tag: %s"), (CString)tagName);
      throw std::exception();
   }

   /* let the connect succeed */
   while (plc_tag_status(tag) == PLCTAG_STATUS_PENDING) {
      Sleep(100);
   }

   int status = plc_tag_status(tag);

  //here I'm getting -6 
   if (status != PLCTAG_STATUS_OK) {
      protokoll(_T("Error setting up tag internal state %s %d."), (CString)tagName, status);
      return false;
   }

   /* get the data */
   rc = plc_tag_read(tag, DATA_TIMEOUT);

   if (rc != PLCTAG_STATUS_OK) {
      protokoll(_T("RockwellConn::ReadBoolTag(): Error reading tag %s, status: %d."), (CString)tagName, rc);
      return false;
   }

   /* return the data */
   int b = plc_tag_get_uint8(tag, 0);

   plc_tag_destroy(tag);

   output = b;

   return true;
```

kyle-github, 2018-08-08:
Hi,

Thanks for using the library.

I am currently on vacation with very little access to the internet.

Can you add "&debug=3" to the end of your tag strings?  Then please capture the output and add it to the bug.  This will help determine the root cause of the problem.

If you are reading these tags every second or two, then you do not need to close the tag and recreate it between each read.

I will go through the code and note all the locations where that error can be returned.  That combined with the log file will help determine the problem and solution.

Best,
Kyle

FabioBentoLuiz, 2018-08-08:
Hi,
I thank you for sharing and for the support. 

The application is already installed in a client and I'll be back there in a few weeks.

I'll send the logs as soon as I have it. 

Enjoy your vacation.

BR,
Fabio

FabioBentoLuiz, 2018-08-27:
Hi Kyle,
here are the logs: https://www.dropbox.com/s/6rzexmd1dkqu9u6/libplctag.zip?dl=0 

I appreciate your help.

BR,
Fabio

FabioBentoLuiz, 2018-09-03:
Hi,
I went some steps further.

The main problem is that some times my Windows Service crashes and that is what I found in Event Viewer:

> Faulting module name: plctag.dll, version: 0.0.0.0, time stamp: 0x5b8479d1
> Exception code: 0xc0000005

In my logs I got just some -6 (PLCTAG_ERR_CREATE), so I thought it was related, but I thing I was wrong.

I don't think it is related [with this issue 73](https://github.com/kyle-github/libplctag/issues/73) as well because the service and all PLCs are in an LAN Gigabit Ethernet.

So instead of using libplctag as an external DLL I've embedded the source code in my application so I can debug all together in MS Visual Studio.

After some time I got the [following exception](https://www.dropbox.com/s/r35q7uezi0r65om/libplctag_crash.png?dl=0):

> req->session was nullptr

So I've changed the following on **eip.c** 

Original
```
/* set up the session sequence ID for this transaction */
      if (encap->encap_command == h2le16(AB_EIP_READ_RR_DATA)) {
         /* get new ID */
         req->session->session_seq_id++; 

         req->session_seq_id = req->session->session_seq_id;
         encap->encap_sender_context = req->session->session_seq_id; /* link up the request seq ID and the packet seq ID */

                                                                     /* mark the session as being used if this is a serialized packet */
                                                                     //~ mark_session_for_request(req);

         pdebug(DEBUG_INFO, "Sending unconnected packet with session sequence ID %llx", req->session->session_seq_id);
      }
```

Changed

```
/* set up the session sequence ID for this transaction */
      if (encap->encap_command == h2le16(AB_EIP_READ_RR_DATA)) {
         /* get new ID */
         if(!req->session)
         {
            pdebug(DEBUG_WARN, "Request session is null!");
            return PLCTAG_ERR_NULL_PTR;
         }

         req->session->session_seq_id++; //it seems that the crash occurs here

         req->session_seq_id = req->session->session_seq_id;
         encap->encap_sender_context = req->session->session_seq_id; /* link up the request seq ID and the packet seq ID */

                                                                     /* mark the session as being used if this is a serialized packet */
                                                                     //~ mark_session_for_request(req);

         pdebug(DEBUG_INFO, "Sending unconnected packet with session sequence ID %llx", req->session->session_seq_id);
      }
```

The application is now running for two days without problem (it used to happen at least every day). I'll continue monitoring for some more days, but it looks promising. 

@kyle-github , let me know your thoughts and if it makes any sense.

BR,
Fabio

kyle-github, 2018-09-03:
I looked through the logs again by thread.  It looks like there is something going wrong with the acquisition of a reference against the session object.   I see the session (I think) reference count mostly going up and up.  

I am trying to see if there is an error path that is taken that does not decrement the reference count. It looks like the counts might not be correct as I thought above.

FabioBentoLuiz, 2018-09-04:
Hi Kyle,

>  Do you have hundreds of tags?
No. Just around 10 on each PLC (at the moment 3 PLCs).

The workaround is stable until now.

I can access this environment just for some more days, so let me know if I can help somehow.

kyle-github, 2018-09-16:
I am trying to get a new test version available to see if that provides a solution.

kyle-github, 2018-10-11:
Can you please test the latest 1.6 version (check out the source if possible).  I have changed some things that may help here.

FabioBentoLuiz, 2018-10-25:
Hi,
I am not onsite anymore, so it will take a time until I can update and test again.

As soon as I can, I'll let you know the results.

BR,
Fabio

FabioBentoLuiz, 2019-01-21:
Ok Kyle. Still didn’t had the chance to upgrade. I’ll keep you posted.