libplctag/libplctag #76: Windows 64 bit compile

lorenzoviola, 2018-08-22:
Hello

I'd like to suggest an addition to the online documentation, about windows build.

I've followed the instructions to compile libplctag on windows 10, and on Win32 platform it works correctly.

I needed to compile it for Windows 64 bit , but VisualStudio 2015 complained with the following error:

LNK1112	module machine type 'x64' conflicts with target machine type 'X86'	

On the project linker properties TargetMachine is "MachineX64 (/MACHINE:X64)"

You need to set also Linker AdditionalOptions to "%(AdditionalOptions) /machine:X64" to make it compile.

Of course I get many :
Warning	C4244	'function': conversion from '__int64' to 'uint16_t', possible loss of data

How to correct that issue ?

thank you for support

kyle-github, 2018-08-24:
I have been on vacation the last few weeks.  I will try to look at this as soon as possible.  Unfortunately, I do not have 64-bit Windows.

The warnings should not be happening.  The fact that they are means that I am missing some flags in my Linux builds to catch them.  I will fiddle with the flags and see if I can find all the problem spots.  There are some areas where I cast from a pointer to intptr_t to one of the uintX_t types just to get a semi-random value, but this sounds different.

Can you attach a build log with the warning about conversion?

The X86 vs. X64 problem seems to be CMake related:

https://stackoverflow.com/questions/3563756/fatal-error-lnk1112-module-machine-type-x64-conflicts-with-target-machine-typ

Are you building the Visual Studio .sln files with CMake on the same machine you are using for compilation?

kyle-github, 2018-08-24:
I will need to find a 64-bit version of Windows and get a newer copy of Visual Studio to make sure I can solve this.

bmhh, 2018-09-06:
Hi, no problem.

You need to specify the platform in your cmake generator like so (example: VS2015)

cmake -G "Visual Studio 14 2015 Win64"

Then you can open the .sln solution and are able to build there.

A list of current generators:
https://cmake.org/cmake/help/v3.7/manual/cmake-generators.7.html

kyle-github, 2018-09-16:
By the way, according to the reporter of issue #79, using CMake from scratch on 64-bit Windows appears to work.

lorenzoviola, 2018-10-03:
Hello 

The library DLL is working in 64 bit

There are still some warnings, but maybe they are related to functions that will not give problems.

Looks like much better now !

Thank you and best regards !

closed 2018-10-03 by lorenzoviola