libplctag/libplctag #79: Reading from modern PLCs (LGX) via PCCC

bmhh, 2018-09-04:
Hi Kyle,

I've been trying to read a Tag from a modern PLC (1769-L32E CLGX), which is equipped with CIP via PCCC. It looks like the library does not use the UCMM when assembling a PCCC packet.
[image]

Debug output is as following(debug level 2):
thread(0001) 2018-08-04 09:13:33.508 WARN check_read_status:325 PCCC command failed, response code: 5
ERROR: Unable to read the data! Got error code -28: PLCTAG_ERR_REMOTE_ERR
thread(0001) 2018-08-04 09:13:33.510 WARN lib_teardown:121 Tag 0x55dc46192b70 at index 1 was not destroyed!

I can see in wireshark that the PLC never returns a response packet, just the request.

I did specify the path=1,0 variable, because as far as I know you need routing to the PLC backplane for modern PLCs.

Is this intended behaviour (no PCCC support for modern PLCs) or what might be happening?
I'm happy to provide further assistance.

By the way, 64-Bit Windows works fine when compiling with CMake.

kyle-github, 2018-09-16:
Hi,

Sorry I seem to have missed this question before now!   I have never tried to force the library to read a Logix-class PLC using PCCC.   Can you provide your initial tag creation string?   And debugging output with debug level 3 or 4? 

The library does support reading PLC 5 data from a PLC 5 using PCCC encapsulated in CIP/EIP.   It also supports reading a PLC 5 on DH+ through an Ethernet DH+ bridge (such as a DHRIO module in a ControlLogix).   The latter might be triggered when you have a path.  I'll have to look more closely at that code.

I am trying to remember how to decode the packet.  It looks like it is sending a read (0x0f CMD with 0x68 FNC) of... N7:0?  

Have you turned on the PLC 5 mapping in the ControlLogix?   I have never used that mode, but I seem to remember that there is some configuration you need to do on the ControlLogix side.

kyle-github, 2018-09-16:
Oh, and thanks for the CMake tip!  I still have not had a chance to set up Windows 10 64-bit.  I need to talk with a friend at Microsoft to get them to buy me a cheap copy of Windows.

bmhh, 2018-09-17:
Sure!
I enabled the PLC5 mapping in controllogix and another lib (which can solely communicate via PCCC) is working fine when I specify the path, so I don't think this is an issue with my PLC.

Here is the path, 

`TAG_PATH "protocol=ab_eip&gateway=192.168.1.11&path=1,0&cpu=PLC5&elem_size=2&elem_count=1&name=N7:0&debug=3"`

Yes, I am sending a read to the N7:0 address of my PLC.
You also guessed correctly that the path triggers the DH+ bridge.
```
thread(0001) 2018-08-17 10:11:31.466 INFO find_tag_create_func:88 Matched protocol=ab_eip
thread(0001) 2018-08-17 10:11:31.469 INFO ab_tag_create:192 Starting.
thread(0001) 2018-08-17 10:11:31.469 INFO rc_alloc_impl:99 Starting, called from ab_tag_create:199
thread(0001) 2018-08-17 10:11:31.469 INFO rc_alloc_impl:131 Done
thread(0001) 2018-08-17 10:11:31.469 INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
thread(0001) 2018-08-17 10:11:31.469 INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
thread(0001) 2018-08-17 10:11:31.469 INFO session_create_unsafe:455 Starting
thread(0001) 2018-08-17 10:11:31.469 INFO rc_alloc_impl:99 Starting, called from session_create_unsafe:459
thread(0001) 2018-08-17 10:11:31.469 INFO rc_alloc_impl:131 Done
thread(0001) 2018-08-17 10:11:31.469 INFO session_create_unsafe:514 Done
thread(0001) 2018-08-17 10:11:31.469 INFO session_init:530 Starting.
thread(0001) 2018-08-17 10:11:31.469 INFO session_connect:560 Starting.
thread(0001) 2018-08-17 10:11:31.471 INFO session_connect:580 Done.
thread(0001) 2018-08-17 10:11:31.471 INFO session_register:664 Starting.
thread(0001) 2018-08-17 10:11:31.471 INFO session_register:699 sending data:
thread(0001) 2018-08-17 10:11:31.471 INFO session_register:700 00000 65 00 04 00 00 00 00 00 00 00
thread(0001) 2018-08-17 10:11:31.471 INFO session_register:700 00010 00 00 00 00 00 00 00 00 00 00
thread(0001) 2018-08-17 10:11:31.471 INFO session_register:700 00020 00 00 00 00 01 00 00 00
thread(0001) 2018-08-17 10:11:31.479 INFO session_register:775 received response:
thread(0001) 2018-08-17 10:11:31.479 INFO session_register:776 00000 65 00 04 00 00 05 02 0c 00 00
thread(0001) 2018-08-17 10:11:31.479 INFO session_register:776 00010 00 00 00 00 00 00 00 00 00 00
thread(0001) 2018-08-17 10:11:31.479 INFO session_register:776 00020 00 00 00 00 01 00 00 00
thread(0001) 2018-08-17 10:11:31.479 INFO session_register:799 Done.
thread(0001) 2018-08-17 10:11:31.479 INFO session_init:545 Done.
thread(0001) 2018-08-17 10:11:31.480 INFO ab_tag_create:394 Done.
thread(0001) 2018-08-17 10:11:31.480 INFO plc_tag_create:307 Returning mapped tag 0x4001
thread(0001) 2018-08-17 10:11:31.480 INFO plc_tag_read:550 Starting.
thread(0001) 2018-08-17 10:11:31.480 INFO eip_pccc_tag_read_start:123 Starting
thread(0001) 2018-08-17 10:11:31.480 INFO rc_alloc_impl:99 Starting, called from request_create:52
thread(0001) 2018-08-17 10:11:31.480 INFO rc_alloc_impl:131 Done
thread(0001) 2018-08-17 10:11:31.480 INFO session_add_request_unsafe:830 Starting.
thread(0001) 2018-08-17 10:11:31.480 INFO session_add_request_unsafe:859 Total requests in the queue: 0
thread(0001) 2018-08-17 10:11:31.480 INFO session_add_request_unsafe:861 Done.
thread(0001) 2018-08-17 10:11:31.480 INFO eip_pccc_tag_read_start:260 Done.
thread(0002) 2018-08-17 10:11:31.481 INFO session_check_outgoing_data_unsafe:949 Readying unconnected packet to send.
thread(0002) 2018-08-17 10:11:31.481 INFO session_check_outgoing_data_unsafe:957 sending packet, so 1 unconnected requests in flight.
thread(0002) 2018-08-17 10:11:31.481 INFO send_eip_request_unsafe:66 Sending unconnected packet with session sequence ID 31055b22
thread(0002) 2018-08-17 10:11:31.481 INFO send_eip_request_unsafe:94 Sending packet of size 67
thread(0002) 2018-08-17 10:11:31.482 INFO send_eip_request_unsafe:95 00000 6f 00 2b 00 00 05 02 0c 00 00
thread(0002) 2018-08-17 10:11:31.482 INFO send_eip_request_unsafe:95 00010 00 00 22 5b 05 31 00 00 00 00
thread(0002) 2018-08-17 10:11:31.482 INFO send_eip_request_unsafe:95 00020 00 00 00 00 00 00 00 00 01 00
thread(0002) 2018-08-17 10:11:31.482 INFO send_eip_request_unsafe:95 00030 02 00 00 00 00 00 b2 00 1b 00
thread(0002) 2018-08-17 10:11:31.482 INFO send_eip_request_unsafe:95 00040 4b 02 20 67 24 01 07 3d f3 45
thread(0002) 2018-08-17 10:11:31.482 INFO send_eip_request_unsafe:95 00050 43 50 21 0f 00 20 5b 68 00 00
thread(0002) 2018-08-17 10:11:31.482 INFO send_eip_request_unsafe:95 00060 01 00 06 07 00 01 00
thread(0002) 2018-08-17 10:11:31.486 INFO recv_eip_response_unsafe:221 Received unconnected packet with session sequence ID 31055b22
thread(0002) 2018-08-17 10:11:31.486 INFO receive_response_unsafe:718 Packet sent initially 4ms ago and was sent 1 times
thread(0002) 2018-08-17 10:11:31.486 INFO receive_response_unsafe:723 got full packet of size 46
thread(0002) 2018-08-17 10:11:31.486 INFO receive_response_unsafe:724 00000 6f 00 16 00 00 05 02 0c 00 00
thread(0002) 2018-08-17 10:11:31.486 INFO receive_response_unsafe:724 00010 00 00 22 5b 05 31 00 00 00 00
thread(0002) 2018-08-17 10:11:31.486 INFO receive_response_unsafe:724 00020 00 00 00 00 00 00 00 00 01 00
thread(0002) 2018-08-17 10:11:31.486 INFO receive_response_unsafe:724 00030 02 00 00 00 00 00 b2 00 06 00
thread(0002) 2018-08-17 10:11:31.486 INFO receive_response_unsafe:724 00040 cb 00 05 01 00 00
[... cut ...]
thread(0001) 2018-08-17 10:11:31.491 WARN lib_teardown:121 Tag 0x558075f9e250 at index 1 was not destroyed!
thread(0001) 2018-08-17 10:11:31.493 INFO lib_teardown:125 Destroying global library mutex.
thread(0001) 2018-08-17 10:11:31.493 INFO lib_teardown:131 Done.
```

Looks like it is intended behaviour.
As far as I understand, if you want to communicate to a modern PLC via PCCC, one has to embed the 0x4b Execute-PCCC service into a 0x52 read data fragmented service with the IOI path specified (1,0 in my case). But I'm not entirely sure. It's not that critical, just something I found.

kyle-github, 2018-09-17:
Can you try without the path element?   That is how direct PLC 5 access works.   Having the path does trigger the DH+ code.   I may need to rethink that logic.   At the time I assumed that no one would want to access a ControlLogix via PCCC.   Why do you want to access it via PCCC?   

Do you have a pointer on how to set up PCCC access to a ControlLogix?  I have only read that it works, not how to do it. I have a ControlLogix so I can test this.

kyle-github, 2018-09-17:
Hmm, from the packet above, it looks like it is NOT using DH+.   But the packet looks like a correct packet for accessing a PLC 5.  The path is not in the packet.  The trailing bytes with 01 00 are the element count field if I am decoding the packet correctly.

bmhh, 2018-09-18:
Yes, it might have been without the PATH argument the first time I created this issue.

I think it's "just" a matter of using the Connection Manager object to point the PCCC command to the *Logix CPU. So, embedding the PCCC command into a 0x52 - unconnected send to the connection manager and append the IOI-path to the packet. (0x52 0x02 0x20 0x67 0x24 (more parameters) ... your packet ... (maybe padding) 0x01 0x00 0x01 0x00) <- sorry, bad mockup.

Nevertheless, here is a detailed log of what happens when I don't specify the CPU-Path=1,0 to the TAG_PATH in the program:

`thread(0001) 2018-08-18 09:28:03.393 INFO find_tag_create_func:88 Matched protocol=ab_eip
thread(0001) 2018-08-18 09:28:03.397 INFO ab_tag_create:192 Starting.
thread(0001) 2018-08-18 09:28:03.397 INFO rc_alloc_impl:99 Starting, called from ab_tag_create:199
thread(0001) 2018-08-18 09:28:03.397 INFO rc_alloc_impl:131 Done
thread(0001) 2018-08-18 09:28:03.397 INFO session_create_unsafe:455 Starting
thread(0001) 2018-08-18 09:28:03.397 INFO rc_alloc_impl:99 Starting, called from session_create_unsafe:459
thread(0001) 2018-08-18 09:28:03.398 INFO rc_alloc_impl:131 Done
thread(0001) 2018-08-18 09:28:03.398 INFO session_create_unsafe:514 Done
thread(0001) 2018-08-18 09:28:03.398 INFO session_init:530 Starting.
thread(0001) 2018-08-18 09:28:03.398 INFO session_connect:560 Starting.
thread(0001) 2018-08-18 09:28:03.401 INFO session_connect:580 Done.
thread(0001) 2018-08-18 09:28:03.401 INFO session_register:664 Starting.
thread(0001) 2018-08-18 09:28:03.401 INFO session_register:699 sending data:
thread(0001) 2018-08-18 09:28:03.401 INFO session_register:700 00000 65 00 04 00 00 00 00 00 00 00
thread(0001) 2018-08-18 09:28:03.401 INFO session_register:700 00010 00 00 00 00 00 00 00 00 00 00
thread(0001) 2018-08-18 09:28:03.401 INFO session_register:700 00020 00 00 00 00 01 00 00 00
thread(0001) 2018-08-18 09:28:03.408 INFO session_register:775 received response:
thread(0001) 2018-08-18 09:28:03.408 INFO session_register:776 00000 65 00 04 00 2b 55 02 0c 00 00
thread(0001) 2018-08-18 09:28:03.408 INFO session_register:776 00010 00 00 00 00 00 00 00 00 00 00
thread(0001) 2018-08-18 09:28:03.408 INFO session_register:776 00020 00 00 00 00 01 00 00 00
thread(0001) 2018-08-18 09:28:03.408 INFO session_register:799 Done.
thread(0001) 2018-08-18 09:28:03.408 INFO session_init:545 Done.
thread(0001) 2018-08-18 09:28:03.408 INFO ab_tag_create:394 Done.
thread(0001) 2018-08-18 09:28:03.408 INFO plc_tag_create:307 Returning mapped tag 0x4001
thread(0001) 2018-08-18 09:28:03.408 INFO plc_tag_read:550 Starting.
thread(0001) 2018-08-18 09:28:03.408 INFO eip_pccc_tag_read_start:123 Starting
thread(0001) 2018-08-18 09:28:03.408 INFO rc_alloc_impl:99 Starting, called from request_create:52
thread(0001) 2018-08-18 09:28:03.408 INFO rc_alloc_impl:131 Done
thread(0001) 2018-08-18 09:28:03.408 INFO session_add_request_unsafe:830 Starting.
thread(0001) 2018-08-18 09:28:03.408 INFO session_add_request_unsafe:859 Total requests in the queue: 0
thread(0001) 2018-08-18 09:28:03.408 INFO session_add_request_unsafe:861 Done.
thread(0001) 2018-08-18 09:28:03.408 INFO eip_pccc_tag_read_start:260 Done.
thread(0002) 2018-08-18 09:28:03.408 INFO session_check_outgoing_data_unsafe:949 Readying unconnected packet to send.
thread(0002) 2018-08-18 09:28:03.408 INFO session_check_outgoing_data_unsafe:957 sending packet, so 1 unconnected requests in flight.
thread(0002) 2018-08-18 09:28:03.408 INFO send_eip_request_unsafe:66 Sending unconnected packet with session sequence ID 3dad11d7
thread(0002) 2018-08-18 09:28:03.408 INFO send_eip_request_unsafe:94 Sending packet of size 67
thread(0002) 2018-08-18 09:28:03.408 INFO send_eip_request_unsafe:95 00000 6f 00 2b 00 2b 55 02 0c 00 00
thread(0002) 2018-08-18 09:28:03.408 INFO send_eip_request_unsafe:95 00010 00 00 d7 11 ad 3d 00 00 00 00
thread(0002) 2018-08-18 09:28:03.408 INFO send_eip_request_unsafe:95 00020 00 00 00 00 00 00 00 00 01 00
thread(0002) 2018-08-18 09:28:03.408 INFO send_eip_request_unsafe:95 00030 02 00 00 00 00 00 b2 00 1b 00
thread(0002) 2018-08-18 09:28:03.408 INFO send_eip_request_unsafe:95 00040 4b 02 20 67 24 01 07 3d f3 45
thread(0002) 2018-08-18 09:28:03.408 INFO send_eip_request_unsafe:95 00050 43 50 21 0f 00 d5 11 68 00 00
thread(0002) 2018-08-18 09:28:03.408 INFO send_eip_request_unsafe:95 00060 01 00 06 07 00 01 00
thread(0002) 2018-08-18 09:28:03.412 INFO recv_eip_response_unsafe:221 Received unconnected packet with session sequence ID 3dad11d7
thread(0002) 2018-08-18 09:28:03.413 INFO receive_response_unsafe:718 Packet sent initially 4ms ago and was sent 1 times
thread(0002) 2018-08-18 09:28:03.413 INFO receive_response_unsafe:723 got full packet of size 46
thread(0002) 2018-08-18 09:28:03.413 INFO receive_response_unsafe:724 00000 6f 00 16 00 2b 55 02 0c 00 00
thread(0002) 2018-08-18 09:28:03.413 INFO receive_response_unsafe:724 00010 00 00 d7 11 ad 3d 00 00 00 00
thread(0002) 2018-08-18 09:28:03.413 INFO receive_response_unsafe:724 00020 00 00 00 00 00 00 00 00 01 00
thread(0002) 2018-08-18 09:28:03.413 INFO receive_response_unsafe:724 00030 02 00 00 00 00 00 b2 00 06 00
thread(0002) 2018-08-18 09:28:03.413 INFO receive_response_unsafe:724 00040 cb 00 05 01 00 00
thread(0001) 2018-08-18 09:28:03.414 WARN check_read_status:325 PCCC command failed, response code: 5
thread(0001) 2018-08-18 09:28:03.414 INFO refcount_cleanup:317 Starting
[... cut ...]
thread(0001) 2018-08-18 09:28:03.414 WARN lib_teardown:121 Tag 0x562f2926c1f0 at index 1 was not destroyed!
thread(0001) 2018-08-18 09:28:03.416 INFO lib_teardown:125 Destroying global library mutex.
thread(0001) 2018-08-18 09:28:03.416 INFO lib_teardown:131 Done.`

As for how to enable PCCC access to a controlLogix, you just need to create a controller-scoped variable (for simplicity an INT-Array), then go to the logic tab -> MAP PLC5/SLC variables, select file number 7 and for the name you just enter the name of your int-Array.

kyle-github, 2018-09-21:
I am still struggling with this.  Can you get a packet capture from the library that works?   There must be something I am setting incorrectly in the packet :-/

Thanks.

bmhh, 2018-09-24:
Sure thing, kyle.

Two types of how this could work:
One: with Unconnected Send:
Req:
[image]
Resp:
[image]

Two: Connected message
Req:
[image]
Resp:
[image]

Table 5 of " Communicating with RA Products Using EtherNet/IP Explicit Messaging
(Rev. 1.2), Appendix C " should give you a detailled explanation of what the PLC expects.

kyle-github, 2018-09-26:
I think I have that document somewhere.  I'll look at it.  The only thing that I can see is that the encoded name has all the fields rather than being somewhat compressed.  

I will have to dig through this for a while.  It looks right to me, but it is not working.

Thanks for the bug report.

bmhh, 2018-10-12:
Hi Kyle,

I was able to fix the problem, however, I needed to modify the code (tried to interfere as little as possible).
in cip.c in the cip_encode_path(...) - function. Whenever a route-path is detected, but it is **not** a DHP path and the CPU is **not** a Micro-CPU, I set a variable 'use_ucmm' (similar to 'use_dhp'). If that variable is checked,
I create a PCCC-request **with** the UCMM wrapped around in order to route it to the correct CPU, if not, nothing changes and a standard-PCCC packet is send to the PLC.
Responses are parsed correctly, but I have only implemented a read so far. This is how the log-output looks now!

`thread(0001) 2018-09-12 10:46:05.004 INFO find_tag_create_func:88 Matched protocol=ab_eip
thread(0001) 2018-09-12 10:46:05.005 INFO ab_tag_create:193 Starting.
thread(0001) 2018-09-12 10:46:05.005 INFO rc_alloc_impl:99 Starting, called from ab_tag_create:200
thread(0001) 2018-09-12 10:46:05.005 INFO rc_alloc_impl:131 Done
thread(0001) 2018-09-12 10:46:05.005 INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
thread(0001) 2018-09-12 10:46:05.005 INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
thread(0001) 2018-09-12 10:46:05.005 INFO cip_encode_path:283 No DH+ Path and no Micro800 detected, falling back to UCMM-packets
thread(0001) 2018-09-12 10:46:05.005 INFO session_create_unsafe:455 Starting
thread(0001) 2018-09-12 10:46:05.005 INFO rc_alloc_impl:99 Starting, called from session_create_unsafe:459
thread(0001) 2018-09-12 10:46:05.005 INFO rc_alloc_impl:131 Done
thread(0001) 2018-09-12 10:46:05.005 INFO session_create_unsafe:514 Done
thread(0001) 2018-09-12 10:46:05.005 INFO session_init:530 Starting.
thread(0001) 2018-09-12 10:46:05.005 INFO session_connect:560 Starting.
thread(0001) 2018-09-12 10:46:05.010 INFO session_connect:580 Done.
thread(0001) 2018-09-12 10:46:05.010 INFO session_register:664 Starting.
thread(0001) 2018-09-12 10:46:05.010 INFO session_register:699 sending data:
thread(0001) 2018-09-12 10:46:05.010 INFO session_register:700 00000 65 00 04 00 00 00 00 00 00 00
thread(0001) 2018-09-12 10:46:05.010 INFO session_register:700 00010 00 00 00 00 00 00 00 00 00 00
thread(0001) 2018-09-12 10:46:05.010 INFO session_register:700 00020 00 00 00 00 01 00 00 00
thread(0001) 2018-09-12 10:46:05.016 INFO session_register:775 received response:
thread(0001) 2018-09-12 10:46:05.016 INFO session_register:776 00000 65 00 04 00 07 c2 02 0c 00 00
thread(0001) 2018-09-12 10:46:05.016 INFO session_register:776 00010 00 00 00 00 00 00 00 00 00 00
thread(0001) 2018-09-12 10:46:05.016 INFO session_register:776 00020 00 00 00 00 01 00 00 00
thread(0001) 2018-09-12 10:46:05.016 INFO session_register:799 Done.
thread(0001) 2018-09-12 10:46:05.016 INFO session_init:545 Done.
thread(0001) 2018-09-12 10:46:05.016 INFO ab_tag_create:394 Done.
thread(0001) 2018-09-12 10:46:05.017 INFO plc_tag_create:306 Returning mapped tag 0x4001
thread(0001) 2018-09-12 10:46:05.017 INFO plc_tag_read:549 Starting.
thread(0001) 2018-09-12 10:46:05.017 INFO eip_pccc_tag_read_start_ucmm:302 Starting
thread(0001) 2018-09-12 10:46:05.017 INFO rc_alloc_impl:99 Starting, called from request_create:52
thread(0001) 2018-09-12 10:46:05.017 INFO rc_alloc_impl:131 Done
thread(0001) 2018-09-12 10:46:05.017 INFO session_add_request_unsafe:830 Starting.
thread(0001) 2018-09-12 10:46:05.017 INFO session_add_request_unsafe:859 Total requests in the queue: 0
thread(0001) 2018-09-12 10:46:05.017 INFO session_add_request_unsafe:861 Done.
thread(0001) 2018-09-12 10:46:05.017 INFO eip_pccc_tag_read_start_ucmm:471 Done.
thread(0002) 2018-09-12 10:46:05.017 INFO session_check_outgoing_data_unsafe:949 Readying unconnected packet to send.
thread(0002) 2018-09-12 10:46:05.017 INFO session_check_outgoing_data_unsafe:957 sending packet, so 1 unconnected requests in flight.
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:66 Sending unconnected packet with session sequence ID 68fac528
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:94 Sending packet of size 82
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:95 00000 6f 00 3a 00 07 c2 02 0c 00 00
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:95 00010 00 00 28 c5 fa 68 00 00 00 00
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:95 00020 00 00 00 00 00 00 00 00 01 00
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:95 00030 02 00 00 00 00 00 b2 00 2a 00
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:95 00040 52 02 20 06 24 01 0a 05 1b 00
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:95 00050 4b 02 20 67 24 01 07 3d f3 45
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:95 00060 43 50 21 0f 00 26 c5 68 00 00
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:95 00070 01 00 06 07 00 01 00 00 01 00
thread(0002) 2018-09-12 10:46:05.017 INFO send_eip_request_unsafe:95 00080 01 00
thread(0002) 2018-09-12 10:46:05.026 INFO recv_eip_response_unsafe:221 Received unconnected packet with session sequence ID 68fac528
thread(0002) 2018-09-12 10:46:05.026 INFO receive_response_unsafe:718 Packet sent initially 9ms ago and was sent 1 times
thread(0002) 2018-09-12 10:46:05.026 INFO receive_response_unsafe:723 got full packet of size 61
[... cut ...]
thread(0001) 2018-09-12 10:46:05.034 INFO ab_teardown:166 Terminating IO thread.
thread(0001) 2018-09-12 10:46:05.078 INFO ab_teardown:174 Freeing global session mutex.
thread(0001) 2018-09-12 10:46:05.078 INFO ab_teardown:178 Removing the read group vector.
thread(0001) 2018-09-12 10:46:05.078 INFO ab_teardown:181 Done.
thread(0001) 2018-09-12 10:46:05.078 INFO lib_teardown:114 Tearing down library.
thread(0001) 2018-09-12 10:46:05.079 INFO lib_teardown:124 Destroying global library mutex.
thread(0001) 2018-09-12 10:46:05.079 INFO lib_teardown:130 Done.`

kyle-github, 2018-10-12:
Wow, thanks!

Can you post the patched cip_encode_path() function?

I think I see the problem here.  We need to have the path part because it is a Logix-class system, but we also need the PCCC part.   In the existing logic, those are exclusive.   Except in the case of DH+.   I think I see how I might work this.   What I really want is to make the PLC CPU type dominant and then trigger the DH+ only if there is actually that in the path.   Unfortunately, the DH+ encoding is an exception to the standard PCCC-within-CIP encoding rules, but only for PLC/5 targets :-(   AB really wants to make me work for this!

In general I am moving to use connected messaging where possible.   While it has overhead on connection set up, the resulting packets are slightly smaller and there is more guarantee that buffer resources will exist for the request and response.  Connected messaging has lower throughput in packets per second, however, it helps a lot with packing multiple requests into one packet.

One long term change I am considering is going to true CSP/PCCC packets (different protocol wrapper) completely for direct PLC/5, SLC and MicroLogix access.   That would be more flexible and cover more systems.   However, I need to get the base framework stable for multiple request packing first.  It is a 2.0 feature.

This gives me a very good clue of where to change things.  Thank you very much!

It might be a few days as I am out for the whole weekend and we are near a deadline at work.

bmhh, 2018-10-17:
Hi Kyle,

haha yes. With modern CIP, everything is easier.
I made a PR to show the changes, it's more of a patch than a rework of the logic-rework you mentioned, and I only implemented the read-function and tested int and floats. Not sure how booleans are handled here, since you can access them bitwise.

kyle-github, 2018-10-21:
Phwew.

I tried many different attempts to see if I could have PLC/5 and Logix work with the same packets.  It just does not seem to work.   Thanks you AB.

I did not take your pull request, but I took your ideas and used them in a different way.  Instead of trying to determine via the path whether to use the special PCCC-in-CIP mode, I created a new CPU type.

Now you use tag creation strings like this:

`protocol=ab-eip&gateway=10.206.1.39&path=1,5&cpu=lgxpccc&elem_size=4&elem_count=1&name=F8:0&debug=4`

The new cpu type is `lgxpccc`.  

I was able to test it reading and writing INT tags and reading REAL tags.

Can you try the new beta release for 1.5.13?

Thank you so much for this effort!  I never would have guessed that you had to wrap the PCCC packets the way that you did.

kyle-github, 2018-10-22:
This has been ported to the 1.6 branch as well.

Please let me know if this works for you!

bmhh, 2018-10-23:
I can confirm your test-results:
INT read/write works fine.
REAL read/write seems to be working fine aswell.
Looks like BOOL does not work R/W, but this can be a bit tricky, since you could just read the corresponding uint8 and then parse the right Bit B3:0/**1** into the return value, or do it another way.

kyle-github, 2018-10-23:
BOOLs are painful.   I do not officially support PLC/5-style BOOLs for a reason or BOOL arrays in Logix-class PLCs for a reason.   They are inconsistent and difficult to handle with simple read/write operations.  You really need to use read-modify-write operations instead.  I have opened issue #82 to track this.

Which version of the library did you test?  1.5.13 or 1.6.2?

Thanks again for finding this and providing a solution!

kyle-github, 2018-12-14:
This is in the 1.5 branch now so I will close this.  Thanks for the push and help!

closed 2018-12-14 by kyle-github