libplctag/libplctag #80: RS Logix 5000 - reading too many tags causes errors in INT (2 byte) reads.

howchmo, 2018-09-26:
To test performance, I wrote a test program to read multiple tags from a RS Logix 5000 processor.  There is a mix of BOOL (size 1) and INT (size 2) values.  When I run to read just the BOOL values, I get good values.  If I attempt to read both sets the BOOL's get set to -96 and I get the proper values for the INT's.  I did a test yesterday and it WAS working but only for 54 boolean tags and 4 int tags.  It is unusually inconsistent, and while I have not been able to totally characterize the bug, I felt I should report the issue.

kyle-github, 2018-09-28:
I am a little confused by what you are testing.  Can you share the program?

What is the model and version of the cpu and firmware?

I can try to set up a similar test and see if I can replicate the results.

Thanks for the report!

howchmo, 2018-10-02:
The program reads tags from CSV files. It is on a new Allen-Bradley L8 (version 31 of the firmware, I believe). I'll try to get more detailed information when I am back in the lab.  It is not too different from the samples you have.  The only difference is that it uses the type of the tag in the CSV to determine how to encode the tag.  Maybe I can post it via gist...

kyle-github, 2018-10-03:
Looking through this now.  Thanks for the gist.

Are you using the master branch of the library?

kyle-github, 2018-10-03:
Huh, I had forgotten just how weird booleans are in this protocol.

Are the BOOLs in an array?  If so, things get weird.

I tweaked your gist until I got it compiling on Linux.  I assume that this code was chopped out of something else because I am not sure it would have compiled since there were syntax errors :-)

I got it working and tried it against about 90 tags in a CSV file.   I was seeing 11-12ms per tag read.  All reads were identical.

I tweaked it so that the reads are done asynchronously (csv_tag_linux2.c) and that runs about 5.5ms per tag read.

Unless I am doing something really wrong, I am not seeing what you are seeing :-(

[csv_tag_linux.zip](https://github.com/kyle-github/libplctag/files/2440306/csv_tag_linux.zip)

kyle-github, 2018-10-03:
If you are reading BOOL arrays, be aware that you get at least 32 values at once.   All BOOL arrays are rounded up to the nearest multiple of 32 bits (a DINT).   So, if you have a BOOL[100] array, it will be stored in four 32-bit values.   And, when reading with the library, you will need to get the right element count (4), not 100 or 128 or any other value that would seem to make sense.

Individual BOOL values (when not in an array) come back as single bytes and are 0 or 0xFF in value.   Usually.  If a BOOL value is in a UDT, then there are packing rules that apply that are different from the way that arrays are treated.

howchmo, 2018-10-03:
Thanks so much for this.  Here is my output when reading a mix of BOOL and INT tags:
"
It took 109.000000 ms to read 14 tags for an average of 7.785714 ms/tag
TagBool0 : -96
TagBool1 : -96
TagBool2 : -96
TagBool3 : -96
TagBool4 : -96
TagBool5 : -96
TagBool6 : -96
TagBool7 : -96
TagBool8 : -96
TagBool9 : -96
TagInt10 : 4000
TagInt11 : 4000
TagInt12 : 4000
TagInt13 : 4000
"

howchmo, 2018-10-03:
I'll try this on a Linux virtual machine at some point.  I am building in Visual Studio 2017.
Another bit of weirdness.  If I read less than 12 BOOL tags, I get errors.  If I read over 12 BOOL tags, I get data.  Sometimes I get good data in the INT's and sometimes I get good data in the BOOL's depending on the number of tags I am reading.  Let me try a destroy between reads.

howchmo, 2018-10-03:
So, I pulled latest from the master and I rebuilt.  I did see some warning about int conversions:

1>------ Rebuild All started: Project: ZERO_CHECK, Configuration: Debug x64 ------
1>Checking Build System
1>CMake does not need to re-run because C:/Users/703099162/ATI/myrad/EIP_APIs/libplctag/build/CMakeFiles/generate.stamp is up-to-date.
2>------ Rebuild All started: Project: plctag, Configuration: Debug x64 ------
2>Building Custom Rule C:/Users/703099162/ATI/myrad/EIP_APIs/libplctag/CMakeLists.txt
2>CMake does not need to re-run because C:/Users/703099162/ATI/myrad/EIP_APIs/libplctag/build/CMakeFiles/generate.stamp is up-to-date.
2>init.c
2>libplctag_tag.c
2>ab_common.c
2>cip.c
2>connection.c
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\connection.c(445): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\connection.c(453): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\connection.c(481): warning C4244: '=': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\connection.c(526): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\connection.c(534): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\connection.c(562): warning C4244: '=': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\connection.c(812): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\connection.c(820): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\connection.c(839): warning C4244: '=': conversion from '__int64' to 'int', possible loss of data
2>eip.c
2>eip_cip.c
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(597): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(600): warning C4244: '=': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(718): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(733): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(736): warning C4244: '=': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(854): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(857): warning C4244: '=': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(999): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(1014): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(1017): warning C4244: '=': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(1223): warning C4244: 'function': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(1227): warning C4244: '=': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(1239): warning C4244: '+=': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(1465): warning C4244: 'function': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(1469): warning C4244: '=': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_cip.c(1481): warning C4244: '+=': conversion from '__int64' to 'int', possible loss of data
2>eip_dhp_pccc.c
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_dhp_pccc.c(194): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_dhp_pccc.c(210): warning C4244: '=': conversion from '__int64' to 'int', possible loss of data
2>c:\users\703099162\ati\myrad\eip_apis\libplctag\src\protocols\ab\eip_dhp_pccc.c(374): warning C4244: 'function': conversion from '__int64' to 'uint16_t', possible loss of data
[... cut ...]
14>------ Rebuild All started: Project: ALL_BUILD, Configuration: Debug x64 ------
14>Building Custom Rule C:/Users/703099162/ATI/myrad/EIP_APIs/libplctag/CMakeLists.txt
14>CMake does not need to re-run because C:/Users/703099162/ATI/myrad/EIP_APIs/libplctag/build/CMakeFiles/generate.stamp is up-to-date.
15>------ Skipped Rebuild All: Project: INSTALL, Configuration: Debug x64 ------
15>Project not selected to build for this solution configuration 
========== Rebuild All: 14 succeeded, 0 failed, 1 skipped ==========

howchmo, 2018-10-03:
I am wondering if there is where possible problems might be introduced on a Windows build.  I will run on Linux within a week.

kyle-github, 2018-10-03:
Thanks very much for posting that.  I am in the process of trying to solve all these conversion warnings.  Unfortunately, GCC/Clang seem to have an "all or nothing" approach to conversion warnings.

Can you send me the CSV file and I will try to make an example run that is exactly the same?

Are any of the booleans in arrays?

howchmo, 2018-10-05:
No booleans in arrays.  I'll put together a test CSV for you.

kyle-github, 2018-10-05:
Thanks.  Please let me know the exact types of each tag.  I will duplicate
them locally and read.

I am still trying to get an L81 CPU.  For some strange reason no one seems
to want to let me borrow one.   I can't imagine why since they cost about
the same as a used car :-/

howchmo, 2018-10-07:
I am going to try and test this on an L72.

kyle-github, 2018-10-07:
Thanks!  Let me know.  I have an L61 and an L55.  Still trying to get an L81.

kyle-github, 2018-10-08:
I just released a beta version of 1.6.0.: https://github.com/kyle-github/libplctag/releases/tag/v1.6.0

I would appreciate it if you could give that a try.   I am very interested to see how it does on the newer ControlLogix firmware.  

Thanks.

howchmo, 2018-10-11:
Just came back to this and I had a little trouble getting 1.6.0 built for Visual Studio 2017.  I had to change a couple of things:

1. Change the following line CMakeList.txt:
`         target_link_libraries(plctag_static ws2_32)`

This allowed CMake to configure and generate correctly.

2. Add the following line to libplctag.h:
`          LIB_EXPORT plc_tag plc_tag_create_sync(const char *attrib_str, int timeout);`

3. Comment out the following line from lib.c:
`           // static plc_tag plc_tag_create_sync(const char *attrib_str, int timeout);`

These allowed the library to build correctly.  I am testing it now against a L8.

howchmo, 2018-10-11:
I tested the new DLL against an L8.  I have not tested it against an L72, but on the L8 (if I try to read a mix of BOOL and INT) the INT values are not correct.  If I read just the INT values they are correct.  For BOOL's, everything seems to be working correctly.  I am reading everything in a loop.

howchmo, 2018-10-11:
I'll test on an L7 tonight.

kyle-github, 2018-10-11:
Hmm, not good.   I was trying to set up support for static builds, but I thought I got it all removed again.  Clearly I did not!   I am working with someone I know at Microsoft to get a copy of 64-bit Windows 10.

Actually, I am trying to NOT export plc_tag_create_sync().  That was the start of the work toward the 2.0 API where plc_tag_create() takes a timeout like read and write.

Thanks for testing this!

howchmo, 2018-10-12:
No problem.  I got it built.  I just ran my same program against 1024 tags (1000 BOOL's and 24 DINT's) on an L7 and I had no issues.  (It read them all in under 7 seconds.)  This appears to be an L8 problem.

It is just a weird problem because if I set my test program just to read the BOOL's or just to read the DINT's it works fine.

kyle-github, 2018-10-12:
Yuck.

Can you provide an output dump?  So this only happens when you have multiple tags created and you call read on them all sequentially.

I am still trying to figure out how this could happen...

Can you set "debug=4" in the tag string and capture the output from the L8?   I keep wondering if there is an error condition that is being hit but I am not catching it.

Thanks so much for working on this!

kyle-github, 2018-10-12:
Some other questions:

- what version is the firmware on each PLC?   I know the L8s must be fairly new.  Are the L7s equally up to date?
- what kind of L7 is this?   ControlLogix or CompactLogix?
- what kind of Ethernet module is the L7 using?
- if it is separate (ControlLogix), what version of the firmware is it using?

Perhaps there are some clues there.   

Hmm, I wonder if the problem is the negotiation for the large packet size?   The L8 will almost certainly support large packets, but the L7 might not.  Or perhaps the L8 supports even larger packets than I support.

kyle-github, 2018-10-12:
I made another update to the 1.6 branch.   In this update I tried to fix the Windows compilation problems you ran into and I forced the negotiated packet size to 508 bytes on both old and newer PLCs.  

If you get a chance, I would appreciate it if you could download the latest code from the 1.6 branch and give that a try.

Thanks.

howchmo, 2018-10-16:
It appears to have worked!  I have run my test program that has 1024 tags with BOOL, INT, DINT, and REAL against a 1756-L72S GuardLogix 5570 Safety Controller and a 1756-L82ES GuardLogix 5580 Safety Controller.  Everything is reading correctly.

I also created a demo using Windows threading to see if I could improve read performance inspired by your PLC5 pthreads example.  I do wonder if with a lot of tags and a need for fast response times, if I should be looking at a solution where my application becomes a Produce/Consume target.  Hmmm...

howchmo, 2018-10-16:
Thanks for the fix!

kyle-github, 2018-10-16:
Hmm, that is interesting.   I suspect that the large buffers will work, but there must be something I am doing wrong somewhere in the set up.  I tested it out on a ControlLogix L6x with firmware version 24 (I think?) and an updated ENBT module.  On that, I could negotiate 4000 byte packets without a problem.

For performance, there is a bit of a regression in the 1.6 line.  I used to allow multiple requests in flight at the same time.   However, in some circumstances, retries were necessary (handled by the library) and there were some users who ran into edge cases where the retries could set up a slowly collapsing condition where the retries caused the next requests to time out and retry which then caused...

I am working toward supporting multiple request packets.   In a single packet, CIP allows you to pack in multiple requests.  So if you are reading a lot of small tags, you can get a huge boost in performance with the multiple request packing.   That said, I was quite pleased with the results I was getting with the large packets.   

I will keep the packet size small until I have access to a later version of firmware (I only have version 20 at most here).

Thanks for your patience and help in debugging this problem!

kyle-github, 2018-10-16:
Sorry, I did not respond to your other point.  Produced/consumed tags can really help if you need a high data rate.  However the library does not support them yet.   I think there are some commercial libraries that might.  It is on my list once 2.0 is out.  It really requires a different structure in the library to support them.  I really want to support them though because they are so much nicer for direct control and telemetry than explicit messaging.

I have worked on a fairly large control project where we used produced/consumed tags for communication between multiple PLCs and it really helped.   Explicit messaging is very heavy.   We had trouble maintaining one packet per 500ms between the PLCs and had some convoluted code to slow down the packet rates when nothing was moving to keep delays down.  We moved to produced/consumed tags and now send a packet every 200ms between all the PLCs without a problem.   Interestingly, the data loss rate went down.   The network in question is fairly large in physical size (approximately 1000 feet by 1000 feet) and has some challenging elements like mobile waveguide-based network end points.

kyle-github, 2018-12-05:
Using the 2.0 code, against a L81, I get the following:

Read 300 tags from the config file.
Read 300 tags in 159ms
Read 300 tags in 111ms
Read 300 tags in 107ms
Read 300 tags in 109ms
Read 300 tags in 111ms
Read 300 tags in 100ms
Read 300 tags in 95ms
Read 300 tags in 92ms
Read 300 tags in 100ms
Read 300 tags in 92ms
Read 300 tags in 98ms
Read 300 tags in 81ms
Read 300 tags in 94ms
Read 300 tags in 109ms
Read 300 tags in 112ms
Read 300 tags in 99ms
Read 300 tags in 98ms
Read 300 tags in 94ms

So far, I am not seeing any bad data.

I rewrote my data_dumper example to be a lot more like your CSV reader.  The new program takes a tab-delimited input file with one line per tag.  It reads the tags at the specified interval (per tag) and then logs the data in an output file that is CSV. 

This provides a pretty good test.   So far, it worked for two days without problems.   As you can see above, my network is pretty flaky.   There is a lot of variation in latency.

howchmo, 2018-12-05:
These are good results.  I think what might be interesting is to see if there is a model for tuning the response time.  If I want to have a 100ms response time, I use 300 tags (X number of bytes).  If I want a 40ms response time, I limit to Y number of tags (Z number of bytes).

kyle-github, 2018-12-06:
The code I am using already handles the timing.   I attached it below.

Each row in the config file is for a single tag.   Each tag has a requested interval for reading.   It is not exact.   The sample above was using an interval of 100ms, so much of the time, it just went from one iteration to the next.

The total number of bytes does not seem to matter too much.   As long as the tags are relatively small, at least.

Note this code is for the 2.0 version of the library, so the API for plc_tag_create() is a little different.  I may port it back to the 1.5 version of the library.  However, it would be a lot less useful there.   If you need it for some reason, let me know.   This is just an example program so it would probably need to be tweaked a bit to be fully usable.   

I am currently using it as a stress test for the 2.0 library. I've already found some bugs :-)

[data_dumper.zip](https://github.com/kyle-github/libplctag/files/2651595/data_dumper.zip)

kyle-github, 2018-12-26:
Please reopen or open another bug if this happens again!   Version 2.0 has been released and should have most of this fixed.

closed 2018-12-26 by kyle-github