libplctag/libplctag #81: Wapper to C#

franklupo, 2018-10-05:
Hi,

Excellent and incredible work. I created a wrapper in .Net C# https://github.com/Corsinvest/cv4ab-api-dotnet and deployed in nuget.
I am still amazed at how well it works and how fast it is.

Thanks again

Best regards

franklupo, 2018-10-07:
Hi,
i want to maintain it separately

best regards

kyle-github, 2018-10-08:
Hi,

I just released version 1.6.0.  It has some new API functions.   I would very much appreciate it if you could test it with any hardware you have!

kyle-github, 2018-12-02:
I have linked to your C# repo and mentioned Nuget in the Welcome wiki page.

I will be releasing 2.0 within a few weeks.  I would appreciate it if you could test as soon as I release the beta version as the C API changed slightly.

Again, thanks for your work and for using the library!

kyle-github, 2018-12-17:
I just released the first Beta of 2.0.   The API is stable and I have tested it on Windows and Linux.   The Java and Python wrappers have been ported and updated.

franklupo, 2019-01-22:
Hi,
I published new version using 2.0.10.

Best regards

kyle-github, 2019-01-22:
Thanks!   Can I close this?

closed 2019-01-22 by franklupo