libplctag/libplctag #82: Library has poor support for boolean values

kyle-github, 2018-10-11:
The library makes it appear that tags like B3:4/1 or flagArray[6] will work to read/write boolean values.   However, except under limited circumstances it will not.

For PCCC/CSP/DF1-based systems (PLC/5, SLC, MicroLogix), the bit index cannot be passed in the PCCC command.   What is sent and returned is a 16-bit word.  The library should handle this case and do the bit twiddling necessary to work with the correct bit.

For CIP-based systems (Control/Compact Logix, Micr800), there is a very large difference in behavior depending on whether the boolean value is an independent tag, part of a BOOL array or in a UDT.

For the case that the tag is an independent tag, then the value is sent and received as a single byte.  This works.   

For the case that the tag points to a single element of a BOOL array, the array segment is sent back as a 32-bit word and the user must pull out the right bit.   What makes this particularly annoying is that the array index, i.e the 6 in flagArray[6], will be passed, will not generate an error, and will be completely ignored by the PLC.  In this case the library should determine that the tag type is a boolean array and behave accordingly.

For the case that the tag is part of a UDT, it depends on whether the tag is a stand alone, independent, BOOL value or part of an array of BOOLs.  If it is stand alone, it will still be packed into an 8-byte array of bits (or more than one if there are more than 8 BOOLs).   The packing rules are arcane and it may simply not be feasible to cover this in the library for a while.

To support this behavior, an API change will be required so this is best handled as part of 2.0.   The user will need to pass the expected tag type.   For instance:

- BOOL - stand alone boolean
- BOOL[100] as part of an array
- BIT (as part of a larger word)
- SINT
- INT
- DINT
- LINT
- array variations on the above...
...

Or perhaps the library can query the PLC (possible on ControlLogix but not obviously supported on anything else)?

For PCCC access it may be possible to get this from the data file type in the tag name.  I.e. N = INT, B = boolean, F = float etc.

xinthose, 2018-10-12:
That is why Modbus is beautiful in its own right.  It's a standard among different PLC's and devices.

kyle-github, 2018-10-12:
Yes, it seems like _most_ other protocols are simpler than this.   While I am sure that each of these "features" of CIP and its whole semi-related family of protocols makes sense in some context, the amount of one-off slight differences really make me annoyed.  It is as if AB's own engineers did not have complete understanding or documentation of their own protocol and so everyone did their part slightly differently.

xinthose, 2018-10-15:
Interesting observation.  Allen-Bradley is supposed to be the premier PLC manufacturer.

xinthose, 2018-10-17:
Nice.  I have not heard of L8x yet.

kyle-github, 2018-12-26:
This may require API changes.   It will only be done in either 2.0 or later.   Some sort of read-modify-write sequence will need to be supported.   Most of the supported protocols have such a command for integer types.

timyhac, 2019-04-03:
Hi Kyle - great work on this library!

What you've suggested about read-modify-write is how I have handled this in my wrapper class, but the problem is that I can run into race conditions with the PLC writing to the same bit in between the read and the write. This could separately be solved by having tags that are read-only to the PLC, but I don't have the luxury in my current project.

My two cents is to keep with the spirit of the AP, which is to be simple - and if the target PLC can't support a read-modify-write command directly, then we should avoid supporting it at all and leave it to the API user to find a way to prevent race conditions.

kyle-github, 2019-04-13:
Each PLC type needs to be broken out with little code sharing.   The commands for RMW for a PLC/5 may not work on a SLC 500.   Sigh.

With version 2.0.14, I have made the first big step to breaking the different PLC code out separately.   Each type of PLC can have its own slight protocol difference.

However, I think this will take a back burner to more important changes.

kyle-github, 2020-04-27:
This is now implemented for Control/CompactLogix and Micro8xx PLCs.

timyhac, 2020-07-04:
Nice work @kyle-github on supporting bools!

kyle-github, 2020-07-04:
Thanks.  There is still the whole PCCC-based PLC mess to handle, as well as Modbus.   But at least the concept is proven and the groundwork laid for those other PLC types.

kyle-github, 2020-08-23:
See also issue #187.

We have some options if we want to make the UX on this as seamless as possible:

1. The user passes the type and we figure it out from there   This might be the easiest to do, but would possibly fail if the field is within a UDT due to the packing rules.   And UDTs become interesting to handle.   This is effectively a large change to the API and would be a lot of work to change for wrappers.
2. The library, when doing the initial read, would read one element, then determine the type and use the elem_count as passed to determine what to do.   CIP, at least, returns the data type and we can parse that and determine the element size.  This takes longer, but would at least not change the effective API.

Note that the second option works for PCCC PLCs because we have the type information from the data file name.  The elem_size parameter is not needed at all for PCCC PLCs.

kyle-github, 2021-11-25:
We are going to punt the UDT stuff to the user application.   Tags all now support single bits including PCCC tags.

closed 2021-11-25 by kyle-github

ppibburr, 2022-08-24:
Am I correct that, a SLC 50x, bit writes would be unsafe?