libplctag/libplctag #87: Deploy problem

franklupo, 2019-01-22:
279/5000
Hello,
I do dll for windows on my pc. When I distribute I get the error in my application: (HRESULT exception: 0x8007007E).
The DLLs VCRUNTIME140D.DLL, UCRTBASED.DLL are required. In the previous version there are no problems.
Is it my problem?

best regards

franklupo, 2019-01-22:
Sorry my problem. I used debug version

Best regards

closed 2019-01-22 by franklupo

kyle-github, 2019-01-22:
Is there a way to combine the DLLs so that users only have one DLL to load?  I am not very familiar with Windows, but in Linux, this kind of thing is possible (not always easy, but possible).

I want to make sure that users can quickly get the library working.

franklupo, 2019-01-22:
one dll for linux and one for windows.
In windows using **cmake** and **visual studio**.
In linux **cmake** and **make**.

In release you can add the dll?