libplctag/libplctag #90: Revisit license - is it a problem for potential users?

kyle-github, 2019-02-06:
The current license is the GNU Lesser/Library GPL.   The intent was to make sure that any fixes that people did would be folded back into the main distribution of the library.  

Is this license a problem for potential users?   If so, what would a better license be (BSD, MIT?) and why?

scontini76, 2019-02-06:
LGPL seems fine to me, it is very suitable for commercial use and assure improvements from developer who are using it. I support your choice for the reason you wrote.

thatcadguy, 2019-02-10:
@kyle-github LGPLed code can be used in closed-source commercial software by dynamic linking, personally I would not make it more permissive than it already is. It's generous enough of you to have this open-source.

xinthose, 2019-03-06:
How does this work with the company you work for?  Do they like open source because it helps improve the library, thus your company's applications?

kyle-github, 2019-03-07:
OmanTek is generally supportive of open source.   A lot of problems in the automation world are caused because things are proprietary and cannot be fixed or maintained.

As you mentioned, the library's users find edge cases and push the library in ways that end up benefiting everyone.   So it is generally a win.

sazindev, 2019-05-03:
Hi, I have been looking at your library and it looks like it would be very usable for us. I would still have to test and prototype something, but the current license is indeed problematic for us. Our closed-source product relies on statically linking everything into one executable and is hence incompatible with LGPL code.

How open are you and your contributors to may-be dual-licensing the code, adding e.g. an Apache 2.0 or MIT license to it?

Even though we are closed-source, we still contribute back our bug-fixes and improvements to third-party libraries which we build on. Thanks for considering this.

kyle-github, 2019-05-27:
Sorry for the delay.

My understanding of the LGPL is that even with static linkage, you can still keep your part proprietary.  It is more difficult but still possible.   

[GPL Tutorial](https://copyleft.org/guide/)

sampaioletti, 2020-03-22:
Kyle, I'm not a lawyer either but if the intent of the library is to allow commercial use but still have benefit back to the community it seems the best option might be MPL v2. Purely due to the nature of libplc and its potential use on embedded systems that don't always allow/prefer dynamic linking for a number of reasons.

Really boils down to how you want the lib used. If you are interested in commercial use then I'll give you my summary based on my research opinion/company policy, ordered by preference (from a commercial mindset) first.

- MIT
  - is the most permissive we use in our commercial software also the easiest for a commercial company to use because it comes with very little requirements.
  - offers you protection but relies on users good nature for contributions
- Apache
  - similar to MIT but we have to list our modifications and can't name similarly
MPL use however your want but must contribute back modifications
  - its more complicated so our company policy requires a valid use case document approved by our council and management
- MPL
  - we have recently started using libraries under this license because it seems like a good mix basically LGPL with little restriction on use
  - company policy says we have to make a document showing MIT/Apache licensed alternatives and make a case for why we should use it.
- LGPL
  - company policy says we have to meet all the requirements of MPL and requires legal review and Management, plus a estimate on time required to develop a library in house in-lieu of using it
  - we really try not to ever use anything LGPL, we actively contribute back to all open source projects that we use (regardless of license), but using software correctly under the LGPL is complicated and generally the extra requirements are more work to keep up with than rolling something in-house

So given the above it could be any version of dual licensing or a paid license option for more permissive/commercial use

Any licensing is only as good as your ability to enforce it, so if your intent is to see more widespread use as appose to enforcing an open source license it makes sense to go more permissive (in my opinion)

Again, thanks for the great lib...

sampaioletti, 2020-04-04:
Finally remembered where I had seen the linking exception for a GPL license

https://github.com/OpenEtherCATsociety/SOEM/blob/master/LICENSE

kyle-github, 2020-04-29:
Thanks for the comments @sampaioletti.   I am seriously thinking about the MPL, but the linking exception is also good.   I have had people ask about that in the past.

I am definitely not a lawyer, so I want to make sure that we are definitely protected, first.  After that, I want to make sure that users have incentive to return fixes and updates to the library and their customers.

sampaioletti, 2020-04-29:
Yeah fair enough, its not an easy problem from any perspective I can appreciate the moral and logical implications. I think any licensing decision goes back to 'a license is only as good as your willingness/ability to enforce it'

Good people will try to do good, bad will try to do bad. Good luck but thanks again for being open about it.

kyle-github, 2020-05-23:
I am leaning toward the MPL 2.0 at this point.   We do not get much code back in the core library and with the new organization.   What we do get are wrappers.   With the new GitHub organization, the core library and the wrappers are separate so the wrappers can have different licenses.

The MPL is compatible with the GPL and LGPL, so existing wrappers should continue to be allowed.

kyle-github, 2020-05-25:
A few users have brought up the challenges of linking LGPL code into single binaries. 
  
Single binary programs are becoming more and more common.   Go really pushes this idea but even Java is getting support for this with Graal.  Apparently .Net has support for this as well now.

With that in mind, the pain of linking an LGPL library becomes a lot higher.   If there is no dynamic linking, then it is much more difficult to have more restrictive permissions on the application source and GPL-like restrictions on the library.

I will be moving to the MPL license as soon as possible once I figure out how to do it.  The wrappers will continue to use whatever license they had when they were contributed.   As part of this, the wrappers will be split out in the near future into separate repo/projects within the libplctag GitHub org.

sampaioletti, 2020-05-25:
Cool I think that's a good move, if your worried about past users you could dual license portions of it in the transition (or perpetually), I've seen that done before. Thanks for listening, I look forward to contributing to the project.

kyle-github, 2020-05-25:
Dual licensing is a good idea.  That would ease some of the transition.

kyle-github, 2020-05-25:
Dual licensing seems to be the key.   I was trying to figure out how to manage the existing wrappers as many were contributed without any explicit license.   The assumption must be that they were compatible with the old license and thus moving to the MPL would mean I would need to remove the wrapper.   But dual licensing seems to solve that problem.  Thanks for the idea!

Getting the release ready now.

kyle-github, 2020-06-12:
The core C library has been released as dual MPL/LGPL! 

For now I will be closing this issue.  I think the MPL is about as far as I want to go toward BSD-style licenses.

Thanks to @sampaioletti for the idea of dual licensing the library.   I could not figure out how to make sure that existing projects that used the LGPL or GPL would be able to keep working with just the MPL.   There are some instructions and ideas on the Mozilla site explaining the MPL, but a dual license seems like a cleaner way to do this.

closed 2020-06-12 by kyle-github

sampaioletti, 2020-06-12:
Awesome! Thanks for listening and hopefully this can help expand adoption. It's a great project!